Koenkk/zigbee2mqtt #5059: SNZB-03 identified as SNZB-02

benpoe, 2020-11-23:
just updated to newest version, because my newly bought SNZB-03 weren't showing any activities and were identified as 666666

however in  1.16.1 is SNZB-03 identified as SNZB-02

```
igbee2MQTT:debug 2020-11-23 00:13:13: Device 'motion1' announced itself
Zigbee2MQTT:info  2020-11-23 00:13:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"motion1"},"type":"device_announced"}'
```

`{"dateCode":"20191107","description":"Temperature and humidity sensor","friendly_name":"motion1","hardwareVersion":1,"ieeeAddr":"0x00124b002242e9fa","lastSeen":1606090197065,"manufacturerID":0,"manufacturerName":"eWeLink","model":"SNZB-02","modelID":"66666","networkAddress":23761,"powerSource":"Battery","type":"EndDevice","vendor":"SONOFF"}],"type":"devices"}'
`

Koenkk, 2020-11-23:
May already have been fixed in the latest dev branch, if not please provide your `data/database.db` entry of this device.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Hveen28, 2020-11-28:
Indeed it is fixed in latest dev branch. I installed the edge version and the SONOFF is identified as motion sensor. Can you tell when this is fixed in the stable version?

Koenkk, 2020-11-28:
Next z2m release will be on 1 december.

closed 2020-11-28 by Koenkk