Koenkk/zigbee2mqtt #5060: PTVO.RGB.LED Support

akinomeroglu, 2020-11-23:
## Information about the device + link

this device refreshed with ptvo.rgb.led firmware

## data/database.db entry of the device

{"id":4,"type":"Router","ieeeAddr":"0x00124b0021ee9890","nwkAddr":59407,"manufId":4447,"manufName":"ptvo.info","powerSource":"Mains (single phase)","modelId":"ptvo.rgb.led","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2821,12,6,8],"outClusterList":[0,2821,6,8],"clusters":{"genBasic":{"attributes":{"modelId":"ptvo.rgb.led","manufacturerName":"ptvo.info","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019-06-21","swBuildId":"2019-06-21"}},"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[12,6,8],"outClusterList":[6,8],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[12,6,8],"outClusterList":[6,8],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"2019-06-21","swBuildId":"2019-06-21","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1606079818059}

akinomeroglu, 2020-11-23:
@Koenkk hi,

As far as I understand, ptvo firmware is highly customizable and thus configurations may be very different. I think that you can add support for a reference design for RGB led's and people can follow this reference design for out of box support in zigbee2mqtt and HA

I will be very happy if you share your thoughts about this issue.

Koenkk, 2020-11-23:
@ptvoinfo can you add support for this?

ptvoinfo, 2020-11-24:
@akinomeroglu, @Koenkk 

Hi, I didn't understand what you need to add in Zigbee2MQTT. The standard converter supports all features of the firmware.

FarukGN, 2020-11-24:
hi @ptvoinfo 

There is no problem with the switch configuration. Zigbee can find it without problems but how should the correct configuration be for RGB led ?

I couldn't find the correct configuration. zigbee issues unsupported device alert when there is no correct configuration

ptvoinfo, 2020-11-25:
Why did you change the "modelId" option in the firmware settings to "ptvo.rgb.led"? It is on the "Expert" tab, and you should know that if you change it, you should create a custom converter in Zigbee2MQTT. Anyone else cannot do it for you.

FarukGN, 2020-11-25:
I changed the modelid to try it out. I know I have to use a converter when the model changes. I did it to simplify and not waste time with the converter. But it didn't work.  Here's the simple question I need right now. What should be modelid for RGB led control ?(Turn it on and off, change color and etc). @ptvoinfo Thank you

ptvoinfo, 2020-11-25:
ModelId: ptvo.switch
I don't know how you connected to RGB led control. Using PWM?
Here are all commands:
https://ptvo.info/zigbee-configurable-firmware-features/pwm/

FarukGN, 2020-11-26:
Soasibo bolsoe @ptvoinfo 

I will try today

FarukGN, 2020-12-05:
Thanks for your support. I wanted to control 20, 25 meter long rgb led by pwm. 

I loaded firmware into cc 2530.it works property. But iam only able to control 3 main colors. (Red, blue, yellow) it cant be fully controlled by pwm. Firmware is perfect for turning on and off zigbee nets. I cant say that rgb is fully led controller because of home assistant cant recognize firmware as rgb led controller.

Thank you so much

@ptvoinfo @Koenkk @akinomeroglu

closed 2021-01-12 by github-actions[bot]