Koenkk/zigbee2mqtt #5061: In-Wall Zigbee 3.0 power outlet with dual USB charger sockets

FoxConsult, 2020-11-23:
## Information about the device + link
Device:			SM-PW801-EZ
Description:		In-Wall Zigbee 3.0 dual USB wireless socket

Manufacturer Link:	http://www.xenon.cn/in-wall-outlet98/sm-pw801-e--.html

Purchased on Aliexpress:	https://www.aliexpress.com/item/4000528906478.html

## data/database.db entry of the device
Device registers itself as Moes ZK-EU-2U (https://www.zigbee2mqtt.io/devices/ZK-EU-2U.html)

USB port reports "state_undefined":"ON". I can issue On or Off to the power outlet but I cannot control the USB outlets On or Off from Domoticz.

'zigbee2mqtt/0xec1bbdfffe8a20ed', payload '{"last_seen":"2020-11-20T14:15:11+01:00","linkquality":47,"state_l1":"OFF","state_undefined":"ON"}'

zigbee2mqtt web GUI:

Router
Description		Zigbee 3.0 dual USB wireless socket plug
Support status		Supported
Vendor			Moes
Model			ZK-EU-2U
Interview completed	Yes

From log running "sudo journalctl -u zigbee2mqtt.service -f" and clicking Domoticz switch number 2 (USB ports On/Off):

Nov 23 11:47:53 raspberrypi-4-lan npm[30880]: Zigbee2MQTT:error 2020-11-23 11:47:53: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:183:52)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:384:44)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:275:14)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at MQTT.emit (events.js:314:20)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:103:18)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at MqttClient.emit (events.js:314:20)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1271:12)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:410:12)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:321:12)
Nov 23 11:47:53 raspberrypi-4-lan npm[30880]:     at Writable.writable._write (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:335:5))

Versions used:
zigbee-herdsman 0.13.31
Zigbee2MQTT version 1.16.1-dev (commit #02a945d)

FoxConsult, 2020-12-24:
Still a request for new device support.

markheloking, 2020-12-31:
Would also like to see this device supported.

svrooij, 2021-01-06:
If I read your description it might be just the same device, but they added two USB ports. Does it even have a switchable circuit?

FoxConsult, 2021-01-07:
The Moes ZK-EU-2U (https://www.zigbee2mqtt.io/devices/ZK-EU-2U.html) also has two USB ports so the SM-PW801-EZ (http://www.xenon.cn/in-wall-outlet98/sm-pw801-e--.html) is functionally identical to the Moes ZK-EU-2U (https://www.moeshouse.com/products/eu-zigbee3-0-dual-usb-wireless-socket-plug-smartthings-app-remote-control-echo-plus-voice-control-work-with-alexa-google-home).

My device has one power outlet and two USB ports and a push button that can be used to switch the power outlet on and off and both USB charger outlets on and off at the same time. This is by design. See https://www.moeshouse.com/products/eu-zigbee3-0-dual-usb-wireless-socket-plug-smartthings-app-remote-control-echo-plus-voice-control-work-with-alexa-google-home

From Domoticz however (zigbee2mqtt gateway), I can turn the power outlet on and off independently from the USB outlets. However, I cannot control the USB ports on or off despite the unit reporting back to zigbe2mqtt.

closed 2021-03-18 by github-actions[bot]