Koenkk/zigbee2mqtt #5064: Bitron AV2010/26: Dimmable Wireless Socket

fabsss, 2020-11-23:
## Information about the device + link
Bitron 2,5A Dimmable Socket
- zigbeeModel: ['902010/26']
Features: 
- On/Off and Dimming
- Power Meter
- Button (Default: Select Brightness levels, on, off, Pairing)

Purchase-Link: https://www.conrad.at/de/p/bitron-video-steckdose-av2010-26-902010-26-1095388.html#productHighlights
Manual: https://asset.conrad.com/media10/add/160267/c1/-/gl/001095388ML01/bedienungsanleitung-1095388-bitron-video-steckdose-av201026-90201026.pdf

## data/database.db entry of the device
not yet available (similar model without dimming: zigbeeModel: ['902010/25']

I tried to create an external converter which recognizes the device by copying the entry of the similar model /25 but are just not capable to get even the simple functions (on/off, power meter, without dimming) working. I just don't understand enough of the code to contribute. I can provide logging data if someone needs them to implement the new device, though. 

It would be really great if someone had a look on the device since there aren't so many zigbee dimmable sockets around and I think its quite useful.

Regards
Fabian

closed 2020-12-31 by github-actions[bot]

gebauer, 2021-01-06:
Hi fabss, 

Have you managed to register the bitron dimmer? 
Best, 
Jan

fabsss, 2021-01-06:
> Hi fabss,
> 
> Have you managed to register the bitron dimmer?
> Best,
> Jan

Hi Jan,

nope, unfortunately not. :(

Regards
Fabian

WordsOfMe, 2022-03-26:
I just found the SMaBiT (Bitron Video) AV2010/26 perfectly registering under zigbee2mqtt 1.24.0.