Koenkk/zigbee2mqtt #5065: Automatic edge build trigger for `zigbee2mqtt/hassio-zigbee2mqtt`

ciotlosm, 2020-11-23:
## Is your feature request related to a problem? Please describe
Automatic edge build trigger for `zigbee2mqtt/hassio-zigbee2mqtt` when dev gets updated. 

## Describe the solution you'd like
Automatic edge build trigger for `zigbee2mqtt/hassio-zigbee2mqtt` when dev gets updated. Similar to:
```
    - name: Trigger danielwelch/hassio-zigbee2mqtt build
      if: github.ref == 'refs/heads/dev' && github.event_name == 'push'
      run: |
        curl -s \
          --request POST \
          --header "Content-Type: application/json" \
          --data '{"definition": {"id": 1}}' \
          -u ${{ secrets.HASSIO_TRIGGER_SECRET }} \
          "https://dev.azure.com/danielwelch2101/hassio-zigbee2mqtt/_apis/build/builds?api-version=5.1"
```

Discussed with @danielwelch as well that we will try migration to the new org for the repo.

Koenkk, 2020-11-23:
Can you make a PR for https://github.com/Koenkk/zigbee2mqtt/blob/4978d09eb1e18bf7e83a15e8087b2fe226e069bd/.github/workflows/ci.yml#L71 ? Note that you need to give me admin access to the azure pipeline.

ciotlosm, 2020-11-23:
@Koenkk you should be a member of that org and have admin access. I will double check and make a PR. I'm a bit unsure on the API URL, but I will try.

ciotlosm, 2020-11-23:
I've checked access and I think it should be there:
[image]

closed 2020-12-24 by ciotlosm