Koenkk/zigbee2mqtt #5069: Ghost devices appeared after changing device names

wojciechczyz, 2020-11-24:
## What happened

After I was changing names of Devices/Entities in Home Assistant
I see ghost devices on zigbee2mqtt assistant  like:

bridge/event | NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?
bridge/info | NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?
bridge/logging | NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?

If I reset state of zigbee2mqtt assistant  I still see them. I tried restarting  Zigbee2Mqtt and Home Assistant - no improvement.

## What did you expect to happen

Zigbee2MQTT version:
1.16.1

Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
CC2652R1

Adapter firmware version:
Latest

fnfid, 2020-11-24:
Since last update "zigbee2mqtt/bridge/config/devices/get" dosn't echo a list of devices to the bridge log anymore (did all the time). Requesting a group list "zigbee2mqtt/bridge/config/groups" still work. I don't know the answer, but think in the future we have to rework some of our code to more recent syntax, as i understand this will be legacy stuff soon.

escees, 2020-12-17:
I'm experiencing the same behavior just after fresh install of latest HA on Debian 10 using Docker.

wojciechczyz, 2021-01-17:
I see those “devices” on the device list, using zigbee2mqtt companion app. They are all listed in red. When at some point switched coordinator stick type and lost all devices, they still preserved

closed 2021-02-26 by github-actions[bot]

user514, 2021-03-11:
I just did a complete new install on a new machine this days, latest HA supervised, latest zigbee2mqtt, latest zigbee2mqtt assistant, did some pairing and after installing assistant, I see:
-bridge/info
-bridge/logging
...with red labels: NO COMPONENTS in the device list of the assistant. When entering those devices in assistant I see a bad formatted web page with options to rename, remove device, etc. i suppose I will try to remove them, but not sure if that will broke something.

wojciechczyz, 2021-03-11:
> I just did a complete new install on a new machine this days, latest HA supervised, latest zigbee2mqtt, latest zigbee2mqtt assistant, did some pairing and after installing assistant, I see: [...]

I still see them on my zigbee2mqtt assistant, not sure if this is issue of zigbee2mqtt or zigbee2mqtt assistant. 
Will you reopen this case?

faman74, 2021-03-11:
same issue here

redpizza69, 2021-03-17:
I see the same, but I also see bridge/event. Full list of ghost devices:
bridge/event
bridge/info
bridge/logging

maciey, 2022-02-06:
Same here...
How to remove those 'not supported' devices ?

[image]