Koenkk/zigbee2mqtt #5072: Power plug Immax Neo started to report current and voltage entities which are not supported.

daha76, 2020-11-24:
## What happened
Power plug Immax Neo started to report current and voltage entities, but obviously this power plug does not support it. Values are always 0. I tried to delete the plug from ZigBee network and pair it back, but even than the current and voltage is reported. Reconfigure didn't help either.

## What did you expect to happen
Not supported attributes are not reported.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: Zigbee2MQTT 1.16.1
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-11-24:
According to @hanzz it works for some: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1002

salopette, 2020-11-24:
with me the values are not there either.

[image]
[image]
[image]

daha76, 2020-11-25:
> According to @hanzz it works for some: [Koenkk/zigbee-herdsman-converters#1002](https://github.com/Koenkk/zigbee-herdsman-converters/pull/1002)

The discussion there is about power measurement in Watts. This works for me, but for some reason zigbee2mqtt started to report also AC Voltage and Current which seems to be not supported.

sheppi72, 2020-11-28:
Same for me, only the power value reported.

```
Firmware build date   20190709-164
Firmware version      2.0
Vendor                Immax
Model                 07048L
```

salopette, 2020-12-29:
.db

`{"id":31,"type":"Router","ieeeAddr":"0xccccccfffeeaae60","nwkAddr":62832,"manufId":4151,"manufName":"Immax","powerSource":"Mains (single phase)","modelId":"Plug-230V-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820,64642],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug-230V-ZB3.0","manufacturerName":"Immax","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190709-164","swBuildId":"2.0"}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"activePower":0}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca168ea","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca168ea","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190709-164","swBuildId":"2.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":9},"lastSeen":1609221249507}
`

salopette, 2020-12-29:
DEBUG: Received Zigbee message from 'Baum Licht', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
DEBUG: Received MQTT message on 'zigbee2mqtt/Baum Licht/set' with data '{"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:26.975Z","linkquality":75,"power":0,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:26.983Z","linkquality":78,"power":0,"state":"ON"}'
DEBUG: Received Zigbee message from 'Baum Licht', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":3}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Baum Licht', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0
DEBUG: Received MQTT message on 'zigbee2mqtt/Baum Licht/set' with data '{"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:42.945Z","linkquality":75,"power":7,"state":"OFF"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:42.951Z","linkquality":75,"power":7,"state":"OFF"}'
DEBUG: Received Zigbee message from 'Baum Licht', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
DEBUG: Received MQTT message on 'zigbee2mqtt/Baum Licht/set' with data '{"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:50.833Z","linkquality":75,"power":0,"state":"ON"}'
INFO: MQTT publish: topic 'zigbee2mqtt/Baum Licht', payload '{"last_seen":"2020-12-29T06:24:50.839Z","linkquality":72,"power":0,"state":"ON"}'
DEBUG: Received MQTT message on 'zigbee2mqtt/Baum Licht/get' with data '{"power":""}'
DEBUG: Received Zigbee message from 'Baum Licht', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":3}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from 'Baum Licht', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":7}' from endpoint 1 with groupID 0

Koenkk, 2020-12-29:
@hanzz can you confirm that your plug supports current and voltage reportings? And if so share your data/database.db entry here?

closed 2021-01-06 by github-actions[bot]

salopette, 2021-01-06:
please do not close, it is not repaired!

daha76, 2021-01-06:
I just ignore those values.

Koenkk, 2021-01-06:
Since I did get no response on https://github.com/Koenkk/zigbee2mqtt/issues/5072#issuecomment-752020610 I've removed them.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)