Koenkk/zigbee2mqtt #5073: Sonoff SNZB-01 identified as eWeLink WB-01

thehijacker, 2020-11-24:
Hello,

I have latest dev build 1.16.1-dev. Received my Sonoff SNZB-01 switch today and when trying to pair it I get following error:

Zigbee model 'WB-01' and manufacturer name 'eWeLink' is NOT supported

Restarting Z2M does not help. I tried to remove it, but it fails. I can only force remove it and pair it again, but get the same error message.

My coordinator is Zzh. No other issues with my Zigbee network.

Let me know if I can provide any additional information to help solve this issue.

Thank you.

Koenkk, 2020-11-24:
Added, 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-24 by Koenkk

thehijacker, 2020-11-25:
Hello. I can confirm that with latest dev branch it is correctly recognised and working.

Thank you.