Koenkk/zigbee2mqtt #5074: IKEA FYRTUR - invert_cover: configuration not working.

Twit123, 2020-11-24:
## What happened
I have 2 identical IKEA FYRTUR blinds (E1757), bought at the same time and both on the same firmware (2.2.009) but strangely one of the blinds is reversed showing open = 0, closed = 100 (The other is open = 100, closed = 0) . I've unpaired and repaired them but still the same. This in tern affects the up/down arrows in HA as it thinks one is up and one is down when in fact both are down and the up arrow on the wrong one is greyed out.

I'm 99.9% sure this is a fault with the blind but though that using the "invert_cover: " option as listed on the device page would fix the issue but it has no effect. Unless I'm doing something really stupid.

`'0x000d6ffffe903a2d':
  friendly_name: FYRTUR_Lounge_Front
  invert cover: true
  retain: false`

If it matters I have my devices in a separate "devices.yaml" file and the log shows the option on start up.

`"0x000d6ffffe903a2d":{"friendly_name":"FYRTUR_Lounge_Front","invert cover":true,"retain":false}`

## What did you expect to happen

I though the 100% and 0% would be reversed

## How to reproduce it (minimal and precise)

Insert "insert cover: true" in device specific yaml and get no effect

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version: 20190608

Twit123, 2020-11-24:
Please ignore me, I've spent days trying to get this to work and suddenly it is, I don't know why but I'm happy.

closed 2020-11-24 by Twit123

beasthouse-au, 2021-05-24:
> Please ignore me, I've spent days trying to get this to work and suddenly it is, I don't know why but I'm happy.

Any idea what the fix was now? I am having this same issue. The blind reports as closed when open and open when closed.

barcelona92, 2021-12-20:
> > Please ignore me, I've spent days trying to get this to work and suddenly it is, I don't know why but I'm happy.
> 
> Any idea what the fix was now? I am having this same issue. The blind reports as closed when open and open when closed.

The same problem

DbilliT, 2024-11-15:
Hey, do you know how to fix it ?
I have the same issue ...