Koenkk/zigbee2mqtt #5081: Can you add ROBB Smart Plug with code  rob_200-017-0

Kees-1953, 2020-11-25:
I bougth the very new Smartplug rob_200-017-0  from robbshop.nl. This Smartplug is a 3680 Watt Zigbee plug with energy measurement.

closed 2021-01-03 by github-actions[bot]

Ramshield, 2021-08-30:
Please see https://github.com/Koenkk/zigbee-herdsman-converters/pull/2973 @Kees-1953

Chamondgl, 2021-12-15:
Hi @Kees-1953, Is your device correctly reporting kWh measurements?

ROBBshop has released hardware revision 1 of the device and I've just added it to the branch pending some final code verifications on line lenghth. Temperature and kWh reporting are confirmed to work locally. However, using the code for rev 0 didn't work for rev 1 didn't work for me.
Might be worth checking out and updating rev 0 code and test locally to see if it works for you in case you have this problem.

Cheers!

turnhob, 2022-01-06:
I do have the ROB_200_017-0 but when i add it to zigbee2mqtt (with a Conbee II) it tells me its not supported.
The model name that shows is: HK-PLUG-A and the manufacturar is Sunricher.
So i don't understand why it does that (I assume ROBB did use Sunricher and uses his own brand). But now im not able to use my Smart Plug. 
Can U help me how to fix this. Im using zigbee2mqtt with home assistant and tried to create a converter file, but when i follow the instructions my add-on wont start anymore when i changed the configuration (as mentioned on your page).

Besides that.. Moving from deconz to zigbee2mqtt was the best thing i could have done.

Ramshield, 2022-01-09:
Hi @turnhob. That's an easy fix! https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/robb.js#L147 
add a new dict with `whiteLabel`, with `vendor` and `model`.  An example would be: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L842