Koenkk/zigbee2mqtt #5082: SEA801-Z01 Unknown device.

Ryanmt2, 2020-11-25:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Ryanmt2, 2020-11-25:
Closing, looks like I am behind the latest devices.js and this is already fixed

closed 2020-11-25 by Ryanmt2

seidler2547, 2020-11-26:
I got the same device today, and while I see the device in devices.js, it still doesn't work. Anything I am missing, or do I just need to wait until the change is released?

Ryanmt2, 2020-11-26:
Depends, the has been supported for a while but there is an unreleased change for some different identifiers eg `_TZE200_c88teujp`

Here is the latest copy/changes
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L13513