Koenkk/zigbee2mqtt #5083: Unresponsiveness, unreliability and broken network

jd1900, 2020-11-25:
# Bug Report
## What happened
I have a testing network in a different place where I've tested all my zigbee devices and they work perfectly.
Most of the devices are within a group and I have several actuators bound to those groups.

However, I've moved to a new house where I've connected all the ZigBee at once, and..., then the errors came.
When zigbee2mqtt is not connected I can use all of the remotes bound to groups and the response is great.
However, once I start zigbee2mqtt problems start to arise.
Bindings start to fail, the work randomly (not always) and very laggy.
Zigbee network seems to be VERY slow. It takes sometimes a few seconds to deliver a command. Sometimes it just timeout (10000ms timeout according to logs), and also I get same 'no network route'.

Devices appear and disappear randomly. 

So..., in summary, when I start Zigbee2Mqtt:
- Network unreliability.
- Very laggy.
- Works randomly.
- Lots of timeouts.
- "Breaks" bindings.
- No network routes.

When I finish Zigbee2Mqtt, bindings start working fine again.

I have several devices (around 30) from different brands, but currently I was only connecting (basically) from two different companies:
- Gledopto.
- Legrand.

I've tried disconnecting all of the Gledopto devices, but it still fails.

Network mesh doesn't seem to work.

I'm on default channel (11). 

## What did you expect to happen
Work fine.

## How to reproduce it (minimal and precise)
¿?

## Debug Info
Zigbee2mqtt version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20191106.zip

jd1900, 2020-11-26:
I've updated the firmware version to the latest one (CC26X2R1_20201026) and still no changes.

I've also looked into the frequency interferences, and the channel seems to be fine:
[image]
[image]

[image]
[image]

Any help of how to have a reliable network?
Once I start Zigbee2Mqtt everything is a nightmare and nothing works.

Koenkk, 2020-11-26:
Actually your channel is not great at all, zigbee channel != wifi channel, wifi channel 1 interferes a lot with zigbee channel 11, see: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html . Try with channel 25 (note changing channel requires repairing everything)

I also recommend switching to https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin because it increases the transmit power.

jd1900, 2020-11-26:
Thank you. I'll try that.

If I switch back to the old channel, will I still keep already paired devices paired?

Koenkk, 2020-11-26:
No, once you switch channel everything is cleared.

jd1900, 2020-11-26:
Thank you again.

Two more questions:
- Why channel 25 and not channel 26? Channel 26 should be "clearer", right?
- Why when the coordinator and Zigbee2mqtt is not up, everything works smoothly? I have a group with 7 devices, all around the house (for the shutter), and they all works smoothly. When the coordinator is up, it doesn't work. Why is that? 
When a device is not connected to the coordinator, does it use a different channel?
Or just it routes differently? 
I don't fully get why it works fine without coordinator and just binding, but everything fails when the coordinator is up.

Koenkk, 2020-11-27:
- Some devices might not work on channel 26
- I also don't understand why this does not work (there should be no difference)

jd1900, 2020-12-09:
Thank you.

After changing the channel to 25, I had to go back due to incompatibilities with Legrand devices and a channel different from 11.
But changing to that firmware got me a little bit of better coverage. I've bough an ZZH and I hope that with the antenna I can have a better user experience.

closed 2021-01-16 by github-actions[bot]