Koenkk/zigbee2mqtt #5084: Zigbee stick does not create /dev/ttyACM0 on USB passthrough

joshuakoh1, 2020-11-26:
When I pass the USB device by device or by port to a VM, /dev/ttyACM0 disappears from proxmox host. But it does not appear on the VM at all. I only see the device under lsusb but no corresponding /dev/ttyACM0 device node.

Remove the passthrough, it shows up again under the proxmox host. 

The dev node should pop up on the VM level but does not. 

Proxmox has an option to pass a serial device directly but the USB stick only generates /dev/ttyACM0 but no /dev/ttyS0 so I cannot pass it through serialX. 

Since zigbee2mqtt only reads from /dev/ttyACM0, even if I managed to pass it as /dev/ttyS0, I would still not be able to access it.

piit79, 2020-11-26:
Hey, I'm afraid this is unrelated to zigbee2mqtt...

> Since zigbee2mqtt only reads from /dev/ttyACM0...

That's not correct - you can have zigbee2mqtt connect to _any_ serial device. For example, I'm using the following configuration:
```yaml
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
```

joshuakoh1, 2020-11-26:
@piit79 I opened this because I'm not sure if it's related to the zigbee-herdsman firmware on the stick that's not allowing it to create the serial port after passthrough.

joshuakoh1, 2020-12-13:
Still looking for a solution. When passed through, the USB device does not fire up cdc_adm. 

```
ubuntu@OpenHAB3:~$ lsusb
Bus 001 Device 002: ID 0451:bef3 Texas Instruments, Inc. CC1352R1 Launchpad
ubuntu@OpenHAB3:~$ dmesg
[ 1594.034672] usb 1-1: USB disconnect, device number 2
[ 1600.478866] usb 1-1: new high-speed USB device number 3 using ehci-pci
[ 1600.637759] usb 1-1: New USB device found, idVendor=0451, idProduct=bef3, bcdDevice= 1.00
[ 1600.637762] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1600.637767] usb 1-1: Product: XDS110 (03.00.00.13) Embed with CMSIS-DAP
[ 1600.637768] usb 1-1: Manufacturer: Texas Instruments
[ 1600.637774] usb 1-1: SerialNumber: L43002MO
[ 1600.643912] hid-generic 0003:0451:BEF3.0002: hiddev0,hidraw0: USB HID v1.11 Device [Texas Instruments XDS110 (03.00.00.13) Embed with CMSIS-DAP] on usb-0000:00:1d.7-1/input5
[ 1862.034842] usb 1-1: USB disconnect, device number 3
[ 1886.466925] usb 1-1: new high-speed USB device number 4 using ehci-pci
[ 1886.626210] usb 1-1: New USB device found, idVendor=0451, idProduct=bef3, bcdDevice= 1.00
[ 1886.626217] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1886.626223] usb 1-1: Product: XDS110 (03.00.00.13) Embed with CMSIS-DAP
[ 1886.626224] usb 1-1: Manufacturer: Texas Instruments
[ 1886.626225] usb 1-1: SerialNumber: L43002MO
[ 1886.632709] hid-generic 0003:0451:BEF3.0003: hiddev0,hidraw0: USB HID v1.11 Device [Texas Instruments XDS110 (03.00.00.13) Embed with CMSIS-DAP] on usb-0000:00:1d.7-1/input5
ubuntu@OpenHAB3:~$ modprobe cdc_acm
modprobe: FATAL: Module cdc_acm not found in directory /lib/modules/5.4.0-56-generic
```

ceesvanegmond, 2021-01-12:
@joshuakoh1 Have you fixed this?

joshuakoh1, 2021-01-13:
@ceesvanegmond Fixed? No. Bandaid solution? Yes. I passed /dev/ttyACM0 from the host to /dev/ttyS0 on the VM then made z2m read from there. Still no fix for the issue of the stick not creating any serial node on passthrough.

piit79, 2021-01-13:
> Still no fix for the issue of the stick not creating any serial node on passthrough.

But this is unrelated to zigbee2mqtt... Possibly related to firmware, but this is not the firmware repository.

closed 2021-02-21 by github-actions[bot]