Koenkk/zigbee2mqtt #5086: Add Support for Philips Hue White E27 LED-Bulb Filament Giant Globe LWO003

mozdi, 2020-11-26:
**## Information about the device + link**
Philips Hue White E27 LED-Bulb Filament Giant Globe LWO003
https://www.amazon.de/Philips-Hue-Vintage-Design-warmwei%C3%9Fes-kompatibel/dp/B08CXYWPK7

..

**## data/database.db entry of the device**
{"id":9,"type":"Router","ieeeAddr":"0x001788XXXXXXXXXX","nwkAddr":8263,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LWO003","epList":[11,242],"endpoints":{"11":{"profId":260,"epId$

..
I tried to fill to devices.js but I´m not pretty sure if this is all correct:
```
{
    zigbeeModel: ['LWO003'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'LWO003', // Vendor model number, look on the device for a model number
    vendor: 'Signify Netherlands B.V.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Philips Hue White E27 LED-Bulb Filament Giant Globe', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off, birghtness', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

What is required in fromZigbee and to Zigbee?

**## Other information**
Here´s my Debug Log:
```
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:45: Device '0x001788XXXXXXXXXX' joined
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:45: Starting interview of '0x001788XXXXXXXXXX'
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788XXXXXXXXXX"},"type":"device_connected"}'
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788XXXXXXXXXX"},"type":"pairing"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:45: Device '0x001788XXXXXXXXXX' announced itself
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x001788XXXXXXXXXX"},"type":"device_announced"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:45: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"modelId":"LWO003"}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:45: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:45: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Signify Netherlands B.V."}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:45: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"appVersion":2}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200124"}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Received Zigbee message from '0x001788XXXXXXXXXX', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1.65.9_hB3217DF4"}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:debug 2020-11-26 10:59:46: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:46: Successfully interviewed '0x001788XXXXXXXXXX', device has successfully been paired
zigbee2mqtt      | Zigbee2MQTT:warn  2020-11-26 10:59:46: Device '0x001788XXXXXXXXXX' with Zigbee model 'LWO003' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt      | Zigbee2MQTT:info  2020-11-26 10:59:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001788XXXXXXXXXX","supported":false},"type":"pairing"}'
```

mozdi, 2020-12-15:
Hi together,

unfortunately I don´t get the bulb running with zigbee2mqtt.

Even with latest release 1.16.2 it does not work.

```
zigbee2mqtt      | Zigbee2MQTT:warn  2020-12-15 11:13:34: Device '0x0017880108ee9af9' with Zigbee model 'LWO003' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt      | Zigbee2MQTT:info  2020-12-15 11:13:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x0017880108ee9af9","supported":false},"type":"pairing"}'
```

It would be great if someone could assist me in adding the device.

Koenkk, 2020-12-15:
Added support for this device.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-15 by Koenkk