Koenkk/zigbee2mqtt #5089: support for Melinera Smart Led String lights (sold by LIDL).

Ruftie, 2020-11-26:
Device is reconsigned as TS0601 and no further actions are possible.

Koenkk, 2020-11-26:
Just merged: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1799

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-26 by Koenkk

Ruftie, 2020-11-26:
thanks will update to dev tomorrow.

Ruftie, 2020-11-27:
updated to dev still the same problem.

meisterfuu, 2020-11-27:
Was just testing this on dev. Unsupported device. Does it still take some time, or do you need some more information?

Ruftie, 2020-11-27:
in domoticz it is visible but cant do anything with the device.
[image]

meisterfuu, 2020-11-27:
After pairing it again a few times it seems to get recognised correctly.

Ruftie, 2020-11-27:
> 
> 
> After pairing it again a few times it seems to get recognised correctly.

will try that to.

meisterfuu, 2020-11-27:
Works perfectly well. Can control it through the new zigbee2mqtt web ui.

Ruftie, 2020-11-27:
> 
> 
> Works perfectly well now. Can control it through the new zigbee2mqtt web ui.
New webgui? picture?

meisterfuu, 2020-11-27:
https://www.zigbee2mqtt.io/information/frontend.html

Ruftie, 2020-11-27:
Loading devices takes too long time. 

error...

Ruftie, 2020-11-28:
got it to work with the zigbee2mqtt web UI, next job is to get the effects working and make switches in Domoticz.