Koenkk/zigbee2mqtt #5091: LVS-SC7 & LVS-ZB500D not working with CC2652

spurgelaurels, 2020-11-26:
Ah bummer this was closed; I think it's a similar issue. 

zigbee2mqtt add-on version (if edge, please report commit hash):  1.16.1

Operating environment (HassOS, Virtual Machine, Device/platform running Home Assistant):  Arch linux 5.9.1-arch1-1, HASS installed manually

Description of problem:
Having an issue with re-pairing an LVS-SC7. It worked dandy with my CC2531, but I moved to a ZZH CC2652 for better stability, along with a move to stack 3.x.0. Initially, it paired while I was re-pairing all of the other devices, but it wasn't until a while later that I realized that my scene controllers weren't working. I checked it out and they were connected and reporting LQI, but they weren't reporting the button presses. 

Tonight I tried re-pairing one, along with pairing a new out-of-the-box LVS-ZB500D, but neither of them will connect or pair and I don't see anything in my logs. 

Your entire configuration from the frontend (with sensitive fields redacted):

```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: <redacted>
  password: <redacted>
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: info
  pan_id: 6755
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
```

Your logs from Home Assistant
Nothing is appearing in the logs during the re-pair operation, even with debug enabled.

_Originally posted by @spurgelaurels in https://github.com/Koenkk/zigbee2mqtt/issues/3940#issuecomment-733403466_

spurgelaurels, 2020-11-26:
``` 
  ./cc2538-bsl.py -p /dev/ttyUSB0 -evw ../../Z-Stack-firmware/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20201026.hex
```

Adding the precise firmware version and command to flash the device. 
Should be noted that all of my Philips, Sengled, Xiaomi, and OSRAM devices are working perfectly splendidly.

Koenkk, 2020-11-26:
Can you provide the zigbee2mqtt debug logging when pressing  buttons on the LVS-ZB500D?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

spurgelaurels, 2020-11-26:
Logs are all in order below, with some comments, and redacted where indicated. 

```
Zigbee2MQTT:info  2020-11-26 13:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"1.16.1","config":{"advanced":{"adapter_concurrent":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6755,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00124b001949f16d":{"friendly_name":"CC2531 Router","retain":false},"0x00124b001969ee55":{"friendly_name":"Stage Scene Switch"},"0x00124b0019854172":{"friendly_name":"Front Scene Switch"},"0x00124b00198544b6":{"friendly_name":"Master Bedroom Scene Switch"},"0x00158d0002c7bd1d":{"friendly_name":"Laundry Room Timer"},"0x00158d0002c92d57":{"friendly_name":"Stage Lights"},"0x00158d00033e0f6e":{"friendly_name":"Big Bath Switch "},"0x00158d0003a0f4b9":{"friendly_name":"Master Bedroom Switch"},"0x00158d0003f42c07":{"friendly_name":"Office Occupancy","retain":false},"0x00158d00041400c3":{"friendly_name":"Big Bath Hall"},"0x00158d0004144081":{"friendly_name":"Kitchen Occupancy"},"0x00158d0004148fd5":{"friendly_name":"Master Bedroom Occupancy"},"0x00178801035c8a4f":{"friendly_name":"Great Hall West","retain":false},"0x00178801035c8da1":{"friendly_name":"Rachel Bed Light"},"0x0017880103699ee7":{"friendly_name":"Daphne Bed Light"},"0x00178801036b84a1":{"friendly_name":"Foyer","retain":false},"0x001788010391869a":{"friendly_name":"Stage Right"},"0x00178801039321fd":{"friendly_name":"Rachel Office"},"0x0017880103c36c28":{"friendly_name":"Stage Left"},"0x00178801049c476d":{"friendly_name":"Daphne Desk Lamp"},"0x7cb03eaa00a494f8":{"friendly_name":"Big Bath Heater"},"0x7cb03eaa00a495a6":{"friendly_name":"TV Lights","retain":false},"0x7cb03eaa00a49765":{"friendly_name":"Master Stairs Switch","retain":false},"0xb0ce1814030300cb":{"friendly_name":"SPARE"},"0xb0ce181403031309":{"friendly_name":"Kitchen","retain":false},"0xb0ce181403031538":{"friendly_name":"Art Display","retain":false},"0xb0ce181403134738":{"friendly_name":"Great Hall East","retain":false},"0xb0ce181403137753":{"friendly_name":"Big Bath Light"},"0xb0ce1814031522a8":{"friendly_name":"0xb0ce1814031522a8","retain":false},"0xb0ce18140316bf4a":{"friendly_name":"SPARE E11"},"0xb0ce18140322a63e":{"friendly_name":"Patio Stairs"},"0xb0ce18140322d00f":{"friendly_name":"Workbench"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","include_device_information":false,"server":"mqtt://core-mosquitto","user":"mqtt"},"passlist":[],"permit_join":false,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyUSB0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6755},"permit_join":true,"version":"1.16.1"}'
```
In the above message all the devices are listed, and the LVS devices in question are  
```
"0x00124b001969ee55":{"friendly_name":"Stage Scene Switch"},"0x00124b0019854172":{"friendly_name":"Front Scene Switch"},"0x00124b00198544b6":{"friendly_name":"Master Bedroom Scene Switch"}
```
Interestingly, the Stage Scene Switch was set up ages ago and I have yet to connect to my mains. The Front Scene Switch is the LVS that I un-paired and attempted to re-pair but doesn't work anymore, and the **Master Bedroom Scene Switch** is still paired although non-functioning. **This is the one that buttons were pressed during this test.** 

```
Zigbee2MQTT:info  2020-11-26 13:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok"}'
```
At this point I also permitted joins, and tried to connect the LVS Dimmer by pressing 5 times and holding until it flashes, to no effect on the Z2M side. The rest of any relevant logs follow....

```
Zigbee2MQTT:debug 2020-11-26 13:29:55: Received MQTT message on 'zigbee2mqtt/Big Bath Heater/set' with data 'OFF'
Zigbee2MQTT:debug 2020-11-26 13:29:55: Publishing 'set' 'state' to 'Big Bath Heater'
Zigbee2MQTT:info  2020-11-26 13:29:56: MQTT publish: topic 'zigbee2mqtt/Big Bath Heater', payload '{"linkquality":66,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2020-11-26 13:30:00: Device 'Master Bedroom Scene Switch' announced itself
[... cut ...] 121680,"url":"https://api.update.ledvance.com/v1/zigbee/firmwares/download?company=4364&product=39&version=0.20.509"}'
Zigbee2MQTT:debug 2020-11-26 13:30:09: Update available for '0x7cb03eaa00a49765': NO
```
It should be noted that **Master Stairs Switch** is a totally unrelated OSRAM device controlling lights by the stairs. 

After this, there are a bunch of device LQI reports, but nothing either related to the above mentioned devices, nor anything out of the ordinary that would be of interest.

Koenkk, 2020-11-30:
@spurgelaurels looks that you don't get any messages, since you have a spare CC2531 (I guess?), can you sniff the traffic when clicking the buttons on the switch? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Please indicate where you pressed the buttons (use first No. column)

spurgelaurels, 2020-11-30:
[LVSSC7.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5619145/LVSSC7.pcapng.gz)

Uploaded. I actually used the Nortek HUSBZB-1 so I hope it works. 

I started the sniff and had my partner push each button in succession, and then I stopped the capture a few seconds later. It's been decoded (default key) so I hope it's all there. I can't make sense of most of the stuff in the Zigbee protocol yet, so I'll be deep in reading tonight too :)

spurgelaurels, 2020-11-30:
[LVS500.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5619158/LVS500.pcapng.gz)
I did the same thing by starting wireshark and trying to connect the LVS dimmer with 5 presses + hold. Blue light flashed, then went out. I waited a few seconds, then closed down the pcap. 

I'm feeling like there wasn't as much activity happening as there should be for a device join.

Koenkk, 2020-12-01:
LVSSC7 looks the easiest, lets start with that (I see it's sending messages correctly but since these are manufacturer specific they appear as "unknown command"  in wireshark).

Can you provide the herdsman debug log when pressing buttons on the LVSSC7?

spurgelaurels, 2020-12-02:
Yeah, I'll dig that up shortly, but in the meantime can you maybe screenshot where you were finding those button presses / device identifiers? I saw nothing in those pcaps that caught my eye at all.

spurgelaurels, 2020-12-02:
Configured the herdsmen debug logs by adding `zigbee_herdsman_debug: true` to my zigbee2mqtt config and restarting. 
Obtained some logs by `docker logs --since 2m addon_7ad98f9c_zigbee2mqtt >> z2m_herds_out.log`
Had to obtain the others by refreshing the z2m UI repeatedly (it only seems to store a few hundred lines before wrapping.... there may be gaps)
Let me know if this isn't the correct method and I'll grab them a different way. 
[Z2M_logs.gz](https://github.com/Koenkk/zigbee2mqtt/files/5626379/Z2M_logs.gz)

Koenkk, 2020-12-02:
I don't see the logging I expect. For HA try to:
- refresh logs
- press button
- immediately refresh logs and post what is added here.

spurgelaurels, 2020-12-02:
That was a challenge to get it lined up (there seemed to be a long delay on the refresh button) but I think I got it...

```
Zigbee2MQTT:info  2020-12-02 13:27:49: MQTT publish: topic 'zigbee2mqtt/Big Bath Heater', payload '{"linkquality":0,"state":"OFF","update":{"state":"idle"},"update_available":false}'
2020-12-02T18:27:59.817Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,59,243,59,243,182,68,133,25,0,75,18,0,142,48]
2020-12-02T18:27:59.817Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,59,243,59,243,182,68,133,25,0,75,18,0,142,48]
2020-12-02T18:27:59.817Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [59,243,59,243,182,68,133,25,0,75,18,0,142] - 48
2020-12-02T18:27:59.817Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":62267,"nwkaddr":62267,"ieeeaddr":"0x00124b00198544b6","capabilities":142}
2020-12-02T18:27:59.818Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 62267
2020-12-02T18:27:59.818Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":62267,"options":0,"radius":30}
2020-12-02T18:27:59.818Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,59,243,0,30,178]
2020-12-02T18:27:59.818Z zigbee-herdsman:controller:log Device announce '0x00124b00198544b6'
Zigbee2MQTT:debug 2020-12-02 13:27:59: Device 'Master Bedroom Scene Switch' announced itself
2020-12-02T18:27:59.818Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2020-12-02 13:27:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Master Bedroom Scene Switch","ieee_address":"0x00124b00198544b6"},"type":"device_announce"}'
Zigbee2MQTT:info  2020-12-02 13:27:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Master Bedroom Scene Switch"},"type":"device_announced"}'
2020-12-02T18:27:59.833Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2020-12-02T18:27:59.833Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2020-12-02T18:27:59.833Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2020-12-02T18:27:59.833Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2020-12-02T18:27:59.833Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2020-12-02 13:28:02: Saving state to file /share/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2020-12-02 13:28:09: Received MQTT message on 'zigbee2mqtt/Big Bath Heater/set' with data 'OFF'
Zigbee2MQTT:debug 2020-12-02 13:28:09: Publishing 'set' 'state' to 'Big Bath Heater'
2020-12-02T18:28:09.453Z zigbee-herdsman:controller:endpoint Command 0x7cb03eaa00a494f8/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-12-02T18:28:09.453Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x7cb03eaa00a494f8:665/3 (0,0,1)
2020-12-02T18:28:09.453Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":665,"destendpoint":3,"srcendpoint":1,"clusterid":6,"transid":137,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,116,0]}}
2020-12-02T18:28:09.454Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,153,2,3,1,6,0,137,0,30,3,1,116,0,86]
2020-12-02T18:28:09.470Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-12-02T18:28:09.470Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-12-02T18:28:09.471Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-12-02T18:28:09.471Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-12-02T18:28:09.471Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-12-02T18:28:09.475Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,137,79]
2020-12-02T18:28:09.475Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,137,79]
2020-12-02T18:28:09.475Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,137] - 79
2020-12-02T18:28:09.475Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":137}
2020-12-02T18:28:09.475Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-12-02T18:28:09.505Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,6,0,153,2,3,1,0,0,0,241,37,161,0,0,5,24,116,11,0,0,153,2,29,210]
2020-12-02T18:28:09.505Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,6,0,153,2,3,1,0,0,0,241,37,161,0,0,5,24,116,11,0,0,153,2,29,210]
2020-12-02T18:28:09.505Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,153,2,3,1,0,0,0,241,37,161,0,0,5,24,116,11,0,0,153,2,29] - 210
2020-12-02T18:28:09.505Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":665,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":10561009,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,116,11,0,0]}}
[... cut ...] ality":0,"groupID":0}'
2020-12-02T18:28:09.506Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2020-12-02 13:28:09: MQTT publish: topic 'zigbee2mqtt/Big Bath Heater', payload '{"linkquality":0,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-12-02 13:28:18: MQTT publish: topic 'zigbee2mqtt/Kitchen Occupancy', payload '{"battery":100,"illuminance":66,"illuminance_lux":66,"linkquality":81,"occupancy":false,"voltage":3025}'
```

spurgelaurels, 2020-12-02:
This might be helpful as well... the device ID is `0x00124b00198544b6`

Koenkk, 2020-12-02:
Can you check if the issue is fixed in the dev branch?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

spurgelaurels, 2020-12-03:
> Can you check if the issue is fixed in the dev branch?

RIGHT, so in the course of removing stable, I found myself in a pile of issues and posts all less than 10 hours old about bogus unhealthy reported states causing blocked installs/upgrades, repos not showing up after being removed, and a whole other 7 hours of troubleshooting fun. I finally just got out of that mess, and the edge branch is running (although my HASS is limping....) 

I'll report back in the morning :D

spurgelaurels, 2020-12-03:
Master Bedroom Scene Switch seems to be working today! Picked up and started working in Nodered right away too. Odd, buttons 3 and 4 aren't functioning, but that might my end as they trigger a few complex functions in nodered. I'll check that out. 

I tried re-pairing the Front Scene Switch, and it's not pairing. It's the same device, and there is absolutely nothing showing up in the logs. Not sure where to start on that one....   

Same with the LVS-ZB500D (dimmer). Also not showing up in the logs at pair time.

Koenkk, 2020-12-03:
> I tried re-pairing the Front Scene Switch, and it's not pairing. It's the same device, and there is absolutely nothing showing up in the logs. Not sure where to start on that one....

Can you sniff the traffic of that? Same for the  LVS-ZB500D. Provide 2 sniffs starting at the point where you put the device into pairing mode and wait 60 seconds.

spurgelaurels, 2020-12-04:
[Dec3_Z2M_Dump.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/5640043/Dec3_Z2M_Dump.pcapng.gz)

Both device attempts are in the one pcap. Not sure it you meant to have them in two files? I can do that if needed, but otherwise I spaced them out pretty far while pairing.

Koenkk, 2020-12-04:
I meant 2 separate files indeed. Looking at your sniff, No. 241 shows that the device tries to join via a Philips device (0x00178801049c476d).

Can you try joining the device via the coordinator by pairing it very close to it? (provide a sniff if it doesn't work).

spurgelaurels, 2020-12-04:
Amazing. Luckily, I had the ZB500D connected to an AC cord before I put it into the wall. I was able to move it closer, and it paired! 
Pulling the other SC7 will be a challenge, so I'll try bringing the HASS host over to the wall where it's plugged in. 

I have about 4 of these devices left to deploy, so if you need any additional packet dumps or logs, let me know. Obviously it's not going to be ideal to have to pull an AC switch from the wall every time it needs to be paired...

closed 2021-01-11 by github-actions[bot]