Koenkk/zigbee2mqtt #5093: Multiple ZigBee networks

dewi-ny-je, 2020-11-26:
I have ZigBee devices in my house connected to a CC2531 and I would like to have some more in my garage.

Due to the distance and the respective wall thickness, no direct ZigBee connection is possible. In the garage however I have wifi thanks to a local base station connected via ethernet.

I prepared a CC2531/CC2530 connected to an ESP8266 (I followed the steps provided) and I thought about adding it to zigbee2mqtt by adding a second "port: " line but it doesn't seem to work.

What is the way to get two (basically independent) Zigbee networks? I hope I don't need to start two instances of zigbee2mqtt!

Koenkk, 2020-11-26:
1 Zigbee2MQTT instance is always 1 adapter and 1 network. If you want to run multiple network you need multiple instances.

dewi-ny-je, 2020-11-26:
Ok, I created a new Docker container. 

Do you think I can safely reuse the same base_topic as the other instance, or should I use a new one?
I tried and the second instance starts even with the same topic, but I have no idea about future problems.

Koenkk, 2020-11-27:
You should use a different base topic

closed 2020-12-28 by Koenkk

dewi-ny-je, 2021-01-20:
It worked.

Maybe it's useful to add the information to the docs. The docker-compose file in my case is similar to this:

version: '3'
services:
  zigbee2mqtt-first:
    image: koenkk/zigbee2mqtt:latest
    volumes:
      - /[first local path for config]:/app/data
#    devices:
#      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    network_mode: host
    environment:
      - TZ=[your timezone]

  zigbee2mqtt-second:
    image: koenkk/zigbee2mqtt:latest
    volumes:
      - /[second local path for config]:/app/data
#    devices:
#      - /dev/null:/dev/ttyACM0
    restart: always
    network_mode: host
    environment:
      - TZ=[your timezone]

Koenkk, 2021-01-20:
Done, added to the FAQ