Koenkk/zigbee2mqtt #5096: Pairing problems of Philips Hue Outdoor Motion Sensor

poerlemans, 2020-11-26:
## What happened
I'm trying to pair this motion sensor. According to the description (see https://www.zigbee2mqtt.io/devices/9290019758.html)  I should press the pairing button for about 10 seconds. The LED should than blink solid green. 
What I see is, that the LED is blinking green for about 3 seconds after about 10 seconds, then alternating green/red for about 5 seconds and then blinking red pulses for about 3 seconds during some minutes.

In the logs I can observe that joining is admitted. In the logs nothing can be observed about having contact with the sensor.
   
## What did you expect to happen
I should expect pairing of the device.

## How to reproduce it (minimal and precise)
I tried the pairing procedure many times. Always the same observations

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: 20190608
Computer System: Ubuntu Server, both Zigbee2mqtt and Home Assistent are running in a docker container, pairing is activated from within Home Assistant.

Log output:
Zigbee2MQTT:debug 2020-11-26 20:29:51: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
Zigbee2MQTT:info  2020-11-26 20:29:51: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-11-26 20:29:51: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"654817a","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}'

poerlemans, 2020-11-28:
Isn't there somebody on this forum who has experiences in pairing this device? I cannot imagine that this is impossible. Shall I go to the shop and try a replacement? Or is this all fruitless and should I go to another motion sensor type? Please give your comment. Thank you very much.

TimWeyand, 2020-11-29:
I have the same Problem, got an new 9290019758 (Hue Outdoor Motion Sensor) but can't pair it. Tried it like 20 Times close to an connected Router or close to the adapter CC2531. (Zigbee2MQTT 1.16.1).

I will try a different adapter next week, since I also had trouble with a new Philips Hue Calla which is located 2 meters next to a working Calla and had connectivity problems with LEDVANCE SMART+ PLUG.

But also keep the notice on [https://www.zigbee2mqtt.io/devices/9290019758.html](https://www.zigbee2mqtt.io/devices/9290019758.html) in mind:
`This specific device has been reported to have issues repairing to a Zigbee network after upgrading from a CC2531 to a CC2652 controller (Zigbee 1.2 to 3.0). (Re)pairing may only work after pairing the device to another network and channel first (has been tested with a Philips Hue 2.0 hub in this instance) before pairing it back to the Zigbee2MQTT network again.`

poerlemans, 2020-12-01:
PROBLEM SOLVED!
By installing an USB cable between my home computer and the CC2531 coordinator the pairing procedure went without interruptions. It was one of the advices, but I underestimated this aspect.
@TimWeyand please test your situation.

closed 2020-12-01 by poerlemans

CodeKlopfer, 2021-08-03:
I had the same issue. Had several other devices, including one Hue motion sensor and one other Hue outdoor motion sensor paired before successfully with my CC2531 directly connected, but not this time. The hint with the USB cable helped, pairing went through immediately :)

Zipties, 2022-02-28:
Just to help future googlers, this doesn’t seem to have anything to do with if you are using a cable or not, simply unplugging the coordinator and reinserting it does the job

scoat76, 2022-08-15:
I tried everything, like removing sensor and try to add it ... still not working ... restarting z2m .. still the same ... only physical unplag and plag again solved the problem

> Just to help future googlers, this doesn’t seem to have anything to do with if you are using a cable or not, simply unplugging the coordinator and reinserting it does the job