Koenkk/zigbee2mqtt #5098: Adding new device Salus Controls SW600

George-andrew, 2020-11-27:
## Information about the device + link

SW600 window/door sensor is small & slim, 
wireless piece of equipment that detects when a window/door is open or closed

Features:
- Contact
- Battery alarm 
- Tamper

[Page SW600](https://www.salus-controls.eu/products/online-control-smart-home-system/system-extension-s-elements/sw600-slim-window-door-sensor) 
[Photo SW600](https://salus-controls.pl/wp-content/plugins/dodawanie_produktow/includes/zdjecia/SW600_zdjecie.png)

I’ve added an entry into devices.js 
 {
        zigbeeModel: ['SW600'],
        model: 'SW600',
        vendor: 'Salus Controls',
        description: 'Door or window contact sensor',
        supports: 'Contact, temper,battery alert',
        fromZigbee: [fz.ias_contact_alarm_1],
        toZigbee: [],
        exposes: [e.contact(), e.battery_low(), e.tamper()],
    ota: ota.salus,
    },

## data/database.db entry of the device
`Zigbee2MQTT:info  2020-11-27 11:39:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20190608","friendly_name":"Coordinator","ieeeAddr":"0x00124b000be895ff","lastSeen":1606473541280,"networkAddress":0,"softwareBuildID":"zStack12","type":"Coordinator"},{"dateCode":"20190508","description":"Door or window contact sensor","friendly_name":"0x001e5e09023c92df","hardwareVersion":0,"ieeeAddr":"0x001e5e09023c92df","lastSeen":1606470434317,"manufacturerID":4216,"manufacturerName":"SALUS","model":"SW600","modelID":"SW600","networkAddress":30156,"powerSource":"Battery","type":"EndDevice","vendor":"Salus Controls"},{"dateCode":"20190415","description":"Relay switch","friendly_name":"0x001e5e0902683992","hardwareVersion":1,"ieeeAddr":"0x001e5e0902683992","lastSeen":1606402562097,"manufacturerID":4216,"manufacturerName":"Computime","model":"SR600","modelID":"SR600","networkAddress":29477,"powerSource":"Mains (single phase)","type":"Router","vendor":"Salus Controls"}],"type":"devices"}'
Zigbee2MQTT:info  2020-11-27 11:39:08: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":true,"linkquality":136,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:10: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":false,"linkquality":128,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:11: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":false,"linkquality":131,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:11: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":false,"linkquality":115,"tamper":true,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:12: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":false,"linkquality":118,"tamper":true,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:12: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":false,"linkquality":126,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:13: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":true,"linkquality":126,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:14: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":true,"linkquality":126,"tamper":false,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:14: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":true,"linkquality":126,"tamper":true,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-27 11:39:15: MQTT publish: topic 'zigbee2mqtt/0x001e5e09023c92df', payload '{"battery_low":false,"contact":true,"linkquality":126,"tamper":false,"update":{"state":"idle"},"update_available":false}'
.`

Koenkk, 2020-11-27:
Added, thanks!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-27 by Koenkk

George-andrew, 2020-11-30:
@Koenkk Thanks you!