Koenkk/zigbee2mqtt #5100: HomeAssistant: Device migration between two z2m instances

Hacker-CB, 2020-11-27:
# What happened

Currently it is impossible to migrate device between two z2m instances connected to single HA instance with different `base_topic` because `unique_id` uses `settings.get().mqtt.base_topic` as suffix:
https://github.com/Koenkk/zigbee2mqtt/blob/cb9699e12abffcfa97a7a4add40275d5a00827f8/lib/extension/homeassistant.js#L854

From the other side, device identifier uses just `zigbee2mqtt_` prefix for its identifier, without using `settings.get().mqtt.base_topic`.

https://github.com/Koenkk/zigbee2mqtt/blob/cb9699e12abffcfa97a7a4add40275d5a00827f8/lib/extension/homeassistant.js#L1065

## What did you expect to happen

Such migration can be usefull when splitting network or just moving device to other area.
We can add something like `homeassistant_unique_id_suffix` to the config to handle this case. This will not break existing instances.

@Koenkk I can add this if you will agree with this point.

Koenkk, 2020-11-28:
If I understand correctly you want to replace the base topic in the unique_id with homeassistant_unique_id_suffix (and set those to the same value??)

Hacker-CB, 2020-11-28:
> If I understand correctly you want to replace the base topic in the unique_id with homeassistant_unique_id_suffix (and set those to the same value??)

Right. Something like this:

```javascript
if (settings.get().homeassistant_unique_id_suffix){
   payload.unique_id = `${resolvedEntity.settings.ID}_${config.object_id}_${settings.get().homeassistant_unique_id_suffix}`; 
} else {
   payload.unique_id = `${resolvedEntity.settings.ID}_${config.object_id}_${settings.get().mqtt.base_topic}`; 
}
```

Koenkk, 2020-11-28:
Not sure about this yet, setting the `homeassistant_unique_id_suffix` when having it not set before will break (I think?) because the unique ID changes. Also there is a reason that the unique_id differs per Zigbee2MQTT instance, otherwise you can get duplicates.

Hacker-CB, 2020-11-28:
Currently in case if we have two devices with same IEEE address on the two z2m instances ALL entitites will be linked to the SINGLE device instance because it has `zigbee2mqtt_` prefix instead of using base_topic:
https://github.com/Koenkk/zigbee2mqtt/blob/cb9699e12abffcfa97a7a4add40275d5a00827f8/lib/extension/homeassistant.js#L1065

Generally, my opinion is that we should keep same `unique_id` for each IEEE address between two z2m instances.
* IEEE address is unique in any case. We will never get duplicates.
* **It will be very easy to migrate device from one z2m instance to another without loosing its history/settings/mapping in HA**. Its main point I opened this ticket.

Suggested `homeassistant_unique_id_suffix` is just workaround.

As more generic solution to keep `unique_id` between different z2m instances we can replace:
```
payload.unique_id = `${resolvedEntity.settings.ID}_${config.object_id}_${settings.get().mqtt.base_topic}`; 
```
to
```
payload.unique_id = `${resolvedEntity.settings.ID}_${config.object_id}_zigbee2mqtt`; 
```
I think in most cases users does not changed `base_topic`

Koenkk, 2020-11-28:
I agree, I also like `payload.unique_id = `${resolvedEntity.settings.ID}_${config.object_id}_zigbee2mqtt`; ` more, but do you know if this is breaking?

Hacker-CB, 2020-11-28:
It can break existing instances ONLY if user changed `base_topic`. In all other cases no any breaking.

Koenkk, 2020-11-29:
Since it's breaking we should keep it for Zigbee2MQTT 2.0.0 (as we cannot do breaking changes now)

closed 2022-08-27 by github-actions[bot]