Koenkk/zigbee2mqtt #5101: Zigbee channel not according configuration.yaml

K-r-i-s-t-i-a-n, 2020-11-27:
## What happened
Zigbee channel 15 is configured in configuration.yaml. However, the network is running on channel 25.
 
## What did you expect to happen
Network is running on channel 15

## How to reproduce it (minimal and precise)
Start Z2M. 

## Debug info
Zigbee2MQTT version: 1.16.1-dev (commit #6548250)
Adapter hardware: Texas Instruments LAUNCHXL-CC1352P-2
Adapter firmware version: 20201026

Relevant log message confirms: 

```
Using` zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6756},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2020-11-27 17:55:42: zigbee-herdsman started
info  2020-11-27 17:55:42: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'
debug 2020-11-27 17:55:42: Zigbee network parameters: `{"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6756}
```

Koenkk, 2020-11-28:
Can you share you configuration.yaml?

K-r-i-s-t-i-a-n, 2020-11-29:
https://hastebin.com/zabeditixe.yaml

The log seems to indicate that it is read correctly from configuration but ignored later. I can also supply a zigbee-herdsman log.

Koenkk, 2020-11-29:
Yes please provide the herdsman debug log when starting zigbee2mqtt.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

K-r-i-s-t-i-a-n, 2020-11-29:
herdsman debug log: https://hastebin.com/atoniwuset.yaml

With another stick I can sniff the traffic and see the network running on channel 25 instead of 15.

Koenkk, 2020-11-30:
The interesting part of this log is:

```
2020-11-27T16:51:04.383Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":4,"value":{"type":"Buffer","data":[0,128,0,0]}}
2020-11-27T16:51:04.383Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-27T16:51:04.384Z zigbee-herdsman:adapter:zStack:startZnp Item 'channelList' is valid
```

Here zigbee-herdsman reads the channel ID and compares it to the configured channel (if there is a mismatch it will reset the adapter and set the new channel). 

osalNvRead is what the adapter returns as configured channel:
- [0,128,0,0] == channel 15
- [ 0, 0, 0, 2 ] == channel 25

As you can see the adapter returns 15 as the channel here, but for some reason that is not the channel where it is running on.

Did you ever run your network on channel 15?

K-r-i-s-t-i-a-n, 2020-12-01:
I ran multiple testing networks and later switched my "production" network to channel 15 to avoid interference. Never checked that the network actually did what I configured, just realized recently the mismatch.

MattWestb, 2020-12-01:
@K-r-i-s-t-i-a-n  Do you have changing pan-id and epan-id 2 for the different networks or you is getting strange collisions and kidnappings between the networks?

Koenkk, 2020-12-01:
@K-r-i-s-t-i-a-n did you also reflash your coordinator at this time? (and when you did did you execute step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html ?)

K-r-i-s-t-i-a-n, 2020-12-02:
The network is running for quite a while (many weeks, propably month), not sure when exactly I started using it in this configuration. I flashed the October 26 release following the guide incl. step 5, but again propably also the firmware releases before that.

I would try reflashing incl. step 5 tonight, but what will happen to the network if it works and the coordinator switches channel? Will I have to repair all devices? (would be quite painful, 100+ devices).

Koenkk, 2020-12-02:
@K-r-i-s-t-i-a-n you will have to repair indeed. So maybe it's better to leave as-is for now.

K-r-i-s-t-i-a-n, 2020-12-03:
Not sure how to proceed here. The issue does not bother me and I want to avoid repairing all devices.

Shall we close the issue or can I supply more information at the moment?

Koenkk, 2020-12-03:
To fix we first have to find a way to reproduce, since I believe this issue is very rare and you are not bothered with it I think we should close it.

closed 2020-12-03 by Koenkk