Koenkk/zigbee2mqtt #5102: scene_add and scene_recall could have better error messages

sjorge, 2020-11-27:
## What happened
While messing around for #4926 I had a bad payload for scene_add, ... the errors here could be improve somewhat

```
Zigbee2MQTT:error 2020-11-27 18:25:14: Publish 'set' 'scene_add' to '0xec1bbdfffeb5154f' failed: 'Error: Invalid payload'
Zigbee2MQTT:error 2020-11-27 18:25:14: Publish 'set' 'scene_recall' to '0xec1bbdfffeb5154f' failed: 'TypeError: Cannot read property '22_0' of undefined'
````

## What did you expect to happen
- scene_recall in this case should just log an error that the scene with ID 22 doesn't exist
- scene_add, could try to be more verbose as to what is bad about the payload but mostly the former needs improving

## How to reproduce it (minimal and precise)
```
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_add -m '{"ID": 22, "state": "OFF", "color": "{"x":0.3818,"y":0.3797}"}'
```

## Debug info
Zigbee2MQTT version: 1.16.1-dev
Adapter hardware: CC2531
Adapter firmware version:
```json
{
  "meta": {
    "maintrel": 3,
    "majorrel": 2,
    "minorrel": 6,
    "product": 0,
    "revision": 20201110,
    "transportrev": 2
  },
  "type": "zStack12"
}
```

K-r-i-s-t-i-a-n, 2020-11-28:
Don publish to ".../set/scene_add" but rather to ".../set" with payload "scene_add" [(documentation):](https://www.zigbee2mqtt.io/information/scenes.html)
```
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set -m '{"scene_add":{"ID": 22, "state": "OFF", "color": "{"x":0.3818,"y":0.3797}"}}'
```

sjorge, 2020-11-28:
They are the same for z2m, less json nesting is just easier when messing around in the command line to debut stuff.

Other properties do work, fine so it’s publishing the scene payload. 

Found my 2nd stick so should be able to sniff later, I’m just hoping it’s something simple like a wrong field being filled in instead of all zigbee bulbs firmware: just ignoring it.

Koenkk, 2020-11-28:
Checks are currently done here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/7599ac0dd002662d0acdd892b694279baa61d60c/converters/toZigbee.js#L3661 so feel free to improve.

closed 2020-11-28 by sjorge