Koenkk/zigbee2mqtt #5103: Motion dont turn light on.

yabbah, 2020-11-27:
## What happened
I have a ikea light bulb and a motion sensor from ikea, and I have paired them into z2m. And I try to bind the sensor to the light so it turns on/off even if z2m is down. I can see it successfully bind to the group where the light is. But still not turning on when trigger motion. I have tried reset and force remove both devices and re-pair them and re-do every step, but no changes. 
## What did you expect to happen
That the light should powered on.
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: Z-Stack_3.x.0 20201026 firmware

yabbah, 2020-11-28:
Its the old motion sensor from ikea that don’t  work to bind. I test one of the new. Worked perfectly. Is there a fix for the old ones?

MattWestb, 2020-11-29:
The old one is LL and its having one random groupe configured after being reseted and paired.
One way is sniffing the groupe and using it or unbinding it from the "factory random groupe" before binding it to one new one.

yabbah, 2020-12-02:
How do I sniff the default random group?

closed 2021-01-09 by github-actions[bot]