Koenkk/zigbee2mqtt #5104: Paulmann Valora 798.26 flashing sporadically

burrima, 2020-11-27:
## What happened

I am able to pair and successfully control my new Paulmann Valora LED panel (798.26). But when switched off, the panel flashes sporadically (short on/off glitch). This happens in random intervals in the order of some minutes.

## What did you expect to happen

Lamp shall stay off when it is switched off.

## How to reproduce it (minimal and precise)

Switch light off, then wait. When zigbee2mqtt is switched off, the lamp may flash once more but then stays off and the problem does not show up again. So, there must be a correlation between the lamp and zigbee2mqtt (lamp not defective).

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2652R stick from slae.sh
Adapter firmware version: https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_20201026.zip

There is only 1 device (LED panel) and 1 zigbee2mqtt instance in the network (using channel 20).

Logfile: https://burrima.ch/dl/zigbee2mqtt.log

## My interpretation as non-expert in this topic
From the herdsman debug log I see that the LED panel seems to be unhappy and sends some messages. Sometimes, I even see some join messages. Can you please have a look if you can see something and help me?

closed 2021-01-04 by github-actions[bot]

sebastianus78, 2021-08-14:
Problem seems still not to be solved. I have 3 paulmann devices, every device has the same issue and flashes randomly.
zigbee2mqtt 1.21.0-1 and for testing zigbee2tmqtt edge

burrima, 2021-08-14:
Try setting permit_join to false in the zigbee2mqtt config. This helped in my case.

sebastianus78, 2021-08-14:
oh, that would explain why after reboot the bulbs don't flash any more. permit join is set to false after reboot. thank you :)

fibu-freak, 2024-11-02:
> Try setting permit_join to false in the zigbee2mqtt config. This helped in my case.

helped me too. Thx