Koenkk/zigbee2mqtt #5106: Aqara QBKG25LM (3gang, no neutral) does not trigger event

talan-z, 2020-11-28:
Hello everyone,

I hope to get some help with the following issue:

## What happened
I purchased some Aqara 3gang wall switches (no neutral). Model QBKG25LM. I can set the lights through mqtt, however, any button presses do not trigger an event in mqtt. The log screen (debug is turned on) is empty.
However what works is to set the lights on and off through mqtt.

It also seems that the devices is properly recogznied and configured
database.db:
{"id":26,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c7cd53b","nwkAddr":8158,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.l3acn3","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.l3acn3","manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"appVersion":27,"stackVersion":2,"hwVersion":1,"dateCode":"06-19-2020"}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":27,"stackVersion":2,"hwVersion":1,"dateCode":"06-19-2020","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606498461542}

## What did you expect to happen
zigbee2mqtt should capture any physical button presses and display it in mqtt

## How to reproduce it (minimal and precise)
Updated 3gang switch to latest firmware with the aqara app (this is the only wall switch that has an update).
Add it to zigbee2mqtt. Press switch button.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: Conbee II
Adapter firmware version: 38.88

talan-z, 2020-11-30:
Hi everyone,

Some additional information:
I have also set the zigbee herdsman into Debug mode. Strange thing is there's NOTHING else logged except for the following:
`debug 2020-11-28 19:21:16: Received Zigbee message from '0x04cf8cdf3c7da4ea-3gang', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":27,"imageType":1288,"manufacturerCode":4447}' from endpoint 1 with groupID null`

When a physical button is pressed, nothing appears in the log or screen.
Does the switch have to get activated first? How to find out? Any other ideas?

Regards,
Talan

Koenkk, 2020-11-30:
Are you able to sniff the traffic when clicking the QBKG25LM? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

talan-z, 2020-12-01:
I'd like to do that. Is it possible with the Conbee 2 and if yes, is there a guide/reocmmended tool?

Koenkk, 2020-12-01:
AFAIK that is not possible

skrobul, 2020-12-07:
Came here to say that I'm experiencing the same problem, using CC2531. Same as in @talan-z case, there is nothing showing up in the herdsman debug output. I have attempted to sniff the Zigbee traffic but whsniff was not able to find my dongle, most likely because it's using coordinator rather than sniffer firmware. I don't have a download/programmer cable but will try to obtain it now and post some updates later.
ps. it looks like @medochino experienced same problem in #4344 

One more thing, which bothers me but I'm not sure if it's actually the problem. Below map includes QBKG25LM with just one way link, while QBKG22LM (LivingRoom_Switch) seems to have connection in both directions. Is that normal?

[image]

talan-z, 2020-12-08:
Thanks, Skrobul. I unfortunately do not have a CC2531 and cannot support. But if there's anything I can do to support, happy to help!

skrobul, 2020-12-08:
@talan-z don't worry, I got this  - my cable arrived an hour ago. Unfortunately, I don't have good news, or at least this is how it looks like.

@Koenkk I got my sniffer dongle hooked up and pressed each button on QBKG25LM twice, thus turning it ON->OFF starting from left to right. I have attached the [Wireshark packet capture](https://github.com/Koenkk/zigbee2mqtt/files/5660705/QBKG25LM_packet_capture_2020-12-08_1714.pcapng.gz), but unless I'm missing something there is no packets exchanged.
Just to make sure that my sniffer actually works, I have executed exactly the same command `whsniff -c 11 | wireshark -k -i -` and pressed the other QBKG22LM switch and the packets did show up. 

Is there anything else I can try?

Koenkk, 2020-12-08:
@skrobul can you change the definition to below and see if that fixes the issue?

```js
{
    zigbeeModel: ['lumi.switch.l3acn3'],
    model: 'QBKG25LM',
    vendor: 'Xiaomi',
    description: 'Aqara D1 3 gang smart wall switch (no neutral wire)',
    fromZigbee: [fz.on_off, fz.QBKG25LM_click, fz.xiaomi_operation_mode_opple],
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode, tz.xiaomi_switch_power_outage_memory, tz.xiaomi_switch_do_not_disturb],
    meta: {multiEndpoint: true, configureKey: 1},
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'), e.action([
        'left_single', 'left_double', 'left_triple', 'left_hold', 'left_release',
        'center_single', 'center_double', 'center_triple', 'center_hold', 'center_release',
        'right_single', 'right_double', 'right_triple', 'right_hold', 'right_release',
    ])],
    onEvent: xiaomi.prevent_reset,
    configure: async (device, coordinatorEndpoint) => {
        await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(device.getEndpoint(1));
        await configureReporting.onOff(device.getEndpoint(2));
        await configureReporting.onOff(device.getEndpoint(3));
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

skrobul, 2020-12-08:
@Koenkk wow, it works! Thank you very much!

talan-z, 2020-12-09:
@Koenkk: It's working for me too! However, only 2 out of 3 switches. I tried to repair the switch, but still doesn't work.
I will look into this this weekend or during winter break. Any ideas on how to troubleshoot, please let me know.
Will the fix also get pushed into the next herdsman release? At the moment i'm injecting the manual git download of herdsman to the docker instance.

Koenkk, 2020-12-09:
Fix is in the devbranch now

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-09 by Koenkk

AbysFlo, 2021-03-05:
Hi there, sorry to reopen this issue but I'm facing the same problem. I'm on zigbee2mqtt_edge and I can't see any event. I've tried adding  the external_converters but it crashes the addon with the following error :
`zigbee2mqtt@1.18.1-dev start /app
> node index.js
(node:365) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token ':'
(Use `node --trace-warnings ...` to show where the warning was created)
(node:365) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:365) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.`

I managed to decouple the buttons but now I'm trying to get them to do something now.

As you said that this fix should be in the Dev branch, I believe adding this external converter is unnecessary and I'm pretty sure the issue is my understanding to either listen to these events correctly or me reading some documentation.

Any hint would be gladly appreciated.

Many thanks

balonchiks, 2021-06-23:
i am having exactly the same issue now. it was working fine and buttons in decoupled mode generated events, but recently everything stopped working. i have tried repairing the device, removing it from z2m and pairing it back, powering it off - it does not matter. as soon as i switch it to decoupled mode, button do not fire any events in zigbee2mqtt :(

Can someone confirm if they are still having this issue? From what i remember - there recently was an OTA update for that device, maybe this is what caused the issue?

P.S. i am on a october version of jethome.ru firmware using modkam stickV3 as the coordinator