Koenkk/zigbee2mqtt #5107: Can't load map with error "Cannot read property 'map' of undefined"

IronButterfly, 2020-11-28:
## What happened

When I click "Load map" button in frontend I see the error message "Cannot read property 'map' of undefined"
Here is debug log:

```
Zigbee2MQTT:info  2020-11-28 15:16:46: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:debug 2020-11-28 15:16:46: LQI succeeded for 'Coordinator'
Zigbee2MQTT:debug 2020-11-28 15:16:46: LQI succeeded for 'AirQMon_CO2'
Zigbee2MQTT:debug 2020-11-28 15:16:46: LQI succeeded for 'Relay'
Zigbee2MQTT:debug 2020-11-28 15:16:46: LQI succeeded for 'Outlet_Rozetka'
Zigbee2MQTT:debug 2020-11-28 15:16:55: Received Zigbee message from 'AirQMon_CO2', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"C","presentValue":37}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-28 15:16:55: MQTT publish: topic 'zigbee2mqtt/AirQMon_CO2', payload '{"co2":616,"linkquality":149,"temperature":37}'
Zigbee2MQTT:debug 2020-11-28 15:16:56: Received Zigbee message from 'AirQMon_CO2', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"ppm","presentValue":624}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2020-11-28 15:16:56: MQTT publish: topic 'zigbee2mqtt/AirQMon_CO2', payload '{"co2":624,"linkquality":149,"temperature":37}'
Zigbee2MQTT:debug 2020-11-28 15:16:56: Received Zigbee message from 'AirQMon_CO2', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"C","presentValue":33}' from endpoint 2 with groupID 0
Zigbee2MQTT:error 2020-11-28 15:16:56: Failed to execute LQI for 'Outlet_Hall'
Zigbee2MQTT:error 2020-11-28 15:16:56: Failed to execute LQI for 'Lamp_Hall'
Zigbee2MQTT:info  2020-11-28 15:16:56: Network scan finished
Zigbee2MQTT:info  2020-11-28 15:16:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{},"error":"Cannot read property 'map' of undefined","status":"error"}'
```

## What did you expect to happen

Loading map

## How to reproduce it (minimal and precise)

Just click "Load map" button in frontend

## Debug info
Zigbee2MQTT version: 1.16.1-dev commit: 26841f5
Frontend version: 0.2.91
Adapter hardware: CC2538
Adapter firmware version: 20201010

1igeza, 2020-11-28:
I had the same problem. I was using custom device (router), when I removed it, map started working again. Now I have "Received message from unsupported device with Zigbee model 'cc2538.router.v1' and manufacturer name 'jethome'" but it is working as router without any problem.

You can try to remove routers one by one to find which one causing a problem.

IronButterfly, 2020-11-28:
> I was using custom device (router), when I removed it, map started working again.

Thank you! 
What do you mean "remove"? Remove from z2m or just switch off it completely?

1igeza, 2020-11-28:
First I removed devices(with router function) to find which is causing problem. Then when I find it (router cc2538 with custom integration) i added this device again. 
I was using:
```
const device = {
    zigbeeModel: ['cc2538.router.v1'],
    model: 'cc2538.router.v1',
    vendor: 'Custom cc2538 devices (DiY)',
    description: '[CC2538 router]',
    supports: 'description, rssi',
    fromZigbee: [],
    toZigbee: [],
};
```

I deleted this custom integration and map started to work again. The router cc2538 is still working correctly but i have in logs:
```
Nov 28 10:59:12 zigbee2mqt npm[4738]: Zigbee2MQTT:warn  2020-11-28 10:59:12: Received message from unsupported device with Zigbee model 'cc2538.router.v1' and manufacturer name 'jethome'
Nov 28 10:59:12 zigbee2mqt npm[4738]: Zigbee2MQTT:warn  2020-11-28 10:59:12: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
But for me this is no a problem because it's working correctly.

IronButterfly, 2020-11-28:
@1igeza thank you for the clarification!
Yes, I found the reason. It is DIY CO2 sensor with my own external converter. After removing this external converter from configuration map works fine. After enabling the external converter back map doesn't work again.
Interestingly, this worked fine in the stable version of z2m HA addon...
Moreover, the map worked fine with a very recent build of z2m edge addon version. Something has changed quite recently.

IronButterfly, 2020-12-02:
Well, adding `exposes: []` to the `const device` section solved the problem:

`const device = {
        zigbeeModel: ['AirQMon'],
    model: 'AirQMon',
    vendor: 'DIY',
    description: 'CO₂ sensor',
    supports: 'CO₂ and temperature',
    fromZigbee: [fz.airqmon_co2, fz.airqmon_temperature],
    toZigbee: [],
    homeassistant: [sensor_co2, sensor_temperature],
    exposes: [],
};`

nurikk, 2020-12-02:
Closing then

closed 2020-12-02 by nurikk