Koenkk/zigbee2mqtt #5108: Zigbee2MQTT 1.16.1 - Can not pair a new SONOFF SNZB-02.

zibous, 2020-11-28:
## What happened
Can not pair a new SONOFF SNZB-02.

## What did you expect to happen
Nothing happens if when i long press the button on SONOFF SNZB-02.
Red light is blinking but not added as new device.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: 
   -  CC2531 as coordinator - zStack12 revision 20190619
   - 2x smartplug2 as router
   - 19 zigbee devices

[image]

herdsman debug logging is o.k but do not show pairing info for SONOFF SNZB-02

1. What can be the reason, because if I try the same with a Conbee2 stick, it works?
2. Am I at the limit of the devices that I can used for the existing configuration (CC2531 + 2 smartplug2 router)?
3. Can I simply replace the CC2531 as coordinator with the Conbee2 stick without having to re-integrate all devices?

# config:
```json
{
    "commit": "cb9699e",
    "config": {
        "advanced": {
            "adapter_concurrent": null,
            "availability_blacklist": [],
            "availability_blocklist": [],
            "availability_passlist": [],
            "availability_timeout": 30,
            "availability_whitelist": [],
            "baudrate": 115200,
            "cache_state": true,
            "cache_state_persistent": true,
            "cache_state_send_on_startup": true,
            "channel": 14,
            "elapsed": true,
            "ext_pan_id": [
                221,
                221,
                221,
                221,
                221,
                221,
                221,
                221
            ],
            "homeassistant_discovery_topic": "homeassistant",
            "homeassistant_legacy_triggers": true,
            "homeassistant_status_topic": "hass/status",
            "last_seen": "ISO_8601_local",
            "legacy_api": true,
            "log_directory": "data/log/%TIMESTAMP%",
            "log_file": "log.txt",
            "log_level": "debug",
            "log_output": [
                "syslog",
                "console"
            ],
            "log_rotation": true,
            "log_syslog": {
                "port": 514,
                "protocol": "udp4",
                "type": "bsd"
            },
            "pan_id": 6754,
            "report": true,
            "rtscts": true,
            "soft_reset_timeout": 0,
            "timestamp_format": "YYYY-MM-DD HH:mm:ss"
        },
        "ban": [],
        "blocklist": [],
        "device_options": {},
        "devices": {
            "0x00158d00028079c0": {
                "friendly_name": "wleakwasher",
                "retain": false
            },
            "0x00158d00028079c1": {
                "friendly_name": "wleakkitchen",
                "retain": false
            },
            "0x00158d00028208ae": {
                "friendly_name": "wleakbad",
                "retain": false
            },
            "0x00158d00033de1dd": {
                "friendly_name": "weatherkz2",
                "retain": false
            },
            "0x00158d0003498508": {
                "friendly_name": "weatherkz1",
                "retain": false
            },
            "0x00158d00034985f3": {
                "friendly_name": "weatherwc",
                "retain": false
            },
            "0x00158d00034d1d58": {
                "friendly_name": "weatherkitchen",
                "retain": false
            },
            "0x00158d000358e8cd": {
                "friendly_name": "motionbad",
                "no_occupancy_since": [
                    60,
                    120,
                    180
                ],
                "occupancy_timeout": 60,
                "retain": false
            },
            "0x00158d00036b56d1": {
                "friendly_name": "weatherbad",
                "retain": false
            },
            "0x00158d00036b7ae0": {
                "friendly_name": "weatherbedroom",
                "retain": false
            },
            "0x00158d00036d3594": {
                "friendly_name": "weatherflur1",
                "retain": false
            },
            "0x00158d00036d6040": {
                "friendly_name": "motiongarden",
                "retain": false
            },
            "0x00158d0003eef3c6": {
                "friendly_name": "magnet1",
                "retain": false
            },
            "0x00158d0003f13798": {
                "friendly_name": "magnet2",
                "retain": false
            },
            "0x00158d000406c3ff": {
                "friendly_name": "outdoors",
                "retain": false
            },
            "0x00158d00042631c2": {
                "friendly_name": "outdoorn",
                "retain": false
            },
            "0x00158d00042f226c": {
                "friendly_name": "smarthomecube",
                "retain": false
            },
            "0x680ae2fffe81f382": {
                "friendly_name": "ikea2",
                "retain": false
            },
            "0x7cb03eaa0a00ebba": {
                "friendly_name": "smartplug1",
                "retain": false
            },
            "0x7cb03eaa0a0145d9": {
                "friendly_name": "smartplug2",
                "retain": false
            },
            "0xec1bbdfffe59df34": {
                "friendly_name": "ikea1",
                "retain": false
            }
        },
        "experimental": {
            "new_api": true,
            "output": "json"
        },
        "external_converters": [],
        "frontend": {
            "port": 8070
        },
        "groups": {},
        "homeassistant": false,
        "map_options": {
            "graphviz": {
                "colors": {
                    "fill": {
                        "coordinator": "#FF3D00",
                        "enddevice": "#448AFF",
                        "router": "#FF6F00"
                    },
                    "font": {
                        "coordinator": "#FFFFFF",
                        "enddevice": "#FFFFFF",
                        "router": "#FFFFFF"
                    },
                    "line": {
                        "active": "#4CAF50",
                        "inactive": "#FF3D00"
                    }
                }
            }
        },
        "mqtt": {
            "base_topic": "zigbee2mqtt",
            "client_id": "rb3zigbee2mqtt",
            "include_device_information": true,
            "reject_unauthorized": false,
            "server": "mqtt://******:1883",
            "user": "*******"
        },
        "passlist": [],
        "permit_join": true,
        "serial": {
            "disable_led": false,
            "port": "/dev/ttyACM0"
        },
        "whitelist": []
    },
    "coordinator": {
        "meta": {
            "maintrel": 3,
            "majorrel": 2,
            "minorrel": 6,
            "product": 0,
            "revision": 20190619,
            "transportrev": 2
        },
        "type": "zStack12"
    },
    "log_level": "debug",
    "network": {
        "channel": 14,
        "extended_pan_id": "0xdddddddddddddddd",
        "pan_id": 6754
    },
    "permit_join": true,
    "version": "1.16.1"
}

```

Koenkk, 2020-11-28:
Probably the CC2531 table is full (5 children for 20190619 firmware). Conbee II might work without repairing, but it's recommended to repair

zibous, 2020-11-28:
@Koenkk 
Thanks, is there another firmware for the CC2531, so that ich an add more children?
The graph shows, that only 3 children are linked to the CC2531, all others are on the smartplug2's router.
CC2531 has a external antenna, so maybe this is better than the Conbee II.

Koenkk, 2020-11-28:
@zibous CC2531 with external antenna does not have better range than a Conbee II (according to my tests).

To clear the CC2531 child table you can reflash it, after that you should be able to pair it.

zibous, 2020-11-28:
@Koenkk 
Thanks 👍

closed 2020-11-28 by zibous