Koenkk/zigbee2mqtt #5109: LIVARNO LUX (Lidl) model HG06106B support

Anjerlaan, 2020-11-28:
## Information about the device + link
Jippie....I managed to get 2 of the LIDL zigbee lamps

After resetting (see page 19 of the PDF to manual link):
https://www.lidl-service.com/cps/rde/xchg/SID-9DC000A7-0D2C39EB/lsp/hs.xsl/product.html?id=5019441699&title=LED-lamp&count=1

The lamp did pair straight away....as a TUYA GU10 device. Could please support this lamp properly?

LIVARNO LUX
IAN: 334403
Model No: HG06106B
Type: Candle
Rated power 6.5 W
Lamp socket E14
Total luminous flux (6500 K) 90° cone: 470 lm

Notes:

- This LIDL candle color lamp is paired as a TUYA TS0505A (GU10)

[image]

- Selecting Color_xy yellow results in light green in the exposure tab of the web front end 

[image]

## data/database.db entry of the device
`{"id":20,"type":"Router","ieeeAddr":"0x588e81fffed5c4aa","nwkAddr":34785,"manufId":4098,"manufName":"_TZ3000_odygigth","powerSource":"Mains (single phase)","modelId":"TS0505A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505A","manufacturerName":"_TZ3000_odygigth","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genLevelCtrl":{"attributes":{"currentLevel":255}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005b6","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005b6","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b00216005b6","endpointID":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1606558232006}`

Anjerlaan, 2020-11-30:
@Koenkk,
I have paired this LIDL Livarnolux Candle E14 model: HG06106B 

[image]

To both the IKEA hub & ZZH Z2m. And both systems interviewed this device as a **TuYa TS0505A_led** device.
Does this mean no entry or addinonal info is needed in zigbee-herdsman-converters for this Lidl Device?
Although the picture of the **TuYa TS0505A_led** device is a GU10

Koenkk, 2020-12-01:
Are you able to control it correctly?

Anjerlaan, 2020-12-01:
>Are you able to control it correctly?

Yes...sofar, the only thing is that if I select the "yellow" color in the "exposes" tab, the color of the lamp becomes greenish and orange becomes more yellow.

I was wondering if these LIDL bulbs are Tuya devices which need to be added to devices.js with a fingerprint?
I can't find anything in the manual of these bulbs about Tuya, only that it used the Zigbee 3.0 protocol.
But I guess that if also the Trädfri Hub recognizes the bulb as a TS0505A, then hardware must be equal?

Koenkk, 2020-12-01:
Device is now recognized as HG06106B, note this is just a cosmetic change.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Anjerlaan, 2020-12-01:
Thanx man! This device has no OTA properties I guess?

Koenkk, 2020-12-02:
No (did not see it for TuYa devices yet)

Anjerlaan, 2020-12-02:
Hmmm...then it maybe better to stick with the IKEA products...same price but with OTA
Thanx sofar!

Koenkk, 2020-12-02:
@Anjerlaan I'm not sure what more features you expect, currently we use light features. Note that Lidl devices actually are white labeled TuYa devices.

Anjerlaan, 2020-12-02:
> @Anjerlaan I'm not sure what more features you expect, currently we use light features. Note that Lidl devices actually are white labeled TuYa devices.

I hoped for devices with OTA capability with the new Lidl smart bulbs...thanx anyway

maichai, 2020-12-02:
There are 3 RGB+CCT models sold by Lidl currently:

GU10 RGB+CCT: Model is TS0505A by _TZ3000_kdpxju99.
E27 RGB+CCT: Model is TS0505A by _TZ3000_dbou1ap4
E14 RGB+CCT: Model is TS0505A by _TZ3000_odygigth.

Can we haz support for all 3 different models in m2z, please :)

Anjerlaan, 2020-12-02:
> E14 RGB+CCT: Model is TS0505A by _TZ3000_odygigth

The "E14 RGB+CCT: Model is TS0505A by _TZ3000_odygigth" version is already added as model **HG06106B** by @Koenkk
It is now in the latest dev branch

Anjerlaan, 2020-12-02:
> Device is now recognized as HG06106B, note this is just a cosmetic change.
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@Koenkk you have added this device with vendor name: Silvercrest, but according to the Lidl site and manual of the bulbs the vendor is Livarno Lux https://www.lidl-shop.nl/LIVARNO-LUX-RGB-LED-lamp-Zigbee-Smart-Home/p100306623.

There are indeed other Lidl smart home products on their site with the vendor name Silvercrest

Koenkk, 2020-12-02:
Closing, see https://github.com/Koenkk/zigbee2mqtt/issues/5164#issuecomment-737492524

closed 2020-12-02 by Koenkk

DenisBY, 2020-12-12:
Sorry for a little off topic. How you did a reset? I tried on/off in different variations. Turned off, then turn on/off 3 times, then long on. Nothing. Turned on, turn off, wait 5 seconds, on/off 3 times, on - nothing. Is there some trick?

Anjerlaan, 2020-12-12:
Start with off, then 3x on/off...and on.
Try short offs

DenisBY, 2020-12-16:
Thanks. Did it in the following way: 3 seconds on, 1 second off. And it starts to blink on 3rd ON instead of 4th.