Koenkk/zigbee2mqtt #5110: Network slow responses and crashes of coordinator

choeflake, 2020-11-28:
Hi,

I have a network of 86 devices. Mainly Hue, but also Xiaomi, Inner, Tradfri and some Tuya based ones. The coordinator crashes several times a month. Sometimes every week, or 3 times a week and sometimes it runs for weeks.
Next to that, we experience slowness in the network.

Regulary when I press a switch, it takes seconds for a bulb to react. I catched the log around a slow event:
`zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:33: Received Zigbee message from 'Kastenkamer_Knop', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:33: Received Zigbee message from 'Kastenkamer_Knop', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:33: Received Zigbee message from 'Kastenkamer_Knop', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 2 with groupID 0
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 09:41:33: MQTT publish: topic 'zigbee2mqtt/Kastenkamer_Knop', payload '{"action":"on-press","battery":100,"brightness":255,"counter":1,"duration":0,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 09:41:33: MQTT publish: topic 'zigbee2mqtt/Kastenkamer_Knop', payload '{"action":"","battery":100,"brightness":255,"counter":1,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 09:41:33: MQTT publish: topic 'zigbee2mqtt/Kastenkamer_Knop/action', payload 'on-press'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:33: Received MQTT message on 'zigbee2mqtt/Kastenkamer_Lamp/set' with data '{"state": "ON", "transition": 0, "brightness": 143}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:33: Publishing 'set' 'brightness' to 'Kastenkamer_Lamp'
zigbee2mqtt    | Zigbee2MQTT:info  2020-11-28 09:41:46: MQTT publish: topic 'zigbee2mqtt/Kastenkamer_Lamp', payload '{"brightness":143,"linkquality":81,"state":"ON","update":{"state":"idle"},"update_available":false}'
zigbee2mqtt    | Zigbee2MQTT:debug 2020-11-28 09:41:46: Publishing 'set' 'transition' to 'Kastenkamer_Lamp'`

I interpret this as that the instruction to the bulb is delayed. Accidentally I cleared my logs, but as far as I can remember I see also timeouts in the log.

When the coordinator crashes, I see these logs: [https://pastebin.com/SeZcTgWT](https://pastebin.com/SeZcTgWT
).

For relevant versions see below in the settings dump.
`
{
    "commit": "...",
    "config": {
        "advanced": {
            "adapter_concurrent": 16,  <- changed recently from 4 to 16
            "availability_blacklist": [],
            "availability_blocklist": [],
            "availability_passlist": [],
            "availability_timeout": 0,
            "availability_whitelist": [],
            "cache_state": true,
            "cache_state_persistent": true,
            "cache_state_send_on_startup": true,
            "channel": 25,
            "elapsed": false,
            "ext_pan_id": [
            ],
            "homeassistant_discovery_topic": "homeassistant",
            "homeassistant_legacy_triggers": true,
            "homeassistant_status_topic": "hass/status",
            "last_seen": "disable",
            "legacy_api": true,
            "log_directory": "/app/data/log/%TIMESTAMP%",
            "log_file": "log.txt",
            "log_level": "info",
            "log_output": [
                "console",
                "file"
            ],
            "log_rotation": true,
            "log_syslog": {},
            "pan_id": 54159,
            "report": false,
            "soft_reset_timeout": 0,
            "timestamp_format": "YYYY-MM-DD HH:mm:ss"
        },
        "ban": [],
        "blocklist": [],
        "device_options": {
            "hue_power_on_behavior": "recover",
            "multiple_press_timeout": 0.25
        },
        "devices": {
            ....
        },
        "experimental": {
            "new_api": true,
            "output": "json"
        },
        "external_converters": [],
        "frontend": {
            "port": 8080
        },
        "groups": {
       },
        "homeassistant": true,
        "map_options": {
            ....
        },
        "mqtt": {
            "base_topic": "zigbee2mqtt",
            "ca": "/app/data/chain.pem",
            "cert": "/app/data/zigbee2mqtt.crt",
            "client_id": "zigbee2mqtt",
            "include_device_information": false,
            "key": "/app/data/zigbee2mqtt.key",
            "server": "mqtt://mosquitto:8883",
            "user": "zigbee"
        },
        "passlist": [],
        "permit_join": false,
        "serial": {
            "disable_led": false,
            "port": "/dev/ttyACM0"
        },
        "whitelist": []
    },
    "coordinator": {
        "meta": {
            "maintrel": 1,
            "majorrel": 2,
            "minorrel": 7,
            "product": 1,
            "revision": 20201026,
            "transportrev": 2
        },
        "type": "zStack3x0"
    },
    "log_level": "debug",
    "network": {
        "channel": 25,
        "extended_pan_id": "0xd5df901cc1b221ab",
        "pan_id": 54159
    },
    "permit_join": false,
    "version": "1.16.1"
}
`

Theses issues are not reproducible 'on demand'.

Where and how to start debugging this?

Koenkk, 2020-11-28:
Can you try with the following firmware first? It contains some optimizations and increases the transmit power.

choeflake, 2020-11-28:
Do you have a link for the firmware?

Koenkk, 2020-11-28:
Forgot to paste, link: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

choeflake, 2020-11-28:
Thanks! The firmware is updated. Now I have to monitor it for the next few days...

choeflake, 2020-12-05:
My general feeling is that the network is more responsive now! But it's a 'feeling' not based on any measurements. The adapter is not crashed last week. So first results look promising.
I counted 3 or 4 times that a button press did not react (or reaction takes several seconds).

For the moment, assume that this firmware is much better. Is that related to the optimisations or increased transmit power? My question about the transmit power, why should that 'help'? In my case I have bulbs everywhere in my house. So there are no long distances to travel without routing devices in between.

Koenkk, 2020-12-06:
Good to hear! hard to say what improved it but the increase of transmit power makes it easier for the coordinator to transmit messages in areas were 2.4GHz channel is used a lot (almost everywhere nowadays).

choeflake, 2020-12-07:
So adding an external antenna could also increase stability? Is it configured using the max transmit power now?

Koenkk, 2020-12-08:
> So adding an external antenna could also increase stability?

Potentially (an external antenna is not always better than a integrated antenna, it depends on the quality of the antenna and how it is connected, RF is complicated)

>  Is it configured using the max transmit power now?

Yes, cc1352 -> 20dbm, cc2652 -> 5dbm

closed 2021-01-15 by github-actions[bot]

choeflake, 2021-02-03:
I was very happy with the dev versions 20201113. But I switched to 20210107 a few weeks ago and my feeling is that this version is a little more worse. Events are missed several times a day. I will try the newest version tonight.
I think the crashes are related to https://github.com/Koenkk/zigbee2mqtt/discussions/6019. It seems that the higher transmit power fixes a lot. But low(er) power -> more timeouts and retries -> more/earlier crashes.