Koenkk/zigbee2mqtt #5113: Failed to interview LivingWise LVS-SN10ZW_SN11 to Zigbee2Mqtt 1.16.1-dev

khaidakin, 2020-11-28:
## What happened
Trying to pair LivingWise LVS-SN10ZW_SN11 (https://www.zigbee2mqtt.io/devices/LVS-SN10ZW_SN11.html)

## What did you expect to happen
I expected the device to connect and work as it did on the old Z2M version, but the device connects with an error and does not show anything in State.

## How to reproduce it (minimal and precise)
try to LivingWise LVS-SN10ZW_SN11 to zigbee2mqtt

## Debug info
Zigbee2MQTT version: 1.16.1-dev
Adapter hardware: CC2652-E72
Adapter firmware version: 20200921 (Egony)
https://pastebin.com/Lcte29yp

spurgelaurels, 2020-11-30:
#5091 

Looks like we're having similar issue. I'm wondering if Livingwise has something weird in their stack that isn't working with 3.x.0. 
I assume you're running 3.x.0 on your end?

Koenkk, 2020-11-30:
Duplicate of #5091

closed 2020-11-30 by Koenkk

khaidakin, 2020-11-30:
> #5091
> 
> Looks like we're having similar issue. I'm wondering if Livingwise has something weird in their stack that isn't working with 3.x.0.
> I assume you're running 3.x.0 on your end?
I think yes.