Koenkk/zigbee2mqtt #5115: Exit docker container when adapter is disconnected

choeflake, 2020-11-28:
Hi,

When the adapter is disconnected, the Zigbee2mqtt process is not proper stopped (I think). It would be very useful when the process exits and the container will be stopped. Combined with a restart policy of docker-compose, Zigbee2mqtt will automatically come back online after the adapter is connected again.

When I start the container without an adapter connected, the container exits after detecting there is no adapter (and docker-compose restarts the container).

Zigbee2mqtt log: [https://pastebin.com/Mfkc8Yna](https://pastebin.com/Mfkc8Yna)

The referred npm debug log: [https://pastebin.com/GqzwfbDi](https://pastebin.com/GqzwfbDi)

Koenkk, 2020-11-29:
Could it be that the docker restart policy does not restart when the exit code is 1? (looking at the logs Zigbee2MQTT stops as expected)

closed 2020-12-30 by Koenkk

hallard, 2021-02-01:
I got the same issue, I have systemd that restart the container, but it stop in a wrong state letting `/tmp/_MEI*` files that fill up the device.

Any chance that when adapter get disconnected that zigbee2mqtt just try to reconnect instead of going to death?

```shell
root@~/docker/zigbee2mqtt# l /tmp/
total 112
drwx------  2 root root 4096 Jan 25 18:51 mc-root
drwx------ 13 root root 4096 Jan 31 23:30 _MEI0wm1XH
drwx------ 13 root root 4096 Feb  1 13:59 _MEI1lyS41
drwx------ 13 root root 4096 Feb  1 21:28 _MEI4ZaQur
drwx------ 13 root root 4096 Feb  1 21:00 _MEI6xFlNu
drwx------ 13 root root 4096 Feb  1 18:50 _MEI6Y4RZX
drwx------ 13 root root 4096 Feb  1 05:45 _MEIAoxLsB
drwx------ 13 root root 4096 Feb  1 16:04 _MEIdZQ2OT
drwx------ 13 root root 4096 Feb  1 07:45 _MEIE6T4kF
drwx------ 13 root root 4096 Jan 31 23:35 _MEIh5sWpL
drwx------ 13 root root 4096 Feb  1 08:00 _MEIHAX4FO
drwx------ 13 root root 4096 Feb  1 01:10 _MEIIEamWh
drwx------ 13 root root 4096 Feb  1 18:45 _MEIIsLfKA
drwx------ 13 root root 4096 Feb  1 07:50 _MEIJSd3bi
drwx------ 13 root root 4096 Feb  1 01:15 _MEIK1hwPX
drwx------ 13 root root 4096 Feb  1 16:10 _MEIM9enfz
drwx------ 13 root root 4096 Jan 31 23:45 _MEInNfTNj
drwx------ 13 root root 4096 Feb  1 08:06 _MEIP2QFrZ
drwx------ 13 root root 4096 Feb  1 15:59 _MEIp4ocb3
drwx------ 13 root root 4096 Jan 31 23:40 _MEIQNGEE9
drwx------ 13 root root 4096 Feb  1 16:15 _MEIRUZIIu
drwx------ 13 root root 4096 Feb  1 14:10 _MEIs9j99u
drwx------ 13 root root 4096 Feb  1 21:10 _MEIT3yOgZ
drwx------ 13 root root 4096 Feb  1 14:05 _MEItxcSfU
drwx------ 13 root root 4096 Feb  1 21:05 _MEIvyAow3
drwx------ 13 root root 4096 Feb  1 07:55 _MEIVzW4i4
drwx------ 13 root root 4096 Feb  1 15:54 _MEIxhTBO4
drwx------  3 root root 4096 Jan 25 18:26 systemd-private-87548fcf7fbf46bfa8414db9aecdc525-ntp.service-PLLfMz
```

Here the log of the crash/restart
```
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:00:25: MQTT publish: topic 'zigbee2mqtt/bridge/groups', payload '[]'
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:00:25: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"07cdc9d","coordinator":{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200211,"transportrev":2},"type":"zStack30x"},"log_level":"info","network":{"channel":11,"extendedPanID":"0x00124b0018eb42df","panID":6754},"permit_join":false,"version":"1.17.0"}'
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[31mZigbee2MQTT:error#033[39m 2021-02-01 20:05:25: Adapter disconnected, stopping
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:25: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:25: Disconnecting from MQTT server
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[31mZigbee2MQTT:error#033[39m 2021-02-01 20:05:25: Failed to stop zigbee
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! code ELIFECYCLE
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! errno 1
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! Exit status 1
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR!
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR! A complete log of this run can be found in:
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m npm ERR!     /root/.npm/_logs/2021-02-01T20_05_25_546Z-debug.log
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m Using '/app/data' as data directory
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m > zigbee2mqtt@1.17.0 start /app
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m > node index.js
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:33: Logging to console and directory: '/app/data/log/2021-02-01.20-05-31' filename: log.txt
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:33: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:33: Starting zigbee-herdsman (0.13.46)
[... cut ...]
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:37: Connected to MQTT server
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:37: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Feb  1 21:07:12 eeebox docker-compose[4164]: #033[36mzigbee2mqtt    |#033[0m #033[32mZigbee2MQTT:info #033[39m 2021-02-01 20:05:37: MQTT publish: topic 'zigbee2mqtt/Cube-e6c8', payload '{"angle":3.98,"battery":100,"side":5,"voltage":3015}'
```