Koenkk/zigbee2mqtt #5116: Zigbee2Mqtt Assistant

rapid1985, 2020-11-28:
Hello

I would like to install the Zigbee2Mqtt Assistant on Docker on a Raspberry Pi 1, but it always restarts and can therefore not work.

`sudo docker run -p 8880:80 -e "Z2MA_SETTINGS__MQTTSERVER=192.168.0.50" -e "Z2MA_SETTINGS__MQTTUSERNAME=<mqttusername>" -e "Z2MA_SETTINGS__MQTTPASSWORD=<mqttpassword>" -e "Z2MA_SETTINGS__MQTTPORT=1024" --restart unless-stopped carldebilly/zigbee2mqttassistant`

What can I do?
Thank you Alex

tavalin, 2020-11-28:
Zigbee2mqtt assistant is not part of this project so you should ask over at the ZigBee2mqtt assistant repo

rapid1985, 2020-11-28:
Thank you for the information

Koenkk, 2020-11-29:
For zigbee2mqtt assistant create an issue here; https://github.com/yllibed/Zigbee2MqttAssistant/issues

closed 2020-11-29 by Koenkk