Koenkk/zigbee2mqtt #5121: sonoff SNZB-03 not able to add it to cc2531

lucaera, 2020-11-29:
Hi guys, i recently purchased a SNZB-03 wanted to add to homeassistant but the cc2531 doesnt add it automatically, on debug shows"Received message from unsupported device with Zigbee model '66666'".

Anyone can help me, please? i tried to add manually but i dont know from where get the friendly name.

I ve already reset it several times but with no results.

Thanks

matiserek-1, 2020-11-29:
I have almost the same situation.
In my case this device is recognised, but as SNZB-02.
Is it anyway to change it or fix it manually? 

[log_from_adding_device.txt](https://github.com/Koenkk/zigbee2mqtt/files/5611990/log_from_adding_device.txt)

sjorge, 2020-11-29:
If you upgrade to the latest dev release this should already be fixed:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L13303

jsrich31, 2020-11-29:
Using the Sonoff SNZB-03 motion sensor to with the Zigbee2mqtt Edge Addon in Home Assistant. When added it shows "Unsupported". How can I fix this?  All suggestion will be appreciated 😁

https://www.zigbee2mqtt.io/devices/SNZB-03.html
What am I missing?

closed 2021-01-06 by github-actions[bot]