Koenkk/zigbee2mqtt #5127: Aqara devices seen but not supported (Failed to interview)

sunseaandpalms, 2020-11-29:
Moved from https://github.com/danielwelch/hassio-zigbee2mqtt/issues/467

## What happened

I set up `Home Assistant + hassio-zigbee2mqtt + Aqara Motion sensor`. Sensor appears in list, but says `Unsupported`. Log says: "Failed to interview '0x00158d0001e51295', device has not successfully been paired".

## What did you expect to happen

Aqara sensor should pop up as a new device in HA.

## How to reproduce it (minimal and precise)

* Put Aqara Motion Sensor (tried version with and without brightness sensor) into pairing mode (Long press -> see 3 blinks -> next single long blink -> followed by 3 blinks)
* Logs say that device connected and interview started ([check on pastebin](https://pastebin.com/TdYBRpng))
* Wait and see failure message (eg. `Failed to interview '0x00158d0001e51295', device has not successfully been paired`, [check pastebin for details](https://pastebin.com/yEPf7265))
* Tried the same happens with Aqara Buttons.
* Go to Zigbee2MQTT addon UI and see sensors are marked as unsupported:
[image]

## Debug info

### Setup details

* RaspberryPi 3 B
* HassOs (4.16)
* ConBee I (plugged into Pi's USB port)
* HassOs -> Hardware options -> [Check on pastebin](https://pastebin.com/GgLZPch4)
* Mosquitto broker (5.1)
* Zigbee2mqtt (1.16.1)
* Zigbee2mqtt -> Configuration (note `deconz` adapter) -> [Check on pastebin](https://pastebin.com/azBC80V2)
* Aqara motion sensor & Aqara buttons
* [Full Zigbee2MQTT addon logs](https://pastebin.com/qTzdEe2L)

### Debugging attempts

* Attempt 1: Bring the Aqara Motion sensor closer to Pi with ConBee -> Brought as close as 1 meter from ConBee stick. Issue remains ([see log](https://pastebin.com/zgq2tvBC))
* Attempt 2: When pairing, after the lights blink in the right way, keep pressing the button every 1 second to avoid device going to sleep and interrupting the pairing process -> Tried and issue remains
* Attempt 3: Watch out the sensor blinking pattern -> it gives 3 short blinks followed by long blink followed by another set of 3 short blinks
[image]
* Attempt 4: potential interference: depending on what you run in your system, you might need an USB cable to move the USB stick a bit away from the machine -> Not tried yet

Koenkk, 2020-11-30:
I really recommend the USB extension cable option first, Pi's are known to generate a lot of intereference

sunseaandpalms, 2020-12-09:
Tried with an extension cable - no luck. I can try moving the PI into other place with less interference, since event 2.4Ghz WiFi has connection breaks in this apartment.

sunseaandpalms, 2020-12-15:
@Koenkk tried all suggestions - no luck yet. Any ideas? Could it be due to `ConBee I` in use? Is there a way I can narrow down the issue?

Koenkk, 2020-12-15:
@sunseaandpalms is anything logged to the hersdman debug log when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sunseaandpalms, 2020-12-16:
Going to try that. Many thanks for reply

sunseaandpalms, 2020-12-17:
I've uploaded the [log](https://pastebin.com/bqgyHvr5). The test was with USB extension cable. Issue seemed to be in  `addr: 0xc34c waiting for response TIMEOUT`. Any thoughts?

Koenkk, 2020-12-17:
Is this https://www.zigbee2mqtt.io/devices/RTCGQ01LM.html or https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html ?

sunseaandpalms, 2020-12-19:
@Koenkk I repeated the experiment with both RTCGQ11LM and RTCGQ01LM. Issue remains the same. I tried WSDCGQ01LM as well, but it shows different kind of error `PERMIT_JOIN FAILED - APS TIMEOUT` and 
```
(node:351) UnhandledPromiseRejectionWarning: APS TIMEOUT
(node:351) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 21)
```
I've updated the issue comment. See `Attempt 7`

Koenkk, 2020-12-20:
@chrishae `PERMIT_JOIN FAILED - APS TIMEOUT` does this mean the connection with the conbee is lost?

sunseaandpalms, 2020-12-25:
@Koenkk any thoughts what else could I check? Could it be due to the fact that devices were linked to AqaraHub previously ( I unplug the Hub from the socket during my testing)?

My next thought is to give up with `ConBee I` and order another adapter

Koenkk, 2020-12-25:
@sunseaandpalms make sure that your complete aqara network is down before pairing (so also all routers, e.g. bulbs sockets etc.)

sunseaandpalms, 2020-12-28:
@Koenkk, the `Attempt 5` (from the description) was done in a completely other place with the devices that I reset a couple of times. I had only 3 Xiaomi sensors with me (all were reset), ConBee I and Xiaomi Wifi Router. Should I retest by keeping the battery only in one Xiaomi device even if I have reset others?

Koenkk, 2020-12-28:
Preferably yes, also try with a different battery (reported to help sometimes)

sunseaandpalms, 2020-12-28:
Ok, thank you. Will do

markheloking, 2020-12-31:
Indeed pairing closer to the router/coordinator and using a fresh battery is known to help with the Aqara devices. It sometimes also helps to keep the device awake (by pressing the button/etc) during the interrogation process as some of these devices sleep quite aggressively.

zigomatichub, 2021-01-04:
Scan are success for the 4 new aqara door sensor but all sensors are reported as Unsupported

zigomatichub, 2021-01-04:
Ok some news: I achieve one to be recognize.  What I did: 

1. Put in debug mode + join to true in config file , restart
2. Remove battery of the sensor, wait some minutes and try again. Just pushing one time and scan : bing ... ->

INFO: Device '0x00158d0003a353b4' joined
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0003a353b4","ieee_address":"0x00158d0003a353b4"},"type":"device_joined"}'
INFO: Starting interview of '0x00158d0003a353b4'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0003a353b4","ieee_address":"0x00158d0003a353b4","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0003a353b4"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0003a353b4"},"type":"pairing"}'
DEBUG: Device '0x00158d0003a353b4' announced itself
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0003a353b4","ieee_address":"0x00158d0003a353b4"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0003a353b4"},"type":"device_announced"}'
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'attributeReport', cluster 'genBasic', data '{"appVersion":3,"modelId":"lumi.sensor_magnet.aq2"}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3055,"10":0,"100":1,"3":29,"4":424,"5":187,"6":[0,1]}}' from endpoint 1 with groupID 0
INFO: MQTT publish: topic 'zigbee2mqtt/0x00158d0003a353b4', payload '{"battery":100,"contact":false,"linkquality":162,"voltage":3055}'
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"hwVersion":2}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20161128"}' from endpoint 1 with groupID 0
DEBUG: Received Zigbee message from '0x00158d0003a353b4', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"3000-0001"}' from endpoint 1 with groupID 0

ChrisHae, 2021-01-11:
@zigomatichub what coordinator/gatway do you use?

zigomatichub, 2021-01-12:
@ChrisHae I use cc2531 as coordinator in Zigbee 1.2 as recommended.  
I achieve to have 4 aqara attached to it but it doesn't work as expected. 
If I stop/start zigbee2mqtt, they not reconnect. If I try to reconnect myself, I see that is linked again and I see some responses but the status is never change for true/false. At least 2 of them still ok but next reboot, those are not giving status as well.
I need to force a remove and reconnect which can take 20 min of trials.

sunseaandpalms, 2021-01-13:
I repeated the test. Used `Xiaomi WSDCGQ01LM` temp sensor that gave `PERMIT_JOIN FAILED - APS TIMEOUT` originally. Now it gives the same error as motion sensors `NODE_DESCRIPTOR FAILED - addr: 0x8670 waiting for response TIMEOUT`.

Env:
* Looked for and disabled Aqara devices around the apartment
* Used new battery
* USB extension cable
* Short-clicking on sensor to prevent going asleep

zigomatichub, 2021-01-14:
> Sorry guys for asking this, but how do I get the full log? I have been copy-pasting from web previously, but it the add-on UI only shows the log partially.

Available under /share/zigbee2mqtt/log

sunseaandpalms, 2021-01-18:
Dear masters, what would be the best next step? Should I sniff the zigbee traffic somehow or order another ZigBee stick?

ChrisHae, 2021-01-19:
Which FW version are you using? If it is not 0x26390500 please update it. https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

sunseaandpalms, 2021-01-19:
The issue is solved after ConBee update to the recent FW. Many thanks!

closed 2021-01-19 by sunseaandpalms