Koenkk/zigbee2mqtt #5131: Aurora Aone 250W Dimmer

jonbrunt123, 2020-11-30:
# Bug Report
## What happened

The Aurora AOne Dimmer 250W profile appears to have changed, with scenes now available through the device options. These scene's do not work. Ever since this change the device does not feed back any data to home assistant. The device is controllable via Home Assistant, however if the device is controlled manually this data is not fed back to Home Assistant.

## What did you expect to happen

Prior to this the device fed back all data.

## How to reproduce it (minimal and precise).

Another user has recreated this. Pair the dimmer to the router and control manually, the device will not feed back data.

## Debug Info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version: 1.2

jonbrunt123, 2020-12-13:
I have downloaded and installed Edge, the problem seems to still exist, however the LQI now updates every so often.

jonbrunt123, 2020-12-27:
@Koenkk sorry to tag, but i can't seem to workout how to mark something as a problem (very very new to GitHub). I'm sure this issue is a very straightforward fix, however i have no coding knowledge whatsover!! Can you assist??

Koenkk, 2020-12-27:
Can you try updating the definition in devices.js to below and see if it works?

```js
{
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module',
    extend: preset.light_onoff_brightness,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

jonbrunt123, 2020-12-27:
Worked brilliant after I worked out how to change the file!! Will that changed make it into the next version out of interest??

Koenkk, 2020-12-27:
Good to hear, will be included in the next z2m version (will be released on 1 january).

closed 2020-12-27 by Koenkk