Koenkk/zigbee2mqtt #5132: Aqara D1 single key WXKG06LM release action?

Raimon93, 2020-11-30:
Details : Aqara D1 single key wireless wall switch (WXKG06LM)

The actions only show Hold, Single and Double. I cannot get the release function working, more people having this issue? With this type Aqara D1 switch

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531

kevincaradant, 2020-12-01:
Hi

Indeed until you said it, I didn't notice that the release action seems to doesn't work.

That doesn't help you but just to confirm that I have also this issue ;)

MagieMalone, 2020-12-23:
I can confirm, this also applies to  Xiaomi WXKG02LM, the release messages are never send.
```
Dec 23 08:56:00 hassbian npm[4798]: Zigbee2MQTT:info  2020-12-23 08:56:00: MQTT publish: topic 'zigbee2mqtt/studeerkamer', payload '{"action":"hold_left","battery":100,"linkquality":18,"voltage":3005}'
Dec 23 08:56:00 hassbian npm[4798]: Zigbee2MQTT:info  2020-12-23 08:56:00: MQTT publish: topic 'zigbee2mqtt/studeerkamer', payload '{"action":"","battery":100,"linkquality":18,"voltage":3005}'
Dec 23 08:56:00 hassbian npm[4798]: Zigbee2MQTT:info  2020-12-23 08:56:00: MQTT publish: topic 'zigbee2mqtt/studeerkamer/action', payload 'hold_left'
Dec 23 09:00:38 hassbian npm[4798]: Zigbee2MQTT:info  2020-12-23 09:00:38: MQTT publish: topic 'zigbee2mqtt/hal', payload '{"battery":100,"illuminance":8,"illuminance_lux":8,"linkquality":5,"occupancy":false,"voltage":3035}'

```

Koenkk, 2020-12-23:
@MagieMalone the WXKG02LM doesn't support it (also own one and checked it)

@Raimon93 therefore I don't expect the WXKG06LM to support it. But to confirm can you provide the debug log when pressing the button, holding it for +- 1 sec and releasing it? In the log clearly indicate where you started pressing.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

MagieMalone, 2020-12-24:
Then the documentation is wrong, as there is says that the  WXKG02LM  also produces release_left, release_right & release_both and funny thing is, it used to work... I'll run some debug later too.

Raimon93, 2020-12-24:
i own a WXKG06LM and 07, both dont have the release function, so far i see..

WXKG06LM 
Zigbee2MQTT:debug 2020-12-24 12:15:23: Received Zigbee message from 'Schakelaar_keuken1', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-24 12:15:23: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken1', payload '{"action":"hold","battery":100,"linkquality":5,"voltage":3035}'
Zigbee2MQTT:info  2020-12-24 12:15:23: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken1', payload '{"action":"","battery":100,"linkquality":5,"voltage":3035}'
Zigbee2MQTT:info  2020-12-24 12:15:23: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken1/action', payload 'hold'

WXKG07LM
Zigbee2MQTT:debug 2020-12-24 12:16:58: Received Zigbee message from 'Schakelaar_keuken_dubbel', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2020-12-24 12:16:58: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken_dubbel', payload '{"action":"right_long","battery":100,"linkquality":5,"voltage":3115}'
Zigbee2MQTT:info  2020-12-24 12:16:58: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken_dubbel', payload '{"action":"","battery":100,"linkquality":5,"voltage":3115}'
Zigbee2MQTT:info  2020-12-24 12:16:58: MQTT publish: topic 'zigbee2mqtt/Schakelaar_keuken_dubbel/action', payload 'right_long'

Koenkk, 2020-12-24:
@Raimon93 if the switch only produces one `Received Zigbee message from` when you do a hold and release it indeed doesn't support it.

MagieMalone, 2020-12-26:
and from the 02 the same, 
```

Dec 26 08:46:29 hassbian npm[21061]: Zigbee2MQTT:debug 2020-12-26 08:46:29: Received Zigbee message from 'studeerkamer', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
Dec 26 08:46:29 hassbian npm[21061]: Zigbee2MQTT:info  2020-12-26 08:46:29: MQTT publish: topic 'zigbee2mqtt/studeerkamer', payload '{"action":"hold_left","battery":100,"linkquality":23,"voltage":3015}'
Dec 26 08:46:29 hassbian npm[21061]: Zigbee2MQTT:info  2020-12-26 08:46:29: MQTT publish: topic 'zigbee2mqtt/studeerkamer', payload '{"action":"","battery":100,"linkquality":23,"voltage":3015}'
Dec 26 08:46:29 hassbian npm[21061]: Zigbee2MQTT:info  2020-12-26 08:46:29: MQTT publish: topic 'zigbee2mqtt/studeerkamer/action', payload 'hold_left'
Dec 26 08:46:31 hassbian npm[21061]: Zigbee2MQTT:debug 2020-12-26 08:46:31: Received Zigbee message from 'studeerkamer', type 'attributeReport', cluster 'genBasic', data '{"65520":{"data":[170,16,5,65,135,104,1,16,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
```
so no release message either.

Koenkk, 2020-12-27:
Confirms these switches only support single actions. Docs will be updated automatically with new release.

closed 2020-12-27 by Koenkk

dungodungo, 2021-02-02:
Hey guys, I just purchased the WXKG06LM. I am trying to pair it with my Samsung SmartThings Hub. This would be my first smart home device. I have absolutely no idea what I'm doing. From what I gather, I need to find device handler code in order to add this device. Any help would be appreciated!