Koenkk/zigbee2mqtt #5137: BlitzWolf BW-SHP13 precise power value?

vogler, 2020-11-30:
How is the power value computed? It seems useless/wrong.
I connected a 9W LED and it reported 2 and 4 whereas `current * voltage = 9.28W` seems to be the right:
~~~json
{"current":0.04,"linkquality":68,"power":2,"voltage":232}
{"current":0.04,"linkquality":68,"power":4,"voltage":232}
{"current":0.04,"linkquality":65,"power":2,"voltage":231}
{"current":0.04,"linkquality":65,"power":4,"voltage":231}
{"current":0.04,"linkquality":68,"power":2,"voltage":231}
{"current":0.04,"linkquality":65,"power":4,"voltage":231}
{"current":0.04,"linkquality":63,"power":2,"voltage":232}
{"current":0.04,"linkquality":63,"power":4,"voltage":231}
~~~
I assume power is reported by the device like that, but why if it has more precise values?

## Debug info
Zigbee2MQTT version: 1.16.1-dev (commit #b1b134f)
Adapter hardware: CC2531

vogler, 2020-11-30:
Ok, I just compared the values with an old Voltcraft Energy Check 3000. I don't have a good reference, but the values seem more trustworthy:

[E27 80 LEDs 9W](https://www.aliexpress.com/item/32844816718.html):
~~~
Voltcraft: 2.2W
SHP13:
	power: 2W and 4 W, alternating
	I*V: 9.28W
~~~
[E27 189leds 20W](https://www.aliexpress.com/item/32859580362.html):
~~~
Voltcraft: 12.3W
SHP13:
	power: 16W
	I*V: 30.16W
~~~
Ikea Ledare E27 1800lm 22W 2700K:
~~~
Voltcraft: 23.7W
SHP13:
	power: 28W
	I*V: 30.03W
~~~

So it seems the reported power is closer to the truth than I*V, but why the difference?
It would be nice to have more data points to see if this can be improved/calibrated. At least in that range, dividing power by 1.2 seems ok, but maybe it's non-linear.

vogler, 2020-12-04:
> Since this is alternating current, you cannot use I*V: https://en.wikipedia.org/wiki/Electric_power#Alternating_current

I thought that's fine for resistive circuits. So you're saying the difference comes from some capacitance in the LED circuit?

JJanssen123, 2020-12-04:
T.b.h. i never opened up a led bulb but i can imagine there's some capacitive and inductive stuff in the ac/dc converter. 

I always thought a measuring device should return P exactly because you cannot always know it's kind of load. So you don't have to know wether I*V is applicable.

stefangries, 2020-12-30:
> Ok, I just compared the values with an old Voltcraft Energy Check 3000. I don't have a good reference, but the values seem more trustworthy:
>
> [E27 80 LEDs [...]

Mine reports values, that are too high as well. For a 60W bulb it reports 70W, so dividing by 1.2 is quite close to the truth for me as well.

Have you found out, if this behavior is linear for higher power consumptions? Can this divisor be added to the device, or do I have to add it in home assistant?

vogler, 2020-12-30:
I shelved it because I wasn't happy with it spamming messages every 10s. Seems like it has now changed to 60s.
(Would be nice if it had configurable polling intervals for the on/off state. Otherwise it's either annoying or imprecise. But that's another issue.)

I just tried a little oven:
~~~
Voltcraft: 575W
SHP13:
	power: 672W
	I*V: 676W
~~~

Assuming it's more precise at higher power, 1.17 would be a good divisor.

markheloking, 2020-12-31:
When comparing measurements, please keep the power factor in mind. Have you got a seperate measurement device on hand that can also measure power factor? It's entirely possible the SHP13 only reports true power and not apparent power leading to the differences in measurements/calculations. Also keep in mind it's AC, so you'll have to RMS for meaningfull calculations.

More about different types of power [here](https://www.allaboutcircuits.com/textbook/alternating-current/chpt-11/true-reactive-and-apparent-power/#:~:text=True%20power%20is%20symbolized%20by,unit%20of%20Watts%20(W).&text=Total%20power%20in%20an%20AC,Volt%2DAmps%20(VA).).

stefangries, 2020-12-31:
Unfortunately, I have little expertise when it comes to electrical engineering. 

What I have done is measure a traditional (non-LED) 60W bulb with the SHP13. The result varied between 68W and 70W. 

I then used my multimeter to determine voltage and current. There, the product came out to 59W.

markheloking, 2020-12-31:
What setting did you put your multimeter to? What kind of bulb is it? Incandescent or CFL? (only incandescent bulbs will have power factor of 1.0)

vogler, 2020-12-31:
> Have you got a seperate measurement device on hand that can also measure power factor?

I have a PZEM-004T laying around, but haven't used it yet.

> It's entirely possible the SHP13 only reports true power and not apparent power leading to the differences in measurements/calculations.

But then it should report less, not more.
As I see it, the only assumptions that would explain the numbers are: 1. SHP13 reports apparent power & 2. Voltcraft reports true power & 3. the circuit is not purely resistive.

Conjunct 3 should be false for @stefangries' 60W bulb, and also my oven (it's very old, has a wind-up wheel to turn it on).

stefangries, 2020-12-31:
> What setting did you put your multimeter to? What kind of bulb is it? Incandescent or CFL? (only incandescent bulbs will have power factor of 1.0)

An traditional incandescent bulb.

With the multimeter I first measured the voltage of the socket. Then I connected the multimeter in series with the light bulb to measure the current. And then I multiplied the results.

markheloking, 2020-12-31:
> I have a PZEM-004T laying around, but haven't used it yet.
>
>
> But then it should report less, not more.
> As I see it, the only assumptions that would explain the numbers are: 1. SHP13 reports apparent [...]

Hmm this would indeed indicate an error of about 20% which would be huge. I'm wondering whether it's a stable offset, or whether every device has it's own offset/error. Using a single calibration value for all devices seems like a bad idea until we figure out some average offset from multiple devices.

stefangries, 2020-12-31:
I have ordered 5 more that are on their way to me. It takes a little time from China, but after arrival I can say if the offset is similar everywhere between the devices.

vogler, 2020-12-31:
Factor, not offset.
By device you mean different SHP13s?
We have data points [(@stefangries, 60W), (@vogler, 575W)]. Could be a coincidence that our factors match, but my guess is that it's consistent.

markheloking, 2020-12-31:
> I have ordered 5 more that are on their way to me. It takes a little time from China, but after arrival I can say if the offset is similar everywhere between the devices.

This should allow for some back-of-napkin conclusions until better info is known. Maybe we should encourage people to check theirs and keep posting measurement info here in the future so that we can keep track of when/of this changes in the future (manufacturers are known to change devices like this without updating model numbers).

> Factor/divisor, not offset (addition).
> By device you mean different SHP13s?
> We have data points [(@stefangries, 60W), (@vogler, 575W)]. Could be a coincidence that our factors match, but my guess is that it's consistent.

I was indeed talking about lineair correction calibration, so both offset and factor. In this instance I used the wrong word as I was talking about that factor of 1.2x.

vogler, 2020-12-31:
> I was talking about that factor of 1.2x.

-pedantic: That would be the divisor, the factor would be 1/1.2 🙈 
I assumed the offset to be 0, but yes, it could be up to 4/1.2-2.2 = 1.13W for my first example.

vogler, 2021-02-01:
bump

> Maybe we should encourage people to check theirs and keep posting measurement info here in the future so that we can keep track of when/if this changes in the future

Where/how? Release notes, on the device page?

JJanssen123, 2021-02-01:
I have recently tested two new BW SHP13's and compared with a Xiaomi plug ZNCZ02LM. I tested with two halogen lamps so there should be no interfering inductive or capacative load. Lamps were on about 10 minutes and there was no change in measured power. I have no others mean of measuring power, so fwiw:

```
            | Xiaomi plug   | BW_SHP13-1        | BW_SHP13-2
Lamp 18 W   | 18.7 W        | 20 W              | 22 W
Lamp 53 W   | 55.67 W       | 62 W              | 62 W
```

It seems the BW SHP13 is indeed somewhat high in power measurement. But I don't think Z2M should correct this in any way. Maybe some warning on the device page indeed. Or in the reviews on AliExpress ;)

vogler, 2021-02-01:
> But I don't think Z2M should correct this in any way. Maybe some warning on the device page indeed. Or in the reviews on AliExpress ;)

Why not? Just mention it on the device page and link this issue.
Fixing it in z2m takes a couple of characters (e.g. `x/1.17`) whereas it may become way more complicated to correct after (e.g. my mqtt values are stored in influxdb, used in node-red, on Android etc.). In general for n different consumers, you have to adjust this value n times (or sub|fix|pub the value on a different topic).

vogler, 2021-03-05:
unstale

frank-f, 2021-03-06:
I got an SHP13 as well. I also have a pretty precise power meter. I measured the SHP13 with a plain old lightbulb connected to it, so power measurement should be as good as it can get. Power meter says 102 W, Zigbee2mqtt reports 115 W.
@Koenkk I'ld like to be able to set an individual calibration value for voltage, power and current, because they're all a bit off, but power really needs it the most. Can you maybe consider that for some future version?
@vogler My correction factor (or divider in your terminology) would be different than yours, so I don't think hardcoding 1.17 would make anyone happier or unhappier than they already are.

vogler, 2021-03-08:
> @vogler My correction factor (or divider in your terminology) would be different than yours, so I don't think hardcoding 1.17 would make anyone happier or unhappier than they already are.

1.13 is still much closer to 1.17 than 1.
Of course there should be options to adjust it, but also a reasonably precise default (e.g. avg of values reported here).

KiPiCoOl, 2021-03-25:
Hi,

My participation :

SHP13 | 1st Wattmeter | 2nd Wattmeter |
329W  |         290W       |         286W        |
662W  |         590W       |         581W        |

vogler, 2021-05-01:
unstale

Dark-Noir, 2021-05-16:
Is it not possible to add device type specific configuration to the configuration.yaml for calibrating this things like with Tasmota https://tasmota.github.io/docs/Power-Monitoring-Calibration/ ?

aw-jansen, 2021-05-18:
Second this ^^ a calibrating feature that would enable you to add, substract and multiply the values would be awesome.

vogler, 2021-06-24:
unstale. This bot is annoying.
@Koenkk is there something similar implemented for another device or do you have some advice how to go at it?

frank-f, 2021-07-26:
unstale

stefangries, 2021-08-24:
I tried the newer BW-SHP15 (external converter [here](https://github.com/Koenkk/zigbee2mqtt/issues/7860)) works much more precisely for me. Nevertheless, there should be a configurable multiplier for the BW-SHP13.

fluffymadness, 2021-08-29:
I second this. Since there is energy monitoring in home assistant now
it would be nice to get a bit more accurate values.

itguy013, 2021-09-29:
unstale

AbeltjeNL, 2021-11-06:
Unstale

adorobis, 2021-12-09:
unstale

adorobis, 2022-01-10:
unstale+

iz3man, 2022-01-19:
Seems I'm not the only one. I never verified the readings until today. There is a power supply connected to the socket which pulls 2.200W from the socket - as stated by the Blitzwolf SHP-13. But the connected battery managment device only sees 1.800W. As well as a clamp current meter, measurng the AC and DC side of the power supply. Could have to do with the PS's power factor, and I will do measurement with a pure ohm' load to verify. But if this is not the case, then I'd say 22% error is not acceptable.

Sungtroll, 2022-01-19:
I did comparition test of my Blitzwolf devices after i noticed the Voltage reading was very far off on some of devices.
Here tested with no-load, 200 Watt load, 400 Watt Load and 800 Watt load. I did not actually test the real load power watt.

[image]

Koenkk, 2022-01-20:
@Sungtroll can you check if it is just an offset? (are we good if a user can set an option to always take the voltage/power/current - 5)?

stefangries, 2022-01-20:
> @Sungtroll can you check if it is just an offset? (are we good if a user can set an option to always take the voltage/power/current - 5)?

Even if I was not addressed, I can tell you that a multiplier would be better than a summand. A correct multiplier for the power value of the SHP13 is about 0.85.

Dark-Noir, 2022-01-20:
> Is it not possible to add a device type specific configuration to the configuration.yaml for calibrating this things like with Tasmota https://tasmota.github.io/docs/Power-Monitoring-Calibration/ ? I mean as a feature ^^

Why not implementing this, instead of a hard multiplier. This should get a much better and precise reading for Current, Voltage and Power.

Sungtroll, 2022-01-20:
I included the voltage differense and i can see it makes no sense to add or substract 5v or 50 mA. The correction factor is different on different devices but correction factor is the right way to do it. My house voltage is different every second so i will use a 12V to 230V pure sine inverter to measure again to get it 100% correct. Tasmota uses power dividing with Volt to Calculate mA, but calculating the mA from devices mA with mA correction factor will do the same or calculating Watt from corrected Volt and corrected Ampere will also work.

[image]

Koenkk, 2022-01-20:
Added this option now, you can set this by clicking on the device in the frontend -> settings (specific). The value is percentual, e.g. filling in -10 will decrease all measurements by 10%.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Sungtroll, 2022-01-20:
I suppose you mean for the power only, if you want to correct the Volt and Ampere too then you have to include correction for those seperatly.

Koenkk, 2022-01-20:
@Sungtroll for all 3, you can configure it seperately.

closed 2022-02-27 by github-actions[bot]

manny199, 2022-10-24:
I have a bit of a weird problem. My Blitzwolf BW-SHP13 plug does not detect watt below 10 watt. Basically, the value it shows with consumption below 8/10 watts is equal to ZERO. 
For example, if I put a webcam it does not detect its consumption in watt. 
Do you have a solution to this problem? The Plug is detected as TS011F_plug_1 and the firmware has been updated. The production is _TZ3000_amdymr7l

NightMean, 2022-11-11:
> I tried the newer BW-SHP15 (external converter [here](https://github.com/Koenkk/zigbee2mqtt/issues/7860)) works much more precisely for me. Nevertheless, there should be a configurable multiplier for the BW-SHP13.

I've ordered couple of BW-SHP15 but it seems the measurement is also not correct. While I know this thread is about active power and not volts, I thought I might as well add my findings to this topic. 
I don't have anything fancy, just a multimeter but I measured constant 240V on the wall socket. The measurements from the plug itself different quite a bit. One plug measured 239V, second 247 and third 250V. 

It'd be really helpful to have some correction for each plug.