Koenkk/zigbee2mqtt #5140: Support For Aurora AOne Two Gang Wireless Battery Rotary Dimmer (AU-A1ZBR2GW)

ashp8i, 2020-11-30:
I have managed to update devices.js in hassos by maintaining a manual copy and was able to pair device and see output when ever the encoder has been turned or pressed alsao there is a secondary mode which I have managed to get the color move function working but I am not sure how I can check that all Scenes, Preset Colors and Cold Warm are supported, also where can we find out if Aurora distribute OTA images, finally unable to check battery status

{
        zigbeeModel: ['2GBatteryDimmer50AU'],
        model: 'AU-A1ZBR2GW',
        vendor: 'Aurora Lighting',
        description: 'AOne Two Gang Wireless Battery Rotary Dimmer',
        supports: 'action',
        fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.greenpower_on_off_switch, fz.command_step, fz.command_move_to_color_temp, fz.command_move_to_color, fz.command_stop, fz.command_move, fz.command_color_loop_set, fz.command_ehanced_move_to_hue_and_saturation, fz.tint404011_scene],
        toZigbee: [tz.light_brightness_step, tz.light_colortemp, tz.light_colortemp, tz.light_color_colortemp],
        exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'color_temperature_move', 'color_move', 'brightness_stop', 'brightness_move_up', 'brightness_move_down', 'color_loop_set', 'enhanced_move_to_hue_and_saturation', 'recall_*', 'scene_*'])],
        ota: ota.zigbeeOTA,
        meta: {multiEndpoint: true, configureKey: 1},
        configure: async (device, coordinatorEndpoint) => { 
        	const endpoint1 = device.getEndpoint(1); 
        	await bind(endpoint1, coordinatorEndpoint, ['genBasic', 'genIdentify','genOnOff', 'genLevelCtrl', 'lightingColorCtrl', 'genPowerCfg']);
        	const endpoint2 = device.getEndpoint(2); 
        	await bind(endpoint2, coordinatorEndpoint, ['genBasic', 'genIdentify','genOnOff', 'genLevelCtrl', 'lightingColorCtrl']);
            const endpoint242 = device.getEndpoint(242); 
            await bind(endpoint242, coordinatorEndpoint, ['greenPower']);
        },

both lqi and action work great
[image]

[image]
[image]

debug 2020-11-30 00:03:47: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:48: Received Zigbee message from 'Ashish Wall Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":95,"currentHue":95}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:48: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":95,"currentHue":95}'
debug 2020-11-30 00:03:48: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:48: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:48: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:48: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:49: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:49: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:49: Received Zigbee message from 'Ashish Wall Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":162,"currentHue":162}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:49: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":162,"currentHue":162}'
debug 2020-11-30 00:03:49: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:49: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:49: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:49: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:49: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:49: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:50: Received Zigbee message from 'Ashish Wall Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":208,"currentHue":208}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:50: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":208,"currentHue":208}'
debug 2020-11-30 00:03:52: Received Zigbee message from 'Ashish Wall Remote', type 'commandMoveToColor', cluster 'lightingColorCtrl', data '{"colorx":0,"colory":0,"transtime":0}' from endpoint 1 with groupID 0
info  2020-11-30 00:03:52: MQTT publish: topic 'zigbee2mqtt/Ashish Wall Remote', payload '{"action":"color_move_1","action_color":{"x":0,"y":0},"action_transition_time":0,"color":{"hue":174,"x":0.3805,"y":0.3769},"color_temp":309,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-11-30 00:03:52: MQTT publish: topic 'zigbee2mqtt/Ashish Wall Remote', payload '{"action":"","color":{"hue":174,"x":0.3805,"y":0.3769},"color_temp":309,"linkquality":42,"update":{"state":"idle"},"update_available":false}'
info  2020-11-30 00:03:52: MQTT publish: topic 'zigbee2mqtt/Ashish Wall Remote/action', payload 'color_move_1'
debug 2020-11-30 00:03:53: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:53: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:53: Received Zigbee message from 'Ashish Wall Remote', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":0,"currentHue":0}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:53: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'attributeReport' and data '{"colorTemperature":0,"currentHue":0}'
debug 2020-11-30 00:03:53: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0
debug 2020-11-30 00:03:53: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
[... cut ...]
debug 2020-11-30 00:03:53: No converter available for 'AU-A1ZBR2GW' with cluster 'lightingColorCtrl' and type 'commandStepColorTemp' and data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}'
debug 2020-11-30 00:03:54: Received Zigbee message from 'Ashish Wall Remote', type 'commandStepColorTemp', cluster 'lightingColorCtrl', data '{"maximum":454,"minimum":153,"stepmode":3,"stepsize":47,"transtime":3}' from endpoint 1 with groupID 0

Would appreciate the help in making this device supported

need to be able to call endpoints Left and Right as labelled on the device endpoint 1 is right and endpoint 2 is the left Knob

Koenkk, 2020-11-30:
You should add `endpoint()` similar to https://github.com/Koenkk/zigbee-herdsman-converters/blob/3a7b95968ea9eec6203e22d603970542bbb44ff3/devices.js#L792 to have `left`/`right` included in the action.

ashp8i, 2020-12-31:
anyway stop this as stale

closed 2021-02-08 by github-actions[bot]