Koenkk/zigbee2mqtt #5141: Innr RB 265 fails to pair with zigbee2mqtt

sagarspatil, 2020-11-30:
# Innr RB 265
## I put Innr RB 265 in pairing mode where it blinks twice after switching it on. I also tried to reset the device multiple times by switching on/off 6 times. The message I see in logs say - 
**Received message from unsupported device with Zigbee model 'undefined'**
The device is supported - [https://www.zigbee2mqtt.io/devices/RB_265.html](url)

## What did you expect to happen
I expected the device to join zigbee2mqtt

## How to reproduce it (minimal and precise)
1. Use Raspberry PI 3B, install Hassio (32 bit)
2. Install zigbee2mqtt addon (I tried it with edge branch too)
3. Switch on/Reset Innr RB 265. Make sure it is in pairing mode (It will blink twice)
4. Check z2m logs and you will see **Received message from unsupported device with Zigbee model 'undefined**

## Debug Info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1 (Zig-a-Zig-ah)
Adapter firmware version: CC26X2R1_20201026

z2m log with zigbee herdsman enabled
https://pastebin.com/K8am4zqd

HA  z2m config 
https://pastebin.com/eTJEkaBR

P.S: Thank you for z2m, I really appreciate everything you are doing.

Koenkk, 2020-11-30:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

sagarspatil, 2020-12-01:
@Koenkk Thanks. I was able to resolve the issue by doing two things, I'm not sure which one helped though.
1. I changed the USB port
2. I moved raspberry pi away from my main network.

closed 2020-12-01 by sagarspatil