Koenkk/zigbee2mqtt #5142: YSR-MINI-Z01 identyfy as ZB-DL01

niqu1982, 2020-11-30:
Again because i'm idiot and made wrong issues
Information about the device + link

..
YSR-MINI-Z01 identyfy as ZB-DL01

https://www.aliexpress.com/item/1005001629650377.html?spm=a2g0s.9042311.0.0.33d95c0foTi2Cd

data/database.db entry of the device

{"id":30,"type":"Router","ieeeAddr":"0x00124b001f945c29","nwkAddr":65008,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"ZB-DL01","epList":[1,13],"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-DL01","manufacturerName":"eWeLink","powerSource":1,"zclVersion":1,"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"20200623","swBuildId":"1.0.6"}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":1,"dateCode":"20200623","swBuildId":"1.0.6","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1606056621638}

Hi i have YSR-MINI-Z01 identyfy as ZB-DL01 i try to add do ode_modules/zigbee-herdsman-converters/devices.js as ZB-DL01 but after restart zigbee2mqtt-edge i lost this. Can anybody tell how to add this model ( i'm probably so stupid to do this)

Koenkk, 2020-11-30:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-30 by Koenkk

niqu1982, 2020-11-30:
Thenx i use dev version

niqu1982, 2020-11-30:
OK there is mistake, this is single channel dimmer. But in interview is rgb+ccw

Koenkk, 2020-12-01:
Fixed,

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)