Koenkk/zigbee2mqtt #5144: Lidl GU10 integration in Home Assistant

goliath888, 2020-11-30:
## What happened

I integrated this GU10 light with no problem. The device is recognized as a TuYa TS0502A model. Can control light brightness and temperature with no problem. 

I have home assistant MQTT integration (auto-discovery enabled by default). 

## What did you expect to happen
I expected that the device will show up in HA, but it did not. I repaired a couple of times, restarted both HA and z2m, but the device will not show up. 

## How to reproduce it (minimal and precise).
Integrate the device in z2m. Wait to appear in HA, but it will not happen. 

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware:  cc2530+cc2591
Adapter firmware version: 20190608

I have the auto-discovery in HA enabled, and lots of other xiaomi sensors working with no issue, so I am not sure why this new device is not showing in HA. 

Is it because the light is recognized as this other device: https://www.zigbee2mqtt.io/devices/TS0502A.html ?

Anjerlaan, 2020-12-01:
Strange.... I have the Candle E14 version of the LIDL, which is recognized as a TuYa TS0505A model in zigbee2mqtt, which is according to the picture a GU10 lamp. https://www.zigbee2mqtt.io/devices/TS0505A.html
This device is also with the IKEA Tradfri HUB recognized as a TuYa TS0505A model 

Is your lamp the color or white version?

goliath888, 2020-12-01:
The white version

Anjerlaan, 2020-12-01:
> The white version

I have the Color version so I can't verify.
If you have a Tradfri HUB, you can try to pair with this hub and what what model is recognized

goliath888, 2020-12-01:
Would be possible to fix this? How can I help?

maichai, 2020-12-02:
I have a Lidl/LivarnoLux GU10 color, it is connected to z2m as:
Model is TS0505A by _TZ3000_kdpxju99.
The icon in z2m Assistant shows a GU10 light.
Also: NO COMPONENTS Maybe this device is not (yet) supported by Zigbee2Mqtt?

In z2m log it reports as 
```
{"dateCode":"","description":"-","friendly_name":"rgb_cct_5w_gu10_test_01","hardwareVersion":1,"ieeeAddr":"0xredacted","lastSeen":1606914174834,"manufacturerID":4098,"manufacturerName":"_TZ3000_kdpxju99","model":"TS0505A","modelID":"TS0505A","networkAddress":redacted,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"}]'`

RGBW Z3","networkAddress":57154,"type":"Router"},{"failed":["lqi"],"friendlyName":"rgb_cct_5w_gu10_test_01","ieeeAddr":"redacted","lastSeen":1606914174834,"manufacturerName":"_TZ3000_kdpxju99","modelID":"TS0505A","networkAddress":redacted,"type":"Router"}]}'

Failed to execute LQI for 'rgb_cct_5w_gu10_test_01'
```

The weird thing is that the Lidl/LivarnoLux E27 RGB-CCT light shows up as (same model!): Model is TS0505A by _TZ3000_dbou1ap4. The Lidl/LivarnoLux E14 RGB-CCT light shows up as: Model is TS0505A by _TZ3000_odygigth. 
For both the icon shown in z2m Assistant is also a GU10 light (although they are E27 and E14 lights).

z2m does not send anything about these devices to the mqtt broker. They seem all not to be supported at the moment.

goliath888, 2020-12-06:
Ok, I have been reading a bit. 

The device is this:
https://zigbeealliance.org/es/zigbee_products/lidl-home-bombilla-led-inteligente-3/

So, I think, this should go in the devices.js:

    {
        fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_dbou1ap4'}],
        model: 'TS0505A',
        vendor: 'Lidl',
        description: 'Livarno Lux GU10',
        extend: generic.light_onoff_brightness_colortemp,
    },

It is the same as the new GU10 RGB, but with the RGB part.

Not sure what to do next. Because if when pairing, the device is recognized as another? how to solve this?

Koenkk, 2020-12-06:
> I expected that the device will show up in HA, but it did not. I repaired a couple of times, restarted both HA and z2m, but the device will not show up.

Can you provide your Zigbee2MQTT debug log when starting zigbee2mqtt? This looks more like a setup problem rather than a Zigbee2MQTT bug.

debuglog

goliath888, 2020-12-06:
> > I expected that the device will show up in HA, but it did not. I repaired a couple of times, restarted both HA and z2m, but the device will not show up.
> 
> Can you provide your Zigbee2MQTT debug log when starting zigbee2mqtt? This looks more like a setup problem rather than a Zigbee2MQTT bug.
> 
> To enable debug logging set in `configuration.yaml`:
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

Of course, here it is:

https://pastebin.com/wJUN90y6

Koenkk, 2020-12-06:
@goliath888 could it be that your homeassistant.js is overridden? (e.g. while adding support for another device some time ago?)

goliath888, 2020-12-06:
> @goliath888 could it be that your homeassistant.js is overridden? (e.g. while adding support for another device some time ago?)

Thank you so much! That was!

The file was not overriden...it just was not updated. I have it in docker, in a mounted file, so I never update it since some time ago. 

Now it's working like a charm.

Thanks again!

closed 2020-12-06 by goliath888

oywino, 2022-08-30:
Hi, my Lidl GU10 HG06492A bulbs is paired easily with deConz and they show up in HA as expected. I can control brightness, but the color temperature is completely crazy. The values seems to span from 148 to 500 in cluster 0300 but only values between 250 and 350 actually changes the color temperature. Anything above or below produces a cold white only.
Has anyone been able to control the color temperature on this bulb correctly ?

hayvan96, 2024-07-01:
> Hi, my Lidl GU10 HG06492A bulbs is paired easily with deConz and they show up in HA as expected. I can control brightness, but the color temperature is completely crazy. The values seems to span [...]

Seems to be the same as #23046