Koenkk/zigbee2mqtt #5148: Devices become unavailable at random time

mustafama, 2020-12-01:
Hi,

I am using Zigbee2mqtt running on docker on raspberry pi for couple of months. Initially I was using raspberry pi zero with Sonoff Zigbee USB Stick. Everything seems to be working fine except that from time to time the Zigbee devices connecting thought Zigbee2mqtt going unavailable at the same time. Most of the time a restart to Zigbee2mqtt would resolve the issue. I switched to Raspberry pi 4B to make sure it is not a hardware issue, still I am having the same issue only not as frequent. At random time the devices becomes unavailable.
I switched debug logging on in Zigbee2mqtt I cannot see anything special. Only MQTT messages received.

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

kevincaradant, 2020-12-01:
Hi

What are the devices which becomes unavailable ? Only devices on battery ? 

Brand ? Xiaomi maybe ? 

Thanks

mustafama, 2020-12-02:
Hello,

Yes I have Three Xiaomi Door sensor but there is also Ikea Zigbee repeater which is also becoming unavailable at the same time.

kevincaradant, 2020-12-02:
Interesting, simple think about it, but are you sure (looking the map of network) that your xiaomi devices are not connected on the router which itself become sometimes unavailable? That could explain the reason of unavailable state of xiaomi devices maybe?

Otherwise I don't know. I just read some xiaomi devices with low battery could have random deconnections and become unstable / unavailable even if they still work after a repairing.

But that's seems to be your AC router which becomes unavailable so no idea.

Wait some better advices from others :)

kevincaradant, 2020-12-02:
Hi

I got exactly the same issue as you today. Don't know why but on my 20 devices Xiaomi, 3 of them stop to answer 5 hours ago.

Really weird. Do you have any idea or advices @Koenkk on what we should try to avoid that ? 

I didn't notice about my temperature sensor but for my motion detection, I directly saw something wrong with some of them.

Also one Wireless Xiaomi Switch D1 and one Wireless Xiaomi mini switch stopped to respond 5 hours ago. (12 minutes of difference). 

It's weird this timing very similar before the unavailable device.

Koenkk, 2020-12-02:
@kevincaradant there are many known reason why Xiaomi devices disconnect, check https://www.zigbee2mqtt.io/devices/WXKG11LM.html#troubleshooting-device-stops-sending-messagesdisconnects-from-network

kevincaradant, 2020-12-02:
Yes Indeed I read them since two hours ( on google, twitter, reddit, github ect). 
No luck for me because I have a good range, full battery and they are connected on the coordinator directly.

So no idea in my case why this happens.

I re-paired them today, I will see if it's happens again or not ;)

mustafama, 2020-12-03:
It seems my issue is different from yours @kevincaradant. In my case all devices connected though the Zigbee2mqtt become unavailable. I checked the connectivity I have two Xiaomi sensors connected through the Ikea repeater and one Xiaomi Sensor connected directly to the Zigbee2mqtt. Also, If I disconnect the repeater, the devices do not become unavailable, only they are not updated inside home assistant. 
I am looking at the Mosquitto logs and I found that Zigbee2mqtt connects to the mqtt broker at the same time this issue happens. I still have to verify this and since this issue occurs only once a day it takes time. The following is the log from Mosquitto broker:

1606970881: New connection from 192.168.8.80 on port 1883.
[INFO] found zigbee2mqtt1 on Home Assistant
1606970881: New client connected from 192.168.8.80 as mqttjs_5d3eafeb (p2, c1, k60, u'zigbee2mqtt1').
1606971262: Saving in-memory database to /data/mosquitto.db.

The following is the log from Zigbee2mqtt with debug logging enabled:

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"stat
us":0,"endpoint":242,"transid":6} +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
0m
Zigbee2MQTT:debug 2020-12-03 04:47:59: Saving state to file /app/data/st
ate.json
Zigbee2MQTT:info  2020-12-03 04:48:01: Connected to MQTT server
  zigbee-herdsman:controller:log Permit joining +3m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq -
 {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,
252,255,254,0,228] +3m

From the zigbee2mqtt log it seems that it is always connected to mqtt the log is full with mqtt messages received.

kevincaradant, 2020-12-03:
Hi

Indeed. What's your HA version? 

When you said, the devices becomes unavailable. Is it in HA or in Zigbee2mqtt? 
If you continue to see the logs, it's a known issue about HA. You have to update it. 

Otherwise I don't know

dudel, 2020-12-03:
@mustafama Are you on zigbee2mqtt 1.16.2 yet? I had the same problem and the cause was that when (for whatever reason) z2m disconnected from MQTT and then reconnected the bridget state was not updated towards HA, thus all devices became unavailable in HA until I restarted z2m. I have not gotten the error again, but it is only two days since 1.16.2...

mustafama, 2020-12-03:
@kevincaradant sorry it was ambiguous. It is not available in home assistant. In Zigbee2mqtt front end I see LQT value more than zero so I assume it is connected. I am not sure 100% how to check the status in zigbee2mqtt.

@dudel that you I will try the update

kevincaradant, 2020-12-03:
OK so you have exactly the same issue than me two weeks ago since I upgraded Z2M over 1.16.0. Its just a related thing with how home assistant manage it now.
I think you're not running on latest (at least 3 latest version of HA), because, for me at least, after to upgraded HA everything works and never got 'unavailable' again. 

But as @dudel said, maybe Z2M is compatible with old and new version for HA now :).
Otherwise, simply update HA and everything should be okay for you.

Let us a comment about what fix your issue. I'm curious to know.

mustafama, 2020-12-07:
Hi,

It seems the new version fixed the issue but not 100%. I still get unavailable status for 10 seconds at a time. Roughtly once a day. It seems to be related to disconnecting from the broker. I see the following in Mosquitto log (my Local time  is GMT+3):

1607349527: Saving in-memory database to /data/mosquitto.db.
1607349600: Socket error on client mqttjs_b2e55212, disconnecting.
1607349604: New connection from 192.168.8.80 on port 1883.
[INFO] found zigbee2mqtt1 on Home Assistant
1607349605: New client connected from 192.168.8.80 as mqttjs_b2e55212 (p2, c1, k60, u'zigbee2mqtt1').

at the same time I find this in the device status in home assistant:

Was opened
17:00:10 - 2 hours ago
Became unavailable
17:00:00 - 2 hours ago
Was opened
14:58:07 - 4 hours ago

kevincaradant, 2020-12-09:
Are you running with the latest version of Z2M and ZA ? 

Don't know what could happens. I didn't any issue like that ( no issue with the broker I want to say)

Be sure to update with latest HA and Z2M ;)

closed 2021-01-09 by Koenkk