Koenkk/zigbee2mqtt #5149: Zemismart M2805EIGB ZB

piotrstanowicz, 2020-12-01:
## Information about the device + link
..
Hi.
I have new device: https://www.zemismart.com/products/zemismart-tuya-zigbee-30-roller-shade-motor-with-battery-for-38mm-tube-smart-life-app-control-alexa-google-home-control

Label on package is "M2805EIGB ZB", Zigbee2MQTT recognized it as Tuya TS0601_curtain -> https://www.zigbee2mqtt.io/devices/TS0601_curtain.html

Device controls works (open, close, stop), but when I try to check position i get "ERROR: No converter available for 'get' 'position' ()"

I'm noob and I don't know much about it. Please help.

## data/database.db entry of the device
..
{"id":19,"type":"Router","ieeeAddr":"0xbc33acfffeda0630","nwkAddr":37815,"manufId":4098,"manufName":"_TZE200_xuzcvlku","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_xuzcvlku","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1606838174759}

Koenkk, 2020-12-01:
Getting the position is not supported for this device.

piotrstanowicz, 2020-12-02:
Yes, yes, it's the same zigbee2rf converter as for this device https://github.com/Koenkk/zigbee-herdsman-converters/issues/1536

Thank you.

closed 2021-01-09 by github-actions[bot]