Koenkk/zigbee2mqtt #5150: Renaming of device breaks HA entities

michalk-k, 2020-12-01:
Not sure it's the issue of: z2m, HA integration or HA itself. Because documentation of z2m mentions mqtt discovery feature, I suppose this this one is built in into z2m. Also because restarting z2m fixes what is broken I assume I'm might be proper place. But again I might be wrong, then please redirect me to right place.

## What happened
When renaming a device which provides more sensors, it very often happens that HA don't pick them all. 
HA recent log records whic are triggered by the change contain information that some sensors cannot be found. The interesting thing is, that those logs points to topic names which existed before name change. 

My quess is it's caused by race condition between z2m rename action and HA trying to pick up new data from discovery topics.
When z2m changes name, HA is somehow triggered to re-read discovery topics from mqtt, but in the meantime zigbee2mqtt changes topic names to new ones. 

Here is the log example from HA:

```
Exception in discovery_callback when dispatching 'mqtt_discovery_updated_('sensor', '0xbc33acfffe63f7ed linkquality')': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/motion1b/availability'}], 'device': {'identifiers': ['zigbee2mqtt_0xbc33acfffe63f7ed'], 'manufacturer': 'Silvercrest', 'model': 'Smart motion sensor (HG06335)', 'name': 'motion1b', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'icon': 'mdi:signal', 'json_attributes_topic': 'zigbee2mqtt/motion1b', 'name': 'motion1b_linkquality', 'state_topic': 'zigbee2mqtt/motion1b', 'unique_id': '0xbc33acfffe63f7ed_linkquality_zigbee2mqtt', 'unit_of_measurement': 'lqi', 'value_template': '{{ value_json.linkquality }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/__init__.py", line 1299, in discovery_callback debug_info.update_entity_discovery_data(self.hass, payload, self.entity_id) File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 89, in update_entity_discovery_data entity_info = hass.data[DATA_MQTT_DEBUG_INFO]["entities"][entity_id] KeyError: 'sensor.motion1_linkquality' 
```

The log has been triggered by changing name from `motion1` to `motion1b`
You can see `motion1b` refered several times, but at the end HA rises exception about `motion1` not found

it's pretty strange because discovery topic is changed in one transaction:
[image]

The might happen even for the first rename from default name. If there are multiple sensors associated with the device, some sensors might fail to be renamed and in result are lost by HA.

What is intresting: broken sensors are being fixed by z2m restart.

## What did you expect to happen
All sensors of device which name is being changed must be renamed and picked up by HA

## How to reproduce it (minimal and precise)
1. Pair new device with z2m which consist of more than 1 sensor (I guess the more sensors the higher probability of failure)
2. Change name of device chosing `Update Home Assistant entity ID` 
at this point some sensors might not recognized by HA (or lost if those were discovered previously)
3. Restart z2m: all HA entities are fixed

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: Conbee II
Adapter firmware version: 2.05.88 / 15. 10. 2020, fw: 26660700
Home Assistant: 0.117.6

Koenkk, 2020-12-02:
Can you provide the debug log when renaming the device?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Erelen-Laiquendi, 2020-12-05:
```
debug 2020-12-05 21:52:21: Received MQTT message on 'zigbee2mqtt/bridge/request/device/rename' with data '{"from":"0x00158d00054d4aad","homeassistant_rename":true,"to":"Room Climate 2"}'
info  2020-12-05 21:52:22: MQTT publish: topic 'zigbee2mqtt/0x00158d00054d4aad', payload ''
info  2020-12-05 21:52:22: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"date_code":"20191205","definition":{"description":"Aqara temperature, humidity and pressure sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"The measured atmospheric pressure","name":"pressure","property":"pressure","type":"numeric","unit":"hPa"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"WSDCGQ11LM","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","msTemperatureMeasurement","msPressureMeasurement","msRelativeHumidity"],"output":["genBasic","genGroups"]}}},"friendly_name":"Room Climate 2","ieee_address":"0x00158d00054d4aad","interview_completed":true,"interviewing":false,"network_address":56839,"power_source":"Battery","software_build_id":"3000-0001","supported":true,"type":"EndDevice"}]'
debug 2020-12-05 21:52:22: Refreshing Home Assistant discovery topic for '0x00158d00054d4aad'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/battery/config', payload 'null'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/temperature/config', payload 'null'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/humidity/config', payload 'null'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/pressure/config', payload 'null'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/linkquality/config', payload 'null'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Room Climate 2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00054d4aad"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Room Climate 2","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/Room Climate 2","name":"Room Climate 2 battery","state_topic":"zigbee2mqtt/Room Climate 2","unique_id":"0x00158d00054d4aad_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Room Climate 2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00054d4aad"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Room Climate 2","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"temperature","json_attributes_topic":"zigbee2mqtt/Room Climate 2","name":"Room Climate 2 temperature","state_topic":"zigbee2mqtt/Room Climate 2","unique_id":"0x00158d00054d4aad_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Room Climate 2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00054d4aad"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Room Climate 2","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"humidity","json_attributes_topic":"zigbee2mqtt/Room Climate 2","name":"Room Climate 2 humidity","state_topic":"zigbee2mqtt/Room Climate 2","unique_id":"0x00158d00054d4aad_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/pressure/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Room Climate 2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00158d00054d4aad"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Room Climate 2","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"pressure","json_attributes_topic":"zigbee2mqtt/Room Climate 2","name":"Room Climate 2 pressure","state_topic":"zigbee2mqtt/Room Climate 2","unique_id":"0x00158d00054d4aad_pressure_zigbee2mqtt","unit_of_measurement":"hPa","value_template":"{{ value_json.pressure }}"}'
info  2020-12-05 21:52:22: MQTT publish: topic 'homeassistant/sensor/0x00158d00054d4aad/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Room Climate 2/availability"}],"device":{"identifiers":["zigbee2mqtt_0 [... cut ...] vice":{"identifiers":["zigbee2mqtt_0x00158d00054d4aad"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Room Climate 2","sw_version":"Zigbee2MQTT 1.16.2"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Room Climate 2","name":"Room Climate 2 linkquality","state_topic":"zigbee2mqtt/Room Climate 2","unique_id":"0x00158d00054d4aad_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
```

Koenkk, 2020-12-06:
@michalk-k @Erelen-Laiquendi can you still reproduce this issue in HA 0.118.5?

Erelen-Laiquendi, 2020-12-06:
HA 0.118.5
z2m 1.16.2 ([this](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt) addon)

Yes, still reproduces. Log about the same.

Koenkk, 2020-12-07:
@emontnemery this look to be a HA bug, Zigbee2MQTT has an option to rename a device and also rename it in HA. When you enable this Zigbee2MQTT will first undiscover the device and then discover it with the new name (see https://github.com/Koenkk/zigbee2mqtt/issues/5150#issuecomment-739335414). I couldn't reproduce locally but it seems there is some race condition in HA here as can be seen in the OP.

emontnemery, 2020-12-08:
@michalk-k @Erelen-Laiquendi can you please reproduce with some additional debug prints in HA:
```yaml
logger:
  default: info
  logs:
    homeassistant.core: debug
    homeassistant.components.mqtt: debug
```

Please also share discovery message before the rename if it's not included in the log.

michalk-k, 2020-12-16:
I can confirm also it happens in HA 0.118.5
Here is log from z2m of addding and renaming Lidl (Silvercrest) door sensor 

```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok"}'
INFO: Device '0xec1bbdfffea0309d' joined
INFO: Starting interview of '0xec1bbdfffea0309d'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffea0309d","ieee_address":"0xec1bbdfffea0309d"},"type":"device_joined"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffea0309d","ieee_address":"0xec1bbdfffea0309d","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xec1bbdfffea0309d"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffea0309d"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffea0309d/availability', payload 'online'
INFO: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffea0309d', payload '{"battery":100,"linkquality":134}'
INFO: Successfully interviewed '0xec1bbdfffea0309d', device has successfully been paired
INFO: Device '0xec1bbdfffea0309d' is supported, identified as: Silvercrest Smart window and door sensor (HG06336)
INFO: Configuring '0xec1bbdfffea0309d'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffea0309d/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffea0309d/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffea0309d"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffea0309d","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffea0309d","name":"0xec1bbdfffea0309d_contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0xec1bbdfffea0309d","unique_id":"0xec1bbdfffea0309d_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffea0309d/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffea0309d/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffea0309d"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffea0309d","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffea0309d","name":"0xec1bbdfffea0309d_battery_low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xec1bbdfffea0309d","unique_id":"0xec1bbdfffea0309d_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffea0309d/tamper/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffea0309d/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffea0309d"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffea0309d","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0xec1bbdfffea0309d","name":"0xec1bbdfffea0309d_tamper","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xec1bbdfffea0309d","unique_id":"0xec1bbdfffea0309d_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0xec1bbdfffea0309d/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffea0309d/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffea0309d"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffea0309d","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffea0309d","name":"0xec1bbdfffea0309d_battery","state_topic":"zigbee2mqtt/0xec1bbdfffea0309d","unique_id":"0xec1bbdfffea0309d_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0xec1bbdfffea0309d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffea0309d/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffea0309d"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffea0309d","sw_version":"Zigbee2MQTT 1.16.2"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffea0309d","name":"0xec1bbdfffea0309d_linkquality","state_topic":"zigbee2mqtt/0xec1bbdfffea0309d","unique_id":"0xec1bbdfffea0309d_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart window and door sensor","exposes":[{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates whether  [... cut ...]
    debug_info.update_entity_discovery_data(self.hass, payload, self.entity_id)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 89, in update_entity_discovery_data
    entity_info = hass.data[DATA_MQTT_DEBUG_INFO]["entities"][entity_id]
KeyError: 'binary_sensor.0xec1bbdfffea0309d_contact' 
```

But I hadn't logging set to debug.
I'll try it yet again.

BTW in case it is HA issue, can you help in reporting that? I'm sure expertize collected and presented by you can be more meaningfulfor HA devs

with regards

emontnemery, 2020-12-16:
@michalk-k HA debug logs with the settings above would be highly appreciated. It does look like a bug in HA for sure.
Is it difficult to reproduce?

michalk-k, 2020-12-16:
no.. I just didn't notice this request.
I'm just after attempt of adding next door sensor but this time everything went without problems. Bad luck ;) Give me few days more.

michalk-k, 2020-12-20:
Finally I found a bit of time to execute the experiment. Again, Lidl Door sensor added to z2m.
Originally, the device is identified by: `0xec1bbdfffe8d2ac4`
Then I renamed it to: `DoorSensorX`

There will be 3 logs: 
z2m for discovery proces
z2m for renaming process
HA log (unrelated messages stripped out, with comment in the middle indicating the place where new name appeared for the first time)

Hope it will help

**Z2m log for joining device**
```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok"}'
INFO: Device '0xec1bbdfffe8d2ac4' joined
INFO: Starting interview of '0xec1bbdfffe8d2ac4'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe8d2ac4","ieee_address":"0xec1bbdfffe8d2ac4"},"type":"device_joined"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xec1bbdfffe8d2ac4","ieee_address":"0xec1bbdfffe8d2ac4","status":"started"},"type":"device_interview"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xec1bbdfffe8d2ac4"},"type":"device_connected"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe8d2ac4"},"type":"pairing"}'
INFO: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe8d2ac4/availability', payload 'online'
INFO: Successfully interviewed '0xec1bbdfffe8d2ac4', device has successfully been paired
INFO: Device '0xec1bbdfffe8d2ac4' is supported, identified as: Silvercrest Smart window and door sensor (HG06336)
INFO: Configuring '0xec1bbdfffe8d2ac4'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffe8d2ac4/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffe8d2ac4"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffe8d2ac4","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","name":"0xec1bbdfffe8d2ac4_contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","unique_id":"0xec1bbdfffe8d2ac4_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffe8d2ac4/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffe8d2ac4"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffe8d2ac4","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","name":"0xec1bbdfffe8d2ac4_battery_low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","unique_id":"0xec1bbdfffe8d2ac4_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
INFO: MQTT publish: topic 'homeassistant/binary_sensor/0xec1bbdfffe8d2ac4/tamper/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffe8d2ac4"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffe8d2ac4","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","name":"0xec1bbdfffe8d2ac4_tamper","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","unique_id":"0xec1bbdfffe8d2ac4_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0xec1bbdfffe8d2ac4/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffe8d2ac4"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffe8d2ac4","sw_version":"Zigbee2MQTT 1.16.2"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","name":"0xec1bbdfffe8d2ac4_battery","state_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","unique_id":"0xec1bbdfffe8d2ac4_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
INFO: MQTT publish: topic 'homeassistant/sensor/0xec1bbdfffe8d2ac4/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4/availability"}],"device":{"identifiers":["zigbee2mqtt_0xec1bbdfffe8d2ac4"],"manufacturer":"Silvercrest","model":"Smart window and door sensor (HG06336)","name":"0xec1bbdfffe8d2ac4","sw_version":"Zigbee2MQTT 1.16.2"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","name":"0xec1bbdfffe8d2ac4_linkquality","state_topic":"zigbee2mqtt/0xec1bbdfffe8d2ac4","unique_id":"0xec1bbdfffe8d2ac4_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart window and door sensor","exposes":[{"access":1,"description":"Indicates if the contact is closed (= true) or open (= false)","name":"contact","property":"contact","type":"binary","value_off":true,"value_on":false},{ [... cut ...]
2020-12-20 14:34:47 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/DoorSensorX, mid: 583
2020-12-20 14:34:47 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/DoorSensorX, mid: 584
2020-12-20 14:34:47 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/DoorSensorX, mid: 585
2020-12-20 14:34:47 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/DoorSensorX, mid: 586
2020-12-20 14:34:47 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/DoorSensorX, mid: 587
```

emontnemery, 2020-12-29:
The problem is due to a race in HA when a discovered entity is removed (by sending an empty discovery message), and then immediately discovered again.

Koenkk, 2020-12-29:
@emontnemery thanks, are there plans to fix this? Otherwise I can do a hacky sleep at the z2m side. (would e.g. 300ms be enough?)

emontnemery, 2020-12-29:
I'll try to get it fixed.

However, how come you do send the empty discovery message?

Koenkk, 2020-12-29:
@emontnemery because otherwise the entity_id is not updated. There is an option to enable/disable this upon renaming 

[image]

emontnemery, 2020-12-29:
OK, got it.

emontnemery, 2021-01-01:
@michalk-k could you give the change in home-assistant/core#44730 a try?

emontnemery, 2021-02-01:
This should be fixed in home-assistant/core#44730 which is included in HA 2021.2.0.

Koenkk, 2021-02-01:
@emontnemery great, many thanks again! I will close this since it's not a z2m issue.

closed 2021-02-01 by Koenkk