Koenkk/zigbee2mqtt #5151: BridgeLegacy problem

aureliosaraiva, 2020-12-01:
## What happened
An error occurs in the logs when I try to pair another device.

```
Zigbee2MQTT:info  2020-11-26 20:29:12: Logging to console and directory: '/app/data/log/2020-11-26.20-29-12' filename: log.txt
Zigbee2MQTT:info  2020-11-26 20:29:13: Starting Zigbee2MQTT version 1.16.1 (commit #6b32f30)
Zigbee2MQTT:info  2020-11-26 20:29:13: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-11-26 20:29:22: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-26 20:29:22: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-11-26 20:29:22: Currently 52 devices are joined:

...

Zigbee2MQTT:info  2020-12-01 16:04:45: Zigbee: allowing new devices to join.

Zigbee2MQTT:error 2020-12-01 16:06:43: Failed to call 'BridgeLegacy' 'onMQTTMessage' (Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```

## Debug info
Zigbee2MQTT version: 1.16.1 
Adapter hardware: CC2531
Adapter firmware version: Source Router

kevincaradant, 2020-12-01:
Hi 

I had it once ( the first time that I paired all my 51 devices. ). I enabled / disabled the join feature 5 or 6 times and 'bim' this error reached on the next try.

A simple unplug / replug coordinator solved this issue and never happens after that.

Or maybe you had reach the maximum devices authorized ? Do you have some "router" devices ? I let you count according to the documentation of Koenkk available here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

Extract for you:
[image]

aureliosaraiva, 2020-12-01:
Router shouldn't be a problem, I have 32 router on the network.

```
Zigbee2MQTT:info  2020-11-26 20:29:12: Logging to console and directory: '/app/data/log/2020-11-26.20-29-12' filename: log.txt
Zigbee2MQTT:info  2020-11-26 20:29:13: Starting Zigbee2MQTT version 1.16.1 (commit #6b32f30)
Zigbee2MQTT:info  2020-11-26 20:29:13: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-11-26 20:29:22: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-26 20:29:22: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2020-11-26 20:29:22: Currently 52 devices are joined:
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d0003230d8f (0x00158d0003230d8f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d0002e6ddb3 (0x00158d0002e6ddb3): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d000394c3d2 (0x00158d000394c3d2): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000006147e (0xf0d1b8000006147e): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000006023d (0xf0d1b8000006023d): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000060ef2 (0xf0d1b80000060ef2): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000061b44 (0xf0d1b80000061b44): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x286d970001081002 (0x286d970001081002): IM6001-MPP01 - SmartThings Multipurpose sensor (2018 model) (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000006229d (0xf0d1b8000006229d): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x286d97000106e3b5 (0x286d97000106e3b5): IM6001-MTP01 - SmartThings Motion sensor (2018 model) (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x286d97000107b88c (0x286d97000107b88c): IM6001-MTP01 - SmartThings Motion sensor (2018 model) (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d000394c437 (0x00158d000394c437): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d000397282e (0x00158d000397282e): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x286d97000106e3d6 (0x286d97000106e3d6): IM6001-MTP01 - SmartThings Motion sensor (2018 model) (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005d30b (0xf0d1b8000005d30b): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b800000618b8 (0xf0d1b800000618b8): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005dff0 (0xf0d1b8000005dff0): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000016750 (0xf0d1b80000016750): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005fd75 (0xf0d1b8000005fd75): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000002fb2e (0xf0d1b8000002fb2e): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b800000f6355 (0xf0d1b800000f6355): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005e0f6 (0xf0d1b8000005e0f6): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000060289 (0xf0d1b80000060289): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000015679 (0xf0d1b80000015679): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b80000061cae (0xf0d1b80000061cae): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005fd53 (0xf0d1b8000005fd53): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d0004a087cf (0x00158d0004a087cf): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b800000f6484 (0xf0d1b800000f6484): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d0002d43f10 (0x00158d0002d43f10): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00124b001c46142a (0x00124b001c46142a): HGZB-043 - Nue / 3A Smart light switch - 3 gang (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00124b001f669083 (0x00124b001f669083): BASICZBR3 - SONOFF Zigbee smart switch (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00124b001f6743e5 (0x00124b001f6743e5): BASICZBR3 - SONOFF Zigbee smart switch (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000006135a (0xf0d1b8000006135a): Not supported (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xf0d1b8000005f532 (0xf0d1b8000005f532): 73693 - Sylvania LIGHTIFY LED RGBW A19 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x04cf8cdf3c773706 (0x04cf8cdf3c773706): GZCGQ01LM - Xiaomi MiJia light intensity sensor (EndDevice)
[... cut ...]
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d00035ac123 (0x00158d00035ac123): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00158d0003432ad3 (0x00158d0003432ad3): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0x00124b001fb4eacb (0x00124b001fb4eacb): GL-S-007Z - Gledopto Smart RGB+CCT 5W GU10 (Router)
Zigbee2MQTT:info  2020-11-26 20:29:22: 0xbc33acfffe616a3c (0xbc33acfffe616a3c): TS0044 - TuYa Wireless switch with 4 buttons (EndDevice)
```

Koenkk, 2020-12-01:
Do I understand correctly that you are running a 52 devices network on a CC2531?

aureliosaraiva, 2020-12-01:
The error stopped after I updated zigbee2mqtt.

But yes, I had 52, now I have 54 devices.

Koenkk, 2020-12-01:
I've seen very few cases were a CC2531 correctly runs a network with 51 devices.

`SRSP - ZDO - mgmtPermitJoinReq after 6000ms` are indications of the adapter getting unstable.

If you encounter more of such issues I recommend upgrading to a more powerful adapter: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

aureliosaraiva, 2020-12-01:
I already ordered a CC2652RB to change the CC2531. As I am from Brazil, it is a bit complicated to get here.

But about the limit, I thought that with more routers on my network, that number should increase. Right?

kevincaradant, 2020-12-01:
the limits of CC2531 is defined by a software limit of X devices without router on the network. 

Using router, indeed you allow to join more devices, you could join on your CC2531 more than one hundred devices (technically)

BUT in reality , your device is cheap with a very few memory and it's powerless as device ( coordinator ). If it's support until now for example (purely random)  3 messages by seconds, adding more devices could increase this number of messages by seconds that it should supported and finally it's will crashed because it's really too much for him.

It depends the devices that you added. My xiaomi aqara devices  talks very few but my plugs talk every seconds and it's very energy-consuming for the coordinator and it's sure with my plugs, your CC2531 will crash :P

aureliosaraiva, 2020-12-02:
I still have activated on my network notification if a device goes offline / online.

aureliosaraiva, 2020-12-02:
Well, now I will need an adapter that supports more than 200 devices.
What can I be using? I have almost 100 devices that I will still pair.

kevincaradant, 2020-12-02:
https://www.zigbee2mqtt.io/information/supported_adapters.html

I'm running under cc1352p-2 in my case.

closed 2021-01-09 by github-actions[bot]