Koenkk/zigbee2mqtt #5152: TuYa TS0601_thermostat becoming non-responsive after a few days

umairyounus, 2020-12-01:
## What happened
TuYa TS0601_thermostat becoming non-responsive after a few days. The is reporting current temperature without any issue but when I try to change the target temperature.

Logs are reporting error:
https://hastebin.com/dehifavutu.apache

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20201026.zip

oopiicaa, 2020-12-02:
I have almost exactly the same setup - Z2M version 1.16.1 (until yesterday when I upgraded to 1.16.2), zzh!'s CC2652R stick with CC26X2R1_20201026 fw and two TS0601 valves. I never noticed these problems. What about setting modes (boost for an ex.) - does this work? Have you tried restarting the system + unplugging/plugging back the stick?

umairyounus, 2020-12-02:
I've upgraded to 1.16.2 and reset pair it, so far so good. It wasn't changing anything even a mode.

umairyounus, 2020-12-04:
A day later it stopped working again and returning the following error:
https://hastebin.com/ixakabimez.js

ycphs, 2020-12-16:
I get a similar message when I try to publish a new `current_heating_setpoint`

Payload is 
`27`
the topic is
`zigbee2mqtt/Thermostat_K/set/current_heating_setpoint`

and my error is 
```Publish 'set' 'current_heating_setpoint' to 'FRIENDLY_NAME' failed: 'Error: Command 0x842e14fffee553fd/1 manuSpecificTuya.setData({"status":0,"transid":28,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,1,14]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - AF - dataRequest after 6000ms)'```

I get the same result with payload
`{"current_heating_setpoint": 27}`
and topic
`zigbee2mqtt/Thermostat_K/set/`

k-zakhariy, 2021-01-19:
I have similar situation, sometime one of my thermostat drops out, i guess that's because low LQI level, because other thermostats have higher connection level : more than 28

oopiicaa, 2021-01-20:
That very well could be the reason (as is with some other devices). Lqi levels on mine valves are > 40. The only problems I've had so far, were because of the low battery, which drain quite fast.

closed 2021-02-27 by github-actions[bot]