Koenkk/zigbee2mqtt #5153: Ikea bulbs won't turn off using scene_recall

nbostrom, 2020-12-01:
## What happened
In our kitchen we use 10w Trådfri drivers and Trådfri GU10 WS bulbs.
I added multiple scenes using the `scene_store` command. One with all lights off and two other scenes with different temps and brightness values. Switching between the two with all the lights on works perfectly, but when I try to switch to the one were all the lights should be off, only the drivers will turn off. I have the same problem in the livingroom where I use one 1000lm Trådfri bulb and two Hue filament bulbs. The filament bulbs will turn off using scene_recall, but the Trådfri bulb wont. 

Z2M do report them as OFF tho...

## What did you expect to happen

I expected all lights to turn off when switching to the scene with all lights off.

## How to reproduce it (minimal and precise)

1. Create a group with a 10W Trådfri driver and a Trådfri GU10 WS bulb
2. Turn both off via Z2M
3. Store a scene for the group with scene_store
4. Turn both on via Z2M
5. Recall the scene with scene_recall
6. Only the driver will turn off.

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: Slaesh stick
Adapter firmware version: 20201026

nbostrom, 2020-12-01:
This is probably a Trådfri firmware issue, right?

sjorge, 2020-12-01:
Have you tried with scene_add? That seems to work for me, with float and gu10 ws bulbs.

nbostrom, 2020-12-01:
Thanks for the tip. I tried this, but with the same result:
```
{
    "scene_add": {
        "ID": 0,
        "state": "OFF"
    }
}
```

sjorge, 2020-12-01:
Have you always used ID 0?

I’ve never used anything < 1, although indent think it should matter.

~ sjorge

nbostrom, 2020-12-01:
It works for the drivers and the Hue lights. It works with the IKEA lights as well, as long as I dont set "state": "OFF", or "brightness": 0. 
Anyhow, I tried turning in off using ID 1 in the livingroom, where I want the ceiling lamp (Trådfri 1000lm) to be off and the "side lamps" to be on in scene 1.

ratenfanger, 2020-12-30:
I have the same problem, but without using the scene functionality of Zigbee2MQTT.

I'm using [iCasa ICZB-KPD18S](https://www.zigbee2mqtt.io/devices/ICZB-KPD18S.html) to create scenes using multiple bulbs:

- [TRADFRI LED bulb E14/E26/E27](https://www.zigbee2mqtt.io/devices/LED1624G9.html) 
- [TRADFRI LED bulb GU10](https://www.zigbee2mqtt.io/devices/LED1537R6_LED1739R5.html)

I bound the remote to the bulbs from the Zigbee2MQTT user interface. Then I created scenes by long pressing a scene button on the remote. 

| Scene | E14 | GU10 |
|-------|-----|------|
| 1     | On  | On   |
| 2     | Off | On   |
| 3     | On  | Off  |

- When I first turn on scene 1 both lights are on as expected. 
- When I then turn on scene 2 the E14 is off and GU10 is on, again as expected.
- When I then turn on scene 3 the E14 goes on but the GU10 stays also on. Strange thing is that Z2M reports the state as being off.

I agree that this looks like a firmware issue. I hope it will be fixed some time soon :) Have you found anything to work around this problem @nbostrom?

nbostrom, 2021-01-04:
@ratenfanger Sorry, but glad to hear there are others.

I haven't had the time to investigate further. [Someone on reddit](https://www.reddit.com/r/tradfri/comments/kiqptr/all_lights_stay_on_whilst_changing_scenes/) had a similar problem using the Trådfri Gateway. He/she fixed it by "adding the bulbs to the scene" first. I don't know what that means in our context. Maybe doing a "scene_add" per bulb?

nbostrom, 2021-01-17:
It works if you add the scene to the bulb one-by-one instead of doing a "scene_store" to the whole group. Closing since this has to be a firmware problem with trådfri and since there are workarounds.

closed 2021-01-17 by nbostrom

EuleMitKeule, 2022-08-01:
I am currently having the exact same problem and I can't figure out how to solve this. @nstadigs could you please explain how you accomplished turning off the tradfri bulbs and other bulbs with the same scene?

nstadigs, 2022-08-08:
Since this came up again I'm thinking that maybe z2m should handle this behind the scenes when doing a scene_store on a group? The Ikea gateway must be doing something similar, no?

nstadigs, 2022-08-08:
> Since this came up again I'm thinking that maybe z2m should handle this behind the scenes when doing a scene_store on a group? The Ikea gateway must be doing something similar, no?

What do you think about this, @Koenkk ?

Koenkk, 2022-08-08:
Preferable we don't do any hacking from the z2m side, maybe the latest IKEA fw solve this?

sjorge, 2022-08-08:
I’m guessing ikea doesn’t use sceneStore but sceneAdd instead.

nstadigs, 2022-08-09:
I remember the Ikea gateway having a similar feature in the app, but as sjorge said it's probably using sceneAdd behind the scenes (no pun intended). I gave this a little more thought and I now agree that solving this in z2m makes little sense.

riskitforabiscuit1252, 2022-08-17:
> I am currently having the exact same problem and I can't figure out how to solve this. @nstadigs could you please explain how you accomplished turning off the tradfri bulbs and other bulbs with the same scene?
> 
> EDIT: Nevermind, I just figured it out. The key was to manually send a scene_add command to every single Tradfri bulb that wouldn't turn off containing the exact same scene ID, scene name, the desired state (OFF) and most importantly the group ID of the group the original scene is assigned to. This is actually exaplained in the [docs](https://www.zigbee2mqtt.io/guide/usage/scenes.html#scene-add).

Can you explain in a little bit more detail how you did this? I'm having the same issue but I can't work out how to set individual bulbs using the group_id feature. No matter what I set the group id number to it ignores this and sets all of the bulbs to off as opposed to just the tradfiri one I wanted. How do you know what this group id is meant to be? Is it a number? Is it the friendly name of the specific tradfri bulb? what is it and how do you get it to work? Here's my Payload that i'm currently entering below. Thanks so much for your guys' help with all this already. Feel like I'm almost there.  

{"scene_add": {"ID": 13, "group_id": 2, "name": "Kam test", "state": "OFF"}}

hvddrift, 2023-08-26:
@riskitforabiscuit1252 your solution has got me 90% on the way to fix my same issues. However, what is the final step to recall the scene? Do you add the bulbs to a group? In other words how do you recall (eg 5 bulbs) scene ?

nstadigs, 2023-08-26:
> @riskitforabiscuit1252 your solution has got me 90% on the way to fix my same issues. However, what is the final step to recall the scene? Do you add the bulbs to a group? In other words how do you recall (eg 5 bulbs) scene ?

You send the scene_recall command to the group by specifying the group friendly name in the topic. See here: https://www.zigbee2mqtt.io/guide/usage/scenes.html#recall-scene

hvddrift, 2023-08-26:
I must be doing something wrong. I can recall the scenes in the individual lights, but as a group recall I cannot do it.

Here are the scenes in my UI, does anything look wrong here?
[image]

and the group (nb, no scenes in the group page are there or can be recalled)
[image]

nstadigs, 2023-08-26:
@hvddrift when adding the scenes you have to provide both group_id and device_id in the payload. There's a section about it in the docs, but it's a little bit hidden I guess: https://www.zigbee2mqtt.io/guide/usage/scenes.html#scene-add

"Equaling scene_store a group_id can be provided when calling scene_add for a single light. In that scenario the state will be applied when the scene is recalled on the group."

Read the full text of both the scene_store and the scene_add docs to get the full picture.

I might make a contribution to make this more clear in the docs if I get the time.

hvddrift, 2023-08-26:
Thank you. I conceptually understand. Are you able to please provide an example? If the blow is the payload I would add to an ikea light (eg: topic zigbee2mqtt/Master Bedroom en-suite sink/set) where or how do I add the "group_id" in the payload? Or do I just have to also make all the device scenes and then a scene_store?

`{
    "scene_add": {
        "ID": 3, // the SCENE_ID
        "name": "Chill scene", // optional: scene name, shown in e.g. the frontend
        "state": "ON", // state, should be 'ON' or 'OFF'

    }
}`

hvddrift, 2023-08-26:
I guess im asking, how do you do this part? "a group_id can be provided when calling scene_add for a single light"

nstadigs, 2023-08-26:
`{
"scene_add": {
"ID": 3,
"name": "Chill scene",
"state": "ON",
"group_id": 1, // Add it to the payload like this
}
}`

hvddrift, 2023-08-28:
After a few days of testing, this is working well. Thank you.