Koenkk/zigbee2mqtt #5154: Moers BHT-002-GCLZB warnings on every message

pgollor, 2020-12-01:
## What happened

I used the [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html) thermostat and i received every two minutes a lot of errors:
```
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":247,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":248,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":249,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":250,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":251,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":252,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":253,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":254,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":255,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":0,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":1,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":2,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":3,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":247,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":248,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":249,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":250,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":251,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":252,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":253,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":254,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":248,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":249,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":250,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":251,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":252,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":253,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":254,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":255,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":0,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":1,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":2,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":3,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":249,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":250,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":251,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":252,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":253,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":254,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":255,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":0,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":1,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":2,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":50,"transid":250,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
WARNING: zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #16 with data {"status":51,"transid":2,"dp":16,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,22]}}
```

## What did you expect to happen
No error messages.

## How to reproduce it (minimal and precise)
Every 2 minutes I got the errors above and also every mqtt status message is repeated 20 times.

## Debug info
Zigbee2MQTT version: 1.16.2 commit: 04c15f7
Adapter hardware: CC2531 (zStack12)
Adapter firmware version: 20190608

Koenkk, 2020-12-01:
@insipiens could you maybe help with this?

vzagaevschi, 2020-12-01:
The same problem happens to me. Can't get current_heating_setpoint (current_heating_setpoint on Exposes from gui is 0), but can send it to device and set it.

insipiens, 2020-12-02:
@vzagaevschi what version of Z2M are you running? 

pulling in @drzony for comment only....

DP:16 should be the current heating setpoint for moes according to the Dev 

i.e New dev version for setting target temperature is calculated as :512+16 = old Z2M version: 528

Could this be a firmware update?

vzagaevschi, 2020-12-02:
> @vzagaevschi what version of Z2M are you running?
1.16.2 and 1.16.2-dev
Problems occurred after the update from 1.15

pgollor, 2020-12-02:
I received the device yesterday, so I could only test it with the current version 1.16.2.

vzagaevschi, 2020-12-02:
On 1.16.1 version it is ok.

drzony, 2020-12-02:
@pgollor @vzagaevschi there is a mixup in data point for this device, the PR above should fix it. If you want, you can try to modify fromZigbee.js yourself and try it out (check my PR on what to change)

drzony, 2020-12-02:
@insipiens thanks for pulling me in :)

closed 2020-12-02 by Koenkk

LolbitAha, 2020-12-14:
Sorry if I sound stupid, just dont get this whole Github format... Is the issue resolved? Or will it be? Cant seem to use the fix proposed above....

drzony, 2020-12-14:
It was merged and it is in latest dev version. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

LolbitAha, 2020-12-14:
> It was merged and it is in latest dev version. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Thank you kind person!!!! Ill be waiting for it to arrive to stable version then!!