Koenkk/zigbee2mqtt #5155: ZNGZDJ11LM: running=true after stop

Erelen-Laiquendi, 2020-12-01:
## What happened
Sending close command - and curtain starts closing:
```
2020-12-01 22:48:16 zigbee2mqtt/Room Curtain/set 'CLOSE'
2020-12-01 22:48:16 zigbee2mqtt/Room Curtain '{"linkquality":120,"options":{"hand_open":false,"reverse_direction":true},"position":100,"running":true}'
```
Sending stop command - and curtain stops:
```
2020-12-01 22:48:18 zigbee2mqtt/Room Curtain/set 'STOP'
2020-12-01 22:48:18 zigbee2mqtt/Room Curtain '{"linkquality":147,"options":{"hand_open":false,"reverse_direction":true},"position":98,"running":false}'
```
And `running=false`. Ok. But after several seconds there is new message:
```
2020-12-01 22:48:20 zigbee2mqtt/Room Curtain '{"linkquality":147,"options":{"hand_open":false,"reverse_direction":true},"position":94,"running":true}'
```
with final position and `running=true` :( And that's all - no new messages with `running=false`. So, curtain is stopped, but running=true.

And when I don't stop it manually, but it reach final position (totally opened, closed or target position - z2m don't send `running=false` at all:
```
2020-12-02 00:26:28 zigbee2mqtt/Room Curtain/set b'{ "position": 4 }'
2020-12-02 00:26:28 zigbee2mqtt/Room Curtain b'{"linkquality":132,"options":{"hand_open":false,"reverse_direction":true},"position":0,"running":true}'
2020-12-02 00:26:31 zigbee2mqtt/Room Curtain b'{"linkquality":129,"options":{"hand_open":false,"reverse_direction":true},"position":5,"running":true}'
```

```
2020-12-01 22:48:02 zigbee2mqtt/Room Curtain/set 'OPEN'
2020-12-01 22:48:02 zigbee2mqtt/Room Curtain '{"linkquality":150,"options":{"hand_open":false,"reverse_direction":true},"position":88,"running":true}'
2020-12-01 22:48:08 zigbee2mqtt/Room Curtain '{"linkquality":114,"options":{"hand_open":false,"reverse_direction":true},"position":100,"running":true}'
```

So, after every change I see `running=true` after curtain stops.

Problem is stable (reproduses every try).

## What did you expect to happen
See `running=false` when curtain don't move.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: slaesh’s CC2652RB stick
Adapter firmware version: Z-Stack_3.x.0 20201026

Erelen-Laiquendi, 2021-01-01:
Last "Edge" - problem still exists.

Koenkk, 2021-01-01:
Can you provide the debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Erelen-Laiquendi, 2021-01-01:
Sending open command:
```
debug 2021-01-01 19:43:12: Received MQTT message on 'zigbee2mqtt/Room Curtain/set' with data 'OPEN'
debug 2021-01-01 19:43:12: Publishing 'set' 'state' to 'Room Curtain'
debug 2021-01-01 19:43:13: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440527,"presentValue":49}' from endpoint 1 with groupID 0
info  2021-01-01 19:43:13: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":99,"options":{"hand_open":false,"reverse_direction":true},"position":49,"running":true}'
debug 2021-01-01 19:43:14: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID 0
```
After several seconds sending stop command:
```
debug 2021-01-01 19:43:23: Received MQTT message on 'zigbee2mqtt/Room Curtain/set' with data 'STOP'
debug 2021-01-01 19:43:23: Publishing 'set' 'state' to 'Room Curtain'
debug 2021-01-01 19:43:23: Received Zigbee message from 'Room Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":51}' from endpoint 1 with groupID 0
info  2021-01-01 19:43:23: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":87,"options":{"hand_open":false,"reverse_direction":true},"position":51,"running":false}'
debug 2021-01-01 19:43:25: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2021-01-01 19:43:25: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":63025664,"presentValue":77}' from endpoint 1 with groupID 0
info  2021-01-01 19:43:25: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":87,"options":{"hand_open":false,"reverse_direction":true},"position":77,"running":true}'
```
As you can see, immideatly after stop, at 19:43:23, there is `"running":false` in mqtt state. But after two more seconds - at 19:43:25 - `"running":true` again.

Or without manual stop.

Open command:
```
debug 2021-01-01 19:44:29: Received MQTT message on 'zigbee2mqtt/Room Curtain/set' with data 'OPEN'
debug 2021-01-01 19:44:29: Publishing 'set' 'state' to 'Room Curtain'
debug 2021-01-01 19:44:29: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440542,"presentValue":77}' from endpoint 1 with groupID 0
info  2021-01-01 19:44:29: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":102,"options":{"hand_open":false,"reverse_direction":true},"position":77,"running":true}'
debug 2021-01-01 19:44:31: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID 0
```
Curtain reached full open state:
```
debug 2021-01-01 19:44:39: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":0}' from endpoint 1 with groupID 0
debug 2021-01-01 19:44:39: Received Zigbee message from 'Room Curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":63025664,"presentValue":100}' from endpoint 1 with groupID 0
info  2021-01-01 19:44:39: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":123,"options":{"hand_open":false,"reverse_direction":true},"position":100,"running":true}'
```
But still `"running":true`.

I can receive `"running":false` only after sending stop command to already stopped curtain:
```
debug 2021-01-01 19:45:28: Received MQTT message on 'zigbee2mqtt/Room Curtain/set' with data 'STOP'
debug 2021-01-01 19:45:28: Publishing 'set' 'state' to 'Room Curtain'
debug 2021-01-01 19:45:29: Received Zigbee message from 'Room Curtain', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":100}' from endpoint 1 with groupID 0
info  2021-01-01 19:45:29: MQTT publish: topic 'zigbee2mqtt/Room Curtain', payload '{"linkquality":135,"options":{"hand_open":false,"reverse_direction":true},"position":100,"running":false}'
```

Koenkk, 2021-01-01:
Probably fixed now, can you confirm?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Erelen-Laiquendi, 2021-01-02:
Yes, now it's ok.
Thanks!

closed 2021-01-02 by Erelen-Laiquendi