Koenkk/zigbee2mqtt #5156: Tuya switch 4 gang device

aureliosaraiva, 2020-12-01:
## What happened
I'm trying to pair a new device. It even pairs correctly.

However the operation is strange.

The switch has 4 buttons, and only shows 2 buttons on the HA.

This is the recognized device.

```js
{
        zigbeeModel: ['qnazj70', 'kjintbl'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_wunufsil'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_vhy3iakz'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_oisqyl4o'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_aqnazj70'}, // 4 gang
        ],
        model: 'TS0601_switch',
        vendor: 'TuYa',
        description: '1, 2, 3 or 4 gang switch',
        exposes: [
            e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
            e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4'),
        ],
        fromZigbee: [fz.tuya_switch, fz.ignore_basic_report, fz.tuya_switch2],
        toZigbee: [tz.tuya_switch_state],
        meta: {configureKey: 1, multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Norklmes', model: 'MKS-CM-W5'},
            {vendor: 'Somgoms', model: 'ZSQB-SMB-ZB'},
            {vendor: 'Moes', model: 'WS-EUB1-ZG'},
        ],
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(2)) await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(4)) await bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
        },
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
        },
    },
```

```
Zigbee2MQTT:info  2020-12-01 18:29:43: Device '0x5c0272fffec1fbd0' joined
Zigbee2MQTT:info  2020-12-01 18:29:43: Starting interview of '0x5c0272fffec1fbd0'
Zigbee2MQTT:info  2020-12-01 18:29:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x5c0272fffec1fbd0"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-12-01 18:29:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffec1fbd0"},"type":"pairing"}'
Zigbee2MQTT:info  2020-12-01 18:29:43: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0/availability', payload 'online'
Zigbee2MQTT:info  2020-12-01 18:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x5c0272fffec1fbd0"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: Successfully interviewed '0x5c0272fffec1fbd0', device has successfully been paired
Zigbee2MQTT:info  2020-12-01 18:29:46: Device '0x5c0272fffec1fbd0' is supported, identified as: TuYa 1, 2, 3 or 4 gang switch (TS0601_switch)
Zigbee2MQTT:info  2020-12-01 18:29:46: Configuring '0x5c0272fffec1fbd0'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"1, 2, 3 or 4 gang switch","friendly_name":"0x5c0272fffec1fbd0","model":"TS0601_switch","supported":true,"vendor":"TuYa"},"type":"pairing"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'homeassistant/switch/0x5c0272fffec1fbd0/switch_l1/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x5c0272fffec1fbd0/availability"}],"command_topic":"zigbee2mqtt/0x5c0272fffec1fbd0/l1/set","device":{"identifiers":["zigbee2mqtt_0x5c0272fffec1fbd0"],"manufacturer":"TuYa","model":"1, 2, 3 or 4 gang switch (TS0601_switch)","name":"0x5c0272fffec1fbd0","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","name":"0x5c0272fffec1fbd0_l1","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","unique_id":"0x5c0272fffec1fbd0_switch_l1_zigbee2mqtt","value_template":"{{ value_json.state_l1 }}"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'homeassistant/switch/0x5c0272fffec1fbd0/switch_l2/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x5c0272fffec1fbd0/availability"}],"command_topic":"zigbee2mqtt/0x5c0272fffec1fbd0/l2/set","device":{"identifiers":["zigbee2mqtt_0x5c0272fffec1fbd0"],"manufacturer":"TuYa","model":"1, 2, 3 or 4 gang switch (TS0601_switch)","name":"0x5c0272fffec1fbd0","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","name":"0x5c0272fffec1fbd0_l2","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","unique_id":"0x5c0272fffec1fbd0_switch_l2_zigbee2mqtt","value_template":"{{ value_json.state_l2 }}"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'homeassistant/switch/0x5c0272fffec1fbd0/switch_l3/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x5c0272fffec1fbd0/availability"}],"command_topic":"zigbee2mqtt/0x5c0272fffec1fbd0/l3/set","device":{"identifiers":["zigbee2mqtt_0x5c0272fffec1fbd0"],"manufacturer":"TuYa","model":"1, 2, 3 or 4 gang switch (TS0601_switch)","name":"0x5c0272fffec1fbd0","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","name":"0x5c0272fffec1fbd0_l3","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","unique_id":"0x5c0272fffec1fbd0_switch_l3_zigbee2mqtt","value_template":"{{ value_json.state_l3 }}"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'homeassistant/switch/0x5c0272fffec1fbd0/switch_l4/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x5c0272fffec1fbd0/availability"}],"command_topic":"zigbee2mqtt/0x5c0272fffec1fbd0/l4/set","device":{"identifiers":["zigbee2mqtt_0x5c0272fffec1fbd0"],"manufacturer":"TuYa","model":"1, 2, 3 or 4 gang switch (TS0601_switch)","name":"0x5c0272fffec1fbd0","sw_version":"Zigbee2MQTT 1.16.2"},"json_attributes_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","name":"0x5c0272fffec1fbd0_l4","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","unique_id":"0x5c0272fffec1fbd0_switch_l4_zigbee2mqtt","value_template":"{{ value_json.state_l4 }}"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: MQTT publish: topic 'homeassistant/sensor/0x5c0272fffec1fbd0/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x5c0272fffec1fbd0/availability"}],"device":{"identifiers":["zigbee2mqtt_0x5c0272fffec1fbd0"],"manufacturer":"TuYa","model":"1, 2, 3 or 4 gang switch (TS0601_switch)","name":"0x5c0272fffec1fbd0","sw_version":"Zigbee2MQTT 1.16.2"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","name":"0x5c0272fffec1fbd0_linkquality","state_topic":"zigbee2mqtt/0x5c0272fffec1fbd0","unique_id":"0x5c0272fffec1fbd0_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2020-12-01 18:29:46: Successfully configured '0x5c0272fffec1fbd0'

```

This is the log when each button is pressed, the fourth button does not generate any log.
```
Zigbee2MQTT:info  2020-12-01 18:37:50: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":99,"state_l1":"ON","state_l2":"OFF","state_l3":"ON","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:37:52: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":107,"state_l1":"ON","state_l2":"ON","state_l3":"ON","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:37:54: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":107,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","undefined":"OFF"}'
```

[image]

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.2
Adapter hardware: CC2531
Adapter firmware version: Source Router

Koenkk, 2020-12-01:
- Can you provide a link to the device? (aliexpress e.g.)
- Can you provide the data/database.db entry?

aureliosaraiva, 2020-12-01:
[image]

It seems that in HA solved, this is showing in the interface. I didn't do anything different.

As I understand it, through the HA interface I can control the l4 button (4 gang)

However when I click on the device it does not send an event to the zigbee2mqtt of the change.

https://pt.aliexpress.com/item/4001181765345.html?spm=a2g0s.9042311.0.0.22f9b90ajazA7q

```
Zigbee2MQTT:info  2020-12-01 18:49:33: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":81,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:33: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":89,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:33: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":89,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:33: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":99,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:34: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":99,"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:34: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":94,"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"OFF","undefined":"OFF"}'
Zigbee2MQTT:info  2020-12-01 18:49:35: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffec1fbd0', payload '{"linkquality":94,"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"ON","undefined":"OFF"}'
```

Koenkk, 2020-12-02:
Can you try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

ARafaelSF, 2020-12-02:
> Can you try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

I have the same issue. The same device (same store no aliexpress).

It works on hub tuya.  Buttons 1, 2, 3 work correctly.
Button 4, I can control it through the Home Assistant. But if I trigger it on the device, I don't see a message in the log. If I activate another button then, I see that its status has not changed in the message.

1 - INFO: MQTT publish: topic 'zigbee2mqtt/Interruptor Sala de Jantar', payload '{"linkquality":97,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
2 - INFO: MQTT publish: topic 'zigbee2mqtt/Interruptor Sala de Jantar', payload '{"linkquality":92,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
3 - INFO: MQTT publish: topic 'zigbee2mqtt/Interruptor Sala de Jantar', payload '{"linkquality":84,"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"OFF","undefined":"OFF"}'
4 - INFO: MQTT publish: topic 'zigbee2mqtt/Interruptor Sala de Jantar', payload '{"linkquality":92,"state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","undefined":"OFF"}'
5 - INFO: MQTT publish: topic 'zigbee2mqtt/Interruptor Sala de Jantar', payload '{"linkquality":84,"state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"ON","undefined":"OFF"}'

Change Switch 3 in device (Messagem 2)
Change Switch 3 in device (Messagem 3)
Change Switch 4 in device (nothing happens)
Change Switch 3 in device (Messagem 4) _ Gang 4 still off (in device is on)
Change Switch 4 in HA (Messagem 5)

ARafaelSF, 2020-12-03:
> However when I click on the device it does not send an event to the zigbee2mqtt of the change.

Did you solved this? My device send event for tree gang, but not for the 4 gang.

pdeprost, 2020-12-19:
Exactly the same problem for me.  Button 4 physical actions are not reported in state_l4

oferwald, 2020-12-25:
Same problem here with the 4th button not sending an event, the "manufName" is "_TZE200_aqnazj70".
Maybe this should be a different ticket? any way to provide useful debug information? Is there a way to debug a single device?

Koenkk, 2021-01-24:
Let's continue here: https://github.com/Koenkk/zigbee2mqtt/issues/4597

closed 2021-01-24 by Koenkk